alsa-project/alsa-lib #6: 1.18: alsa-lib build fails with LTO

kloczek, 2019-03-30:
<pre>make[2]: Entering directory &apos;/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.1.8/src&apos;
/bin/sh ../libtool  --tag=CC   --mode=link gcc  -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto -version-info 2:0:0 -Wl,--version-script=Versions  -Wl,--no-undefined -Wl,-z,relro -Wl,--as-needed  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -flto -fuse-linker-plugin -o libasound.la -rpath /usr/lib64 conf.lo confmisc.lo input.lo output.lo async.lo error.lo dlmisc.lo socket.lo shmarea.lo userfile.lo names.lo control/libcontrol.la mixer/libmixer.la pcm/libpcm.la timer/libtimer.la rawmidi/librawmidi.la hwdep/libhwdep.la seq/libseq.la ucm/libucm.la topology/libtopology.la  -lm -ldl -lpthread -lrt 
libtool: link: gcc -shared  -fPIC -DPIC  .libs/conf.o .libs/confmisc.o .libs/input.o .libs/output.o .libs/async.o .libs/error.o .libs/dlmisc.o .libs/socket.o .libs/shmarea.o .libs/userfile.o .libs/names.o  -Wl,--whole-archive control/.libs/libcontrol.a mixer/.libs/libmixer.a pcm/.libs/libpcm.a timer/.libs/libtimer.a rawmidi/.libs/librawmidi.a hwdep/.libs/libhwdep.a seq/.libs/libseq.a ucm/.libs/libucm.a topology/.libs/libtopology.a -Wl,--no-whole-archive -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -O2 -g -fstack-protector-strong -grecord-gcc-switches -m64 -mtune=generic -flto -Wl,--version-script=Versions -Wl,--no-undefined -Wl,-z -Wl,relro -Wl,--as-needed -Wl,-z -Wl,now -flto -fuse-linker-plugin    -lm -ldl -lpthread -lrt -Wl,-soname -Wl,libasound.so.2 -o .libs/libasound.so.2.0.0
/tmp/ccROQp1D.s: Assembler messages:
/tmp/ccROQp1D.s: Error: invalid attempt to declare external version name as default in symbol `snd_dlopen@@ALSA_1.1.6&apos;
lto-wrapper: fatal error: gcc returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
</pre>

perexg, 2019-04-09:
Fixed in 5366bdb4fbc2d569522ca8cf6a161190d2f76e98 . Let me know, if it works for you.

kloczek, 2019-05-06:
No, still it does not work.
(FTR: full build log is in attachment)
[alsa-lib.log](https://github.com/alsa-project/alsa-lib/files/3147773/alsa-lib.log)

<pre>make[2]: Entering directory &apos;/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.1.8/src&apos;
/bin/sh ../libtool  --tag=CC   --mode=link gcc  -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto -version-info 2:0:0 -Wl,--version-script=Versions  -Wl,--no-undefined -Wl,-z,relro -Wl,--as-needed  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -flto -fuse-linker-plugin -o libasound.la -rpath /usr/lib64 conf.lo confmisc.lo input.lo output.lo async.lo error.lo dlmisc.lo socket.lo shmarea.lo userfile.lo names.lo control/libcontrol.la mixer/libmixer.la pcm/libpcm.la timer/libtimer.la rawmidi/librawmidi.la hwdep/libhwdep.la seq/libseq.la ucm/libucm.la topology/libtopology.la  -lm -ldl -lpthread -lrt 
libtool: link: gcc -shared  -fPIC -DPIC  .libs/conf.o .libs/confmisc.o .libs/input.o .libs/output.o .libs/async.o .libs/error.o .libs/dlmisc.o .libs/socket.o .libs/shmarea.o .libs/userfile.o .libs/names.o  -Wl,--whole-archive control/.libs/libcontrol.a mixer/.libs/libmixer.a pcm/.libs/libpcm.a timer/.libs/libtimer.a rawmidi/.libs/librawmidi.a hwdep/.libs/libhwdep.a seq/.libs/libseq.a ucm/.libs/libucm.a topology/.libs/libtopology.a -Wl,--no-whole-archive -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -O2 -g -fstack-protector-strong -grecord-gcc-switches -m64 -mtune=generic -flto -Wl,--version-script=Versions -Wl,--no-undefined -Wl,-z -Wl,relro -Wl,--as-needed -Wl,-z -Wl,now -flto -fuse-linker-plugin    -lm -ldl -lpthread -lrt -Wl,-soname -Wl,libasound.so.2 -o .libs/libasound.so.2.0.0
/tmp/cc1Rv6SH.s: Assembler messages:
/tmp/cc1Rv6SH.s: Error: invalid attempt to declare external version name as default in symbol `snd_dlopen@@ALSA_1.1.6&apos;
lto-wrapper: fatal error: gcc returned 1 exit status
</pre>

tiwai, 2019-05-06:
It's a known issue of gcc.  LTO still doesn't support the versioned symbols.
  https://gcc.gnu.org/bugzilla/show_bug.cgi?id=48200

We may put some workaround, but I'd rather wait for the upstream resolution in the gcc side:
  https://stackoverflow.com/questions/46304742/how-to-combine-lto-with-symbol-versioning

perexg, 2019-05-06:
@kloczek : Use hints from gitcompile:

<pre>
gcc: -flto -flto-partition=none
export AR="gcc-ar"
export RANLIB="gcc-ranlib"
</pre>

kloczek, 2019-05-06:
So what is the difference with using -flto -flto-partition=none and -flto -fuse-linker-plugin?

perexg, 2019-05-06:
I don't know. This magic can explain only a gcc developer. You may ask in the mentioned gcc bug.

kloczek, 2020-02-17:
This issue can be closed now as -flto-partition=none seems is working now.

closed 2020-02-17 by kloczek

unhappy-ending, 2022-06-21:
Hi, I'd like to report that -flto-partition=one works fine, you don't have disable -flto-partition outright since you can use a single one. I guess it probably doesn't matter in the case of GCC if using no partition or a single partition, so long as external symbol versioning can be seen.

On the Clang side of things, you have to use -flto=full for single threaded LTO since when using multithreaded -flto=thin you get a similar problem. Clang can't see everything using split and threaded LTO unlike when using a single threaded and not split code generation during LTO.

Adding some Gentoo build logs.

[alsa-lib-clang-flto-full.log](https://github.com/alsa-project/alsa-lib/files/8946730/alsa-lib-clang-flto-full.log)
[alsa-lib-clang-flto-thin.log](https://github.com/alsa-project/alsa-lib/files/8946731/alsa-lib-clang-flto-thin.log)
[alsa-lib-gcc-flto-partition-one.log](https://github.com/alsa-project/alsa-lib/files/8946732/alsa-lib-gcc-flto-partition-one.log)
.