alsa-project/alsa-lib #7: aplay - pcm_read: Input/output error

mrajwa, 2019-05-31:
Hello,

I came across an issue with custom built aplay (system one works OK on Ubuntu 18.04):

I have fetched both alsa-lib and alsa-utils repositories, built them now try to run:

`path_to_aplay/aplay -C -D hw:0,8 -r 16000 -f S16_LE -c 2 tmp.wav -vvv`

Immediately after the start it prints hw_params and the and error message saying:

`arecord: pcm_read:2103: read errorL Input/output error`

As I said system aplay works fine. Moreover I compared hw_params for both system and this custom built aplay and they are exactly the same!

I also tried to built alsa-utils from debian package here I did:

apt source alsa-lib && sudo apt build-dep alsa-utils && fakeroot debian/rules clean && fakeroot debian/rules build && fakeroot debian/rules binary

All finished fine, now I repeated above steps for alsa-utils and here is aproblem at fakeroot debian/rules binary step. At some point it stops with error message saying:

`dpkg-shlibdeps: error: no dependency information found for /usr/lib/libasound.co.s2 (used by debian/alsa-utils/usr/bin/alsaucm)`

Does anybody have any idea what is going on here?

cladisch, 2019-05-31:
Apparently, something went wrong with the custom build. But you have not provided any information about that.

I do not use Debian, but I'd guess that compiling alsa-lib and alsa-utils in two different sandboxes prevents the second one from finding the first one.

perexg, 2019-10-14:
I am sorry, we cannot help you with the custom build. Use the mailing list or the distribution specific forum.

closed 2019-10-14 by perexg