alsa-project/alsa-lib #12: Record at 384 KHz

paddygoat, 2019-10-10:
Tried everything I can think of to record at 384. How do I do it?

Thanks!

paddygoat, 2019-10-14:
Any ideas?

perexg, 2019-10-14:
What's the exact problem? What this command does output:

<pre>
arecord -v -c 2 -r 384000 -f S16_LE a.wav
</pre>

paddygoat, 2019-10-14:
paddy@nobody:~$ arecord -v -c 2 -r 384000 -f S16_LE a.wav
arecord: main:619: bad speed value 384000
paddy@nobody:~$ arecord -v -c 2 -r 192000 -f S16_LE a.wav
Recording WAVE 'a.wav' : Signed 16 bit Little Endian, Rate 192000 Hz, Stereo
ALSA <-> PulseAudio PCM I/O Plugin
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 192000
  exact rate   : 192000 (192000/1)
  msbits       : 16
  buffer_size  : 96000
  period_size  : 24000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 24000
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 96000
  silence_threshold: 0
  silence_size : 0
  boundary     : 6755399441055744000

perexg, 2019-10-14:
Upgrade alsa-utils to 1.1.6+.

paddygoat, 2020-01-19:
sudo apt-get install libncurses5-dev libncursesw5-dev
wget ftp://ftp.alsa-project.org/pub/utils/alsa-utils-1.1.7.tar.bz2
tar xvjf alsa-utils-1.1.7.tar.bz2
cd alsa-utils-1.1.7
./configure
make && sudo make install

perexg, 2020-01-20:
aplay --version