alsa-project/alsa-lib #22: When storing settings which contain a [ or ], the setting is not quoted.

Bertware, 2020-01-10:
I am using a Raspberry Pi 1 B+ with a Hifiberry DAC+ADC Pro HAT.
This hat contains a PCM5122 Stereo DAC and an ADC. Now to the issue: The ADC has parameters such as `VINR1[SE]` and `VINR2[SE] + VINR1[SE]`. When using `alsactl store` or`alsactl restore`, restoring will fail due to an invalid file.

`ALSA lib conf.c:1887:(_snd_config_load_with_include) _toplevel_:354:30:Unexpected char`

This error points to the parameter `VINR1[SE]`, which is not surrounded by quotes. The other parameters, which also contain other special characters such as spaces, are surrounded by quotes.

An excerpt of the file is included below. This clearly illustrates which names were encapsulated between quotes.

```

control.28 {
                iface MIXER
                name 'ADC Right Capture Source'
                value VINR1[SE]
                comment {
                        access 'read write'
                        type ENUMERATED
                        count 1
                        item.0 'No Select'
                        item.1 VINR1[SE]
                        item.2 VINR2[SE]
                        item.3 'VINR2[SE] + VINR1[SE]'
                        item.4 VINR3[SE]
                        item.5 'VINR3[SE] + VINR1[SE]'
                        item.6 'VINR3[SE] + VINR2[SE]'
                        item.7 'VINR3[SE] + VINR2[SE] + VINR1[SE]'
                        item.8 VINR4[SE]
                        item.9 'VINR4[SE] + VINR1[SE]'
                        item.10 'VINR4[SE] + VINR2[SE]'
                        item.11 'VINR4[SE] + VINR2[SE] + VINR1[SE]'
                        item.12 'VINR4[SE] + VINR3[SE]'
                        item.13 'VINR4[SE] + VINR3[SE] + VINR1[SE]'
                        item.14 'VINR4[SE] + VINR3[SE] + VINR2[SE]'
                        item.15 'VINR4[SE] + VINR3[SE] + VINR2[SE] + VINR1[SE]'
                        item.16 '{VIN2P, VIN2M}[DIFF]'
                        item.17 '{VIN3P, VIN3M}[DIFF]'
                        item.18 '{VIN2P, VIN2M}[DIFF] + {VIN3P, VIN3M}[DIFF]'
                }
        }
```

Creation of the output data and writing it to a file is handled by the save_state method in alsactl: https://github.com/alsa-project/alsa-utils/blob/master/alsactl/state.c#L1539
This code in turn calls snd_config_save in the alsa-lib project: https://github.com/alsa-project/alsa-lib/blob/b20b400e2f598c86abaf697d66396cecd49b3e14/src/conf.c#L3080
Then further to string_print: https://github.com/alsa-project/alsa-lib/blob/b20b400e2f598c86abaf697d66396cecd49b3e14/src/conf.c#L1508
And then in this loop which decides which characters need to be quoted: https://github.com/alsa-project/alsa-lib/blob/b20b400e2f598c86abaf697d66396cecd49b3e14/src/conf.c#L1523

*Proposed solution*
The characters `[` and `]` should be a reason to put text between quotes.
These characters should be added to the switch statement in print_string here:
https://github.com/alsa-project/alsa-lib/blob/b20b400e2f598c86abaf697d66396cecd49b3e14/src/conf.c#L1535

Regards,
Bert

Bertware, 2020-01-13:
Applied in https://github.com/alsa-project/alsa-lib/commit/a78dd1bd1e9b47c2aee9b1e066de515b6854695a

closed 2020-01-13 by Bertware

PanderMusubi, 2020-02-22:
See also https://support.hifiberry.com/hc/en-us/community/posts/360009452418-DAC-ADC-PRO-issue-alsactl?page=1#community_comment_360001645217