alsa-project/alsa-lib #24: MIDI 2.0 universal packet protocol

symdeb, 2020-01-14:
Since MIDI 2.0 only support USB and non MIDI DIN (for now), will MIDI 2.0 need a new or updated OS class driver for the 32/64 bit universal packet interface and device descriptor for devices ? Page 16 of the USB MIDI class driver spec defines support for 1,2 or 3 bytes transfers only, How will this have to be implemented by MIDI USB device manufacturers and well as on the Linux OSes side ?  How will this impact Linux ALSA ? Would the first byte of the message with b7=0 not confuse the driver of running status ? Reference for packet format : [https://www.midi.org/articles-old/details-about-midi-2-0-midi-ci-profiles-and-property-exchange](url)

tiwai, 2020-02-10:
USB MIDI 2.0 protocol isn't defined yet, so it's hard to judge.  But, yes, we'll need to extend the USB MIDI driver code accordingly.  And no, we can't do anything right now unless the USB MIDI 2.0 spec is published.

ALSA rawmidi is nothing but the dumb byte stream, so at most we might need the alignment per 32bit packet size.  A bigger change would be ALSA sequencer, and this would need a new event type for MIDI 2.0 packet.  But the existing ALSA sequencer event has already 96 bits at least (128 bits on 64bit arch), so it should fit within the current scheme.

Meanwhile, the support of MIDI 2.0 CI is a totally different question.  It's still unclear to me how the whole MIDI-CI things can be implemented in details, so it's a bit difficult to judge right now.

closed 2020-02-10 by tiwai

symdeb, 2020-02-11:
Hi Takashi-san,

USB MIDI 2.0 was approved on NAMM 2020. The specs are on MIDI.ORG (not a
PDF, just a webpage). Not sure if that is the whole spec.
Not sure if they are going to publish a formal spec publicly soon. The CI
spec is published in more detailed already.
AFAIK CI is 1.0 SYSEX but does list a command sequence to negotiate devices
to switch from 1.0 to 2.0. I would assume that 2.0 mode is moving to UMP
mode. It is shown n the presentation as well.

BTW MIDI,org just replied: *MIDI 2.0 uses a Universal MIDI Packet (UMP)
format for MIDI 1.0 Protocol messages and MIDI 2.0 Protocol messages. There
is a Version 2 of the USB Device Class Definition for MIDI Devices in
development to support the UMP format. It will require new class drivers.
All of the major OS vendors are aware of the progress. I think the ALSA
team is also aware. *

P.S: For using  JACK2 using MIDI ports , JACK need the ALSA "seq" mode. Wo
would assume JACK also needs those changes then ? It's not all that
backwards compatible as it seems. Not sure why they chose UMP instead of
just an SYSEX extension At high speed interfaces (USB) a byte more or less
does not matter that much.

Cheers.

tiwai, 2020-02-11:
Thanks, it's an interesting information.
Unfortunately we have neither contact nor information about MIDI 2.0 at all, so the support on ALSA (Linux kernel / user-space) isn't expected right now.  Could you ask them for helping the implementation on Linux?
It might be that Google or other vendor have worked on it, but still we need to know who is doing what for now.

symdeb, 2020-02-11:
Hello Takashi-san.

Thanks, How should MIDI reach out to the ALSA developer community or key
contact to provide information ?  at  alsa-lib@noreply.github.com ?
Not sure MIDI.org will providing code or changes for a updated or new ALSA
driver. :-)
I am not sure if UMP needs different device descriptor but it looks like
that.
The USB MIDI Gadgets driver for embedded devices is probably thus also
impacted.
If a new descriptor is defined then the Roland A-88.mk2 must have support
this such new descriptor then.
https://www.roland.com/us/products/a-88mk2/

tiwai, 2020-02-11:
The easiest would be to just get me (Takashi Iwai <tiwai@suse.de>) in the loop, then I'll delegate to other parties.  Of course we don't expect that they develop or submit changes by themselves :)  It's only about the specs and the details.

Also another question is whether other groups have already started working on Linux (e.g. Android or ChromeOS) enablement.  If yes, we'd better to build up a working group.

symdeb, 2020-02-12:
Hi Takashi-san,

Ok, let me see if I can your email address to them. BTW statement about OS
support here @36:00
https://www.youtube.com/watch?v=zXnHaoN2Cig&list=PLe2skUvADfhvu_pyet1veIIEAH0LA4iFK&index=15&t=0s

symdeb, 2020-02-12:
Hi Takashi-san,

Ok, let me see if I can your email address to them. BTW statement about OS
support here @36:00
https://www.youtube.com/watch?v=zXnHaoN2Cig&list=PLe2skUvADfhvu_pyet1veIIEAH0LA4iFK&index=15&t=0s

Interestingly, MIDI 2.0 was wrapped up in standard SYSEX and then send to a
Cubase receiver that unpacked the Sysex and then send to a VST.
Wished UMP would just be a Sysex extension then all the device driver and
OS changes could have been avoided perhaps. there must be a reason for
choising a new package format.
Who knows there will be a MIDI 1.0, UMP sysex extension someday :-)

symdeb, 2020-02-12:
Sent a message to MIDI.ORG

BTW statement about OS support here @36:00
https://www.youtube.com/watch?v=zXnHaoN2Cig&list=PLe2skUvADfhvu_pyet1veIIEAH0LA4iFK&index=15&t=0s

Interestingly, MIDI 2.0 was wrapped up in standard SYSEX and then send to a
Cubase receiver that unpacked the Sysex and then send to a VST.
Wished UMP would just be a Sysex extension then all the device driver and
OS changes could have been avoided perhaps. there must be a reason for
choising a new package format.
Who knows there will be a MIDI 1.0, UMP sysex extension someday :-)

cladisch, 2020-02-12:
I was contacted two years ago, but it was not possible to get an NDA in place for various reasons.

So far, the information available is:
> There is a Version 2 of the USB Device Class Definition for MIDI Devices in development to support the UMP format. It will require new class drivers.

tiwai, 2020-02-12:
> I was contacted two years ago, but it was not possible to get an NDA in place for various reasons.

Oh that's interesting.  Let me know if you get further contact.

symdeb, 2020-02-13:
I wish i could help on the driver/kernel.. Just started to work on customizing embedded ARM SoC based Linux kernels for USB hosts and devices, but assume when ALSA for x64 is updated for 2.0 embedded Linux can automatically benefit from that work.  I could well help with testing though.

saji8k, 2020-07-01:
The USB MIDI 2.0 Specification has now been released:
https://www.usb.org/sites/default/files/USB%20MIDI%20v2_0.pdf

The MIDI 2.0 specs can be found here (You will have to create a free account first to view):
https://www.midi.org/midi2

symdeb, 2020-07-05:
Good news ! Now need USB MIDI 2.0 Gadget support for embedded devices :-)

symdeb, 2020-07-25:
The USB MIDI 2.0 spec has been released a few weeks ago now.
MIDI 2.0 uses an alternate setting for the interface.
This means that the device can support Legacy MIDI 1.0 and 2.0 but not
both a the same time. (MIDI 1.0 as in legacy MIDI USB, not the UMP MIDI
1 mode) How will this work for ALSA  Will the ALSA driver send a
request to the device to switch back and forth. Would a  user (or
program) be able to (force) switch the interface back to MIDI 1 ?  How
would that work for the device to know which mode to work in ?
Working on A MIDI 2.0 device (FS), but it will need ALSA code to
debug/play with. Would ALSA raw be first candidate to develop ?

symdeb, 2020-08-15:
Have a device now with the descriptors to support MIDI 2.0. Now need driver software to test it against with 
Any idea when work on this will commence ? Is there a possibility for a MIDI 2.0 branch ?

DocSunset, 2021-03-27:
Just curious if there has been any progress on ALSA support for MIDI 2.0? This issue was closed before the specs were available, so I suppose the fact that it remains closed doesn't imply that the feature is supported. I also don't see any commits or pull requests that mention MIDI.

How are things progressing now that the core MIDI 2.0 specs and the USB MIDI 2.0 spec are publicly accessible?

tiwai, 2022-08-23:
Does anyone have a real or a virtual MIDI 2.0 device?  I'm interested in how the USB descriptor looks like, as a starter.

symdeb, 2022-08-24:
The descriptor details are in the USB spec.
https://www.usb.org/sites/default/files/USB%20MIDI%20v2_0.pdf
Dev kit: MIDI 2.0 devkit: https://amenote.com/?page_id=74  (available 4Q22)
I created a simple USB MIDI 2.0 Full speed device a back in 2020 but there was no  USB host/driver to verify the desc is correct
Android 13 should be able to help with that.

symdeb, 2022-08-25:
[descriptor.txt](https://github.com/alsa-project/alsa-lib/files/9424479/descriptor.txt)