alsa-project/alsa-lib #25: ALSA raw MIDI API - client hangup and amidi exits midiC3D0 failed: Device or resource busy

MusicMaker, 2020-02-19:
While developing a client using the raw MIDI interface, bumped into an issue that
went away after rebooting Ubuntu 19.10 - 64 bit. Using a MIDISPORT 4x4 one of the 4 ports could not be opened. The client (written in C) call hang and the client thus went unresponsive. No timeout, no error, no nothing. CTRL-BREAK the client and retrying many times gave the exact same result, no other apps were using the port. The other 3 ports on the interface and any other MIDI USB device connected could be opened and used without any problem.

When opening the port with amidi, an error was reported, it seems amidi did not gracefully exit as well.

amidi -receive -p hw:3,0,1
ALSA lib rawmidi_hw.c:233:(snd_rawmidi_hw_open) open /dev/snd/midiC3D0 failed: Device or resource busy
cannot open port "hw:3,0,1": Device or resource busy

Looking at the rawmidi_hw.c code there is a GOTO and SYSERR statement. Not sure if those caused this .

   if (fd < 0) {
            snd_ctl_close(ctl);
            SYSERR("open %s failed", filename);
            return -errno;
.
After reboot the problem went away. The problem could have been caused while fixing some by coding issue. Maybe someone with more understanding of the code can spot an issue in the code. While  many applications use the raw MIDI interface. believe Bitwig and most others use the sequencer API to open ports.

Just wanted to log this just in case anyone else experiences this

perexg, 2020-02-19:
Could you run gdb / strace to determine the exact syscall where the program hangs?

takaswie, 2020-02-19:
Hi @MusicMaker 

> cannot open port "hw:3,0,1": Device or resource busy

When in-kernel driver registers sound card, the rawmidi device reserved by the driver is added to Linux system. At the same time, in-kernel client of ALSA sequencer is also added[1].

I suspect that the in-kernel client of ALSA sequencer is used by any application, then your application is going to corrupt due to EBUSY from the call of open(2) to rawmidi device since the in-kernel client uses the device.

For example:

```
$ amidi -l
Dir Device    Name
IO  hw:2,0,0  003Rack MIDI 1
 O  hw:2,0,1  003Rack MIDI 2
```

```
$ aplaymidi -l
 Port    Client name                      Port name
 14:0    Midi Through                     Midi Through Port-0
 24:0    003Rack                          003Rack MIDI 1
 24:1    003Rack                          003Rack MIDI 2
```

The client '24:0' in ALSA Sequencer is for the device 'hw:2,0,0' for rawmidi interface, therefore they can't be used at the same time by applications:

```
$ aplaymidi -p 24:0 ~/Desktop/yousei.mid &
[1] 31986
$ amidi -p hw:2,0,0 -s /dev/urandom
ALSA lib rawmidi_hw.c:233:(snd_rawmidi_hw_open) open /dev/snd/midiC2D0 failed: Device or resource busy
cannot open port "hw:2,0,0": Device or resource busy
```

I thinks you have the similar case. I'd like you to check connections between clients of ALSA Sequencer and check no clients establishes connections for the client of your device.

```
$ aconnect -l
...
```

[1] https://git.kernel.org/pub/scm/linux/kernel/git/tiwai/sound.git/tree/sound/core/rawmidi.c#n1796

symdeb, 2020-02-19:
Was able to create the device busy situation by just running midiloop.c example. 
`/a.out -i hw:5,0,0 -o hw:6,0,0`
When pressing CTRL-Z to  stop midiloop(it ddid not show/receive any data) and then start it again it shows 
> `ALSA lib rawmidi_**hw.c:233:**(snd_rawmidi_hw_open) open /dev/snd/midiC5D0 failed: Device or resource busy`
> snd_rawmidi_open hw:5,0,0 failed: -16

SYSERR is near line 233 so believe that must be the call. I then ran my own code  under gdb and when it hanged at opening the input port  use CTRL-C and showed
```
0x00007ffff7daafdb in __libc_open64 (file=0x7fffffffd530 "/dev/snd/midiC2D0", oflag=524288)
    at ../sysdeps/unix/sysv/linux/open64.c:48
48	../sysdeps/unix/sysv/linux/open64.c: No such file or directory.
```
After unplugging the USB MIDI device when it is stuck, the code continues with is a port opening error)  It's not specific to the MIDISPORT.  When running the code without CTRL-C  (not closing the ports correctly, it seems th is is culprit) the stuck issues do not occur and reading/writing is all well. Why would there not be a timeout and is the perhaps way to recover from such port busy situation ?
Any suggestions to use gdb or strace to pin down where it hangs would be nice as well.

MusicMaker, 2020-02-19:
Was able to create the device busy situation by just running midiloop.c example. 
`/a.out -i hw:5,0,0 -o hw:6,0,0`
When pressing CTRL-Z to  stop midiloop(it ddid not show/receive any data) and then start it again it shows 
> `ALSA lib rawmidi_**hw.c:233:**(snd_rawmidi_hw_open) open /dev/snd/midiC5D0 failed: Device or resource busy`
> snd_rawmidi_open hw:5,0,0 failed: -16

SYSERR is near line 233 so believe that must be the call. I then ran my own code  under gdb and when it hanged at opening the input port  use CTRL-C and showed
```
0x00007ffff7daafdb in __libc_open64 (file=0x7fffffffd530 "/dev/snd/midiC2D0", oflag=524288)
    at ../sysdeps/unix/sysv/linux/open64.c:48
48	../sysdeps/unix/sysv/linux/open64.c: No such file or directory.
```
After unplugging the USB MIDI device when it is stuck, the code continues with is a port opening error)  It's not specific to the MIDISPORT.  When running the code without CTRL-C  (not closing the ports correctly, it seems th is is culprit) the stuck issues do not occur and reading/writing is all well. Why would there not be a timeout and is the perhaps way to recover from such port busy situation ?
Any suggestions to use gdb or strace to pin down where it hangs would be nice as well.

perexg, 2020-02-19:
If you don't open the device without SND_RAWMIDI_NONBLOCK, then there is the wait loop inside kernel until the device is available again. You may check which task owns the device using `lsof /dev/snd/midi*` command. If there is no task and you cannot use the device, it's probably a kernel driver issue. Also, as @takaswie suggested, you should look if the sequencer use the specific device.

MusicMaker, 2020-02-20:
Thanks for that !  Not sure if I understand this "If you **don't** open the device **without** SND_RAWMIDI_NONBLOCK, but for what it is worth, all ports are opened in non-blocking mode and no other (user) application is running (that I know of) that using ALSA. amidi seems to exit with a failure . Using my code and using the API the call waits forever and an timeout/error return form the call would be fine, An indefinite wait loop isn't good from user experience point of view. Anyhow will look into this further.  P.S:  so far, I have not had any issues like this using JACK2.

cladisch, 2020-02-20:
The `/proc/asound/cardX/midiY` file should show the PID of the process that keeps the file open. Is there one?

MusicMaker, 2020-02-22:
This issue can be closed if not deleted, Totally embarrassed to find that, due to a silly mistake, the port got actually opened in blocked mode. However, when  the port is open by a process due to a crashed application. It seems the only way to get it open is to kill the process. A kind of "closed_forced" function would be nice to force ALSA kill the process for the prtt. Apologies for wasting your time.