alsa-project/alsa-lib #26: usage of snd_rawmidi_status_get_avail()

MusicMaker, 2020-02-20:
In non blocked raw MIDI mode, is snd_rawmidi_status_get_avail supposed to return the number of available bytes than can read via snd_rawmidi_read. (port is opened for read only)  It seems always return 0 even there is MIDI data when using a read.  Else is there a(nother) way to find the number of unread received bytes in the buffer?
snd_rawmidi_status_alloca (&status);  	
snd_rawmidi_status(client, status); 
snd_rawmidi_status_get_avail(status);

cladisch, 2020-02-20:
`snd_rawmidi_status_get_avail()` indeed should return the number of bytes that can be read.

Please check that `snd_rawmidi_status()` did not return an error.

How do you determine that there is MIDI data? What are the contents of `/proc/asound/cardX/midi0` at that point of time?

symdeb, 2020-02-22:
It now works, only it seems at least one read must be done to the port, before avail will return values  > 0, otherwise it always will return 0. I used avail before the read and if was zero the port was never read.

symdeb, 2020-02-22:
So far debugging this, it looks like at least one read to the port must be done before avail will return values > 0.

MusicMaker, 2020-02-22:
So far debugging this, it looks like at least one read to the port must be done before avail() will return values > 0

cladisch, 2020-02-22:
Yes, you need to actually read bytes (or poll for them) to trigger the driver to actually read from the hardware.

symdeb, 2020-02-27:
Ok, Thanks , with that, it works and could get multiple bytes in a red. Closed.

MusicMaker, 2020-02-27:
Ok, Thanks , with that, it works and could get multiple bytes in a read.
P.S should avail return bytes that can be written to the output buffer (before it is full ) ?

cladisch, 2020-02-27:
Yes, "available" means bytes that can be read or written.

closed 2020-04-28 by perexg