alsa-project/alsa-lib #29: snd_rawmidi_params_set_no_active_sensing()

MusicMaker, 2020-02-27:
Does this filter out receiving 0xFE incoming and outgoing messages when enabled ? Still see incoming active sensing messages after using this.

takaswie, 2020-02-27:
> Does this filter out receiving 0xFE incoming and outgoing messages when enabled ? Still see incoming active sensing messages after using this.

The API `snd_rawmidi_params_set_no_active_sensing()` is just to set flag to the data of `snd_rawmidi_params_t` structure. The substream of rawmidi is configured for it when the call of `snd_rawmidi_params()` with the data.

cladisch, 2020-02-27:
The documentation is not very clear.

This setting enables or disables sending of a single Active Sensing message when the stream is closed.