alsa-project/alsa-lib #34: How to make ALSA relocatable in the filesystem?

probonopd, 2020-03-07:
We want to bundle a private copy of ALSA with our application, so that the ALSA from the Linux distribution is not used at all. Yet it seems ALSA is relying on some absolute paths compiled in with no clear way to pass in a different location for ALSA at runtime. 

How to do this?

Reference:
https://github.com/mumble-voip/mumble/issues/3959#issuecomment-596095335

probonopd, 2020-03-07:
It appears that https://github.com/flatpak/freedesktop-sdk-images/blob/1.6/alsa-lib-plugin-path.patch might be addressing this?

perexg, 2020-06-04:
Commited to master.

closed 2020-06-04 by perexg

bkuhls, 2020-06-13:
This patch breaks building alsa-lib with uClibc:
```
dlmisc.c:79:51: error: ‘RTLD_DL_LINKMAP’ undeclared (first use in this function)
   if (dladdr1(&snd_dlpath, &info, (void**)&links, RTLD_DL_LINKMAP) == 0)
dlmisc.c:81:38: error: ‘RTLD_DI_ORIGIN’ undeclared (first use in this function)
   if (links != NULL && dlinfo(links, RTLD_DI_ORIGIN, origin) == 0) {
```
RTLD_DL_LINKMAP & RTLD_DI_ORIGIN, are unsupported on uClibc:
https://cgit.uclibc-ng.org/cgi/cgit/uclibc-ng.git/tree/include/dlfcn.h#n106

perexg, 2020-06-13:
> This patch breaks building alsa-lib with uClibc:
>
> ```
> dlmisc.c:79:51: error: ‘RTLD_DL_LINKMAP’ undeclared (first use in this function)
>    if (dladdr1(&snd_dlpath, &info, (void**)&links, [...]

Why uclibc build defines __GLIBC__ ? https://github.com/alsa-project/alsa-lib/blob/6eb78d41a1688b14b458af4a57e7bcd92120c9fe/src/dlmisc.c#L72

If you have a better way to check non-glibc build, we can change it.