alsa-project/alsa-lib #35: Sample project request...

mnini, 2020-03-11:
Can you make sample of alsa processor that is have input and output in the code?

I know only syntax of C and I want to program DSP for linux output sound.

flatmax, 2020-03-11:
There is a good example of ALSA Full duplex here :
https://github.com/flatmax/gtkiostream/blob/master/test/ALSAFullDuplexTest.C

It is written in C++, however you can specify your DSP code in the process method :
https://github.com/flatmax/gtkiostream/blob/master/test/ALSAFullDuplexTest.C#L37

There is a thread on the people@audioinjector.net email list on how to compile it.
https://lists.audioinjector.net/pipermail/people/2020-March/000021.html

perexg, 2020-06-03:
https://github.com/alsa-project/alsa-lib/tree/master/test

closed 2020-06-03 by perexg