alsa-project/alsa-lib #38: Regression: Unity3D games — Crash on launch if pulseaudio is not running

vv221, 2020-03-16:
Multiple Unity3D games crash on launch if you try to launch them while pulseaudio is not running. For games using the `ScreenSelector.so` plugin, the crash happen after the screen resolution selection.

It seems to only happen on systems using Mesa + amdpgu.

The crash happens with `alsa-lib` v1.2.2, but does not happen with `alsa-lib` 1.1.8. This allowed me to run a regression test and find the faulty commit:
```
commit 5ee5ef31b5ff3fb7c904054cb9cac7478a727f7c
Author: Jaroslav Kysela <perex@perex.cz>
Date:   Sun Dec 1 14:26:40 2019 +0100

    namehint: correct the @args check
    
    BugLink: https://github.com/alsa-project/alsa-plugins/issues/3
    Signed-off-by: Jaroslav Kysela <perex@perex.cz>

diff --git a/src/control/namehint.c b/src/control/namehint.c
index 808df6b5..4927ef97 100644
--- a/src/control/namehint.c
+++ b/src/control/namehint.c
@@ -348,6 +348,12 @@ static int try_config(snd_config_t *config,
                goto __cleanup;
        if (snd_config_search(res, "@args", &cfg) >= 0) {
                snd_config_for_each(i, next, cfg) {
+                       /* skip the argument list */
+                       snd_config_get_id(snd_config_iterator_entry(i), &str);
+                       while (*str && *str >= '0' && *str <= '9') str++;
+                       if (*str == '\0')
+                               continue;
+                       /* the argument definition must have the default */
                        if (snd_config_search(snd_config_iterator_entry(i),
                                              "default", NULL) < 0) {
                                err = -EINVAL;
```
From what I understand, this commit has been used to fix the following issue: https://github.com/alsa-project/alsa-plugins/issues/3

The following patch applies cleanly on tag `v1.2.2` and current `master`, and fixes this crash:
```
From a236046a14b438c96889a46cf8baf38b19b7b45e Mon Sep 17 00:00:00 2001
From: vv221 <vv221@dotslashplay.it>
Date: Mon, 16 Mar 2020 16:05:24 +0100
Subject: [PATCH] Revert 5ee5ef31b5ff3fb7c904054cb9cac7478a727f7c "namehint:
 correct the @args check"

This fixes a regression trigerring a crash with Unity3D games if
pulseaudio is not running.

Research about this crash can be found on ./play.it issues tracker:
https://forge.dotslashplay.it/play.it/games/issues/355
---
 src/control/namehint.c | 7 -------
 1 file changed, 7 deletions(-)

diff --git a/src/control/namehint.c b/src/control/namehint.c
index 60c48ae3..808df6b5 100644
--- a/src/control/namehint.c
+++ b/src/control/namehint.c
@@ -348,13 +348,6 @@ static int try_config(snd_config_t *config,
                goto __cleanup;
        if (snd_config_search(res, "@args", &cfg) >= 0) {
                snd_config_for_each(i, next, cfg) {
-                       /* skip the argument list */
-                       if (snd_config_get_id(snd_config_iterator_entry(i), &str) < 0)
-                               continue;
-                       while (*str && *str >= '0' && *str <= '9') str++;
-                       if (*str == '\0')
-                               continue;
-                       /* the argument definition must have the default */
                        if (snd_config_search(snd_config_iterator_entry(i),
                                              "default", NULL) < 0) {
                                err = -EINVAL;
-- 
2.25.1
```
It is probably not ideal, as I guess it would restore the regression it was supposed to fix.

---

The original research about this regression can be found on ./play.it issues tracker: [Unity3D + amgdpu (Mesa) — Crash on launch if pulseaudio is not running](https://forge.dotslashplay.it/play.it/games/issues/355)

perexg, 2020-03-16:
Already fixed in https://github.com/alsa-project/alsa-lib/issues/27 via https://github.com/alsa-project/alsa-lib/commit/09e78da4cade7b8fecb8f36717b85c456f2e5487 .

The used sound routines are weird and probably buggy.

closed 2020-03-16 by perexg

vv221, 2020-03-16:
My bad, I tried the latest tag (`v1.2.2`) but forgot to try current master…

vv221, 2020-03-16:
@perexg 

I still get the crash described here with a current master build, including https://github.com/alsa-project/alsa-lib/commit/09e78da4cade7b8fecb8f36717b85c456f2e5487. But I do not get it if I apply the reverting of https://github.com/alsa-project/alsa-lib/commit/5ee5ef31b5ff3fb7c904054cb9cac7478a727f7c on top of master.

Please consider reopening this issue, as I do not think it is a real duplicate of #27.

vv221, 2020-03-16:
In case it may help, is here the diff I get in `aplay -L` output after reverting https://github.com/alsa-project/alsa-lib/commit/5ee5ef31b5ff3fb7c904054cb9cac7478a727f7c (left is current master, right is current master + reverted commit):
```
null                                                            null
    Discard all samples (playback) or generate zero samples (       Discard all samples (playback) or generate zero samples (
lavrate                                                       <
    Rate Converter Plugin Using Libav/FFmpeg Library          <
samplerate                                                    <
    Rate Converter Plugin Using Samplerate Library            <
speexrate                                                     <
    Rate Converter Plugin Using Speex Resampler               <
jack                                                            jack
    JACK Audio Connection Kit                                       JACK Audio Connection Kit
oss                                                           <
    Open Sound System                                         <
pulse                                                         <
    PulseAudio Sound Server                                   <
upmix                                                         <
    Plugin for channel upmix (4,6,8)                          <
vdownmix                                                      <
    Plugin for channel downmix (stereo) with a simple spacial <
default:CARD=SB                                                 default:CARD=SB
    HDA ATI SB, VT1708S Analog                                      HDA ATI SB, VT1708S Analog
    Default Audio Device                                            Default Audio Device
```
Output is truncated, everything after the `default:CARD=SB` entry is the same on both sides.

perexg, 2020-03-16:
And it's correct. I'd be happy to fix a bug in alsa-lib, but the bug is really in the app code. See the list which the app produce with the debug output: https://github.com/alsa-project/alsa-lib/issues/27#issuecomment-592920831 . It looks like a wrong memory management which ends with the crash in some cases.

You can disable listing of those plugins via alsa-lib's configuration files (remove hint {} section), but then the you won't see them in all apps. Or you can make a private alsa-lib configuration for this app using the ALSA_CONFIG_DIR environment variable. There are several ways to override the alsa-lib's configuration.

vv221, 2020-03-17:
I could not find a way to provide application-specific configuration overrides without having to set up a full ALSA configuration, did I miss something in the documentation? Being able to set a custom path for `~/.asoundrc` through a variable environment is the kind of thing I am looking for, but I had no success so far.

It seems `ALSA_CONFIG_DIR` and `ALSA_CONFIG_PATH` would require to first copy the full ALSA configuration, meaning that any change happening to the shipped default configuration afterwards would be ignored. I does not seem like a reasonable way to set application-specific settings.

---

The only workaround I have so far is to hide `libasound2.so` from the application, using a trick similar to this:
```
mkdir libs
ln --symbolic /dev/null libs/libasound.so.2
LD_LIBRARY_PATH="libs:$LD_LIBRARY_PATH" ./application_binary
```
But I have no complete understanding of the side effects it could have. Actually I was quite surprised to notice that applications play sound without any issue with this trick, I was expecting to work around the crash at the cost of no sound in application.

perexg, 2020-03-17:
The ~/.asoundrc is loaded via /usr/share/alsa/alsa.conf (look to the start).

If the app does not require alsa-lib (linked to pulseaudio, too?), you may try something like:

````ALSA_CONFIG_PATH=/tmp aplay  -L````

vv221, 2020-03-17:
Well, that is the tricky part I have some difficulties to understand.

To give some more context:
* The test system I use (a Debian Bullseye/Sid) exposes `libasound.so.2` in its 1.2.2 version;
* pulseaudio is not installed (`libpulse.so.0` being exposed or not has no noticeable effect);
* the applications have no hard requirement on alsa-lib;
* all tested applications are games using the Unity3D engine, some of them 32-bit and other ones 64-bit.

If I use no workaround, I get the described crash. Using a current alsa-lib build from master instead of the 1.2.2 build provided by my distribution brings no change, I still get the crash.

If I set `ALSA_CONFIG_PATH` to a path with not valid configuration, there is no crash but the applications have no sound.

If I use the symlink trick to hide `libasound.so.2`, I get no crash and the applications have working sound. I get the same experience if the library is not installed at all.

---

To make it clear, I am OK to settle with the hiding of `libasound.so.2` from the games as a workaround, as it seems effective. But I am very frustrated by not understanding how this actually works ;)

DarkWav, 2020-03-20:
Hmm.... On Arch with alsa-lib build directly from latest git with ABS i can run Vaporum (a unity game) without problems, with pulseaudio not running at all. Needless to say my aplay -L output looks very different than yours.
Could you post the output of aplay -L from the version compiled from latest git master without reverting 5ee5ef3?

vv221, 2020-03-20:
Full output of `aplay -L`, using a build of current master (6137d263f3805c72781ea74b813b0e3754f5fc34):
```
surround21
    2.1 Surround output to Front and Subwoofer speakers
surround40
    4.0 Surround output to Front and Rear speakers
surround41
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50
    5.0 Surround output to Front, Center and Rear speakers
surround51
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
null
    Discard all samples (playback) or generate zero samples (capture)
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
default:CARD=SB
    HDA ATI SB, VT1708S Analog
    Default Audio Device
sysdefault:CARD=SB
    HDA ATI SB, VT1708S Analog
    Default Audio Device
front:CARD=SB,DEV=0
    HDA ATI SB, VT1708S Analog
    Front speakers
surround21:CARD=SB,DEV=0
    HDA ATI SB, VT1708S Analog
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=SB,DEV=0
    HDA ATI SB, VT1708S Analog
    4.0 Surround output to Front and Rear speakers
surround41:CARD=SB,DEV=0
    HDA ATI SB, VT1708S Analog
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=SB,DEV=0
    HDA ATI SB, VT1708S Analog
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=SB,DEV=0
    HDA ATI SB, VT1708S Analog
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=SB,DEV=0
    HDA ATI SB, VT1708S Analog
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
hdmi:CARD=SB,DEV=0
    HDA ATI SB, VT1708S Digital
    HDMI Audio Output
dmix:CARD=SB,DEV=0
    HDA ATI SB, VT1708S Analog
    Direct sample mixing device
dmix:CARD=SB,DEV=3
    HDA ATI SB, VT1708S Digital
    Direct sample mixing device
dsnoop:CARD=SB,DEV=0
    HDA ATI SB, VT1708S Analog
    Direct sample snooping device
dsnoop:CARD=SB,DEV=3
    HDA ATI SB, VT1708S Digital
    Direct sample snooping device
hw:CARD=SB,DEV=0
    HDA ATI SB, VT1708S Analog
    Direct hardware device without any conversions
hw:CARD=SB,DEV=3
    HDA ATI SB, VT1708S Digital
    Direct hardware device without any conversions
plughw:CARD=SB,DEV=0
    HDA ATI SB, VT1708S Analog
    Hardware device with all software conversions
plughw:CARD=SB,DEV=3
    HDA ATI SB, VT1708S Digital
    Hardware device with all software conversions
usbstream:CARD=SB
    HDA ATI SB
    USB Stream Output
hdmi:CARD=HDMI,DEV=0
    HDA ATI HDMI, HDMI 0
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=1
    HDA ATI HDMI, HDMI 1
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=2
    HDA ATI HDMI, HDMI 2
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 3
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=4
    HDA ATI HDMI, HDMI 4
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=5
    HDA ATI HDMI, HDMI 5
    HDMI Audio Output
dmix:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 0
    Direct sample mixing device
dmix:CARD=HDMI,DEV=7
    HDA ATI HDMI, HDMI 1
    Direct sample mixing device
dmix:CARD=HDMI,DEV=8
    HDA ATI HDMI, HDMI 2
    Direct sample mixing device
dmix:CARD=HDMI,DEV=9
    HDA ATI HDMI, HDMI 3
    Direct sample mixing device
dmix:CARD=HDMI,DEV=10
    HDA ATI HDMI, HDMI 4
    Direct sample mixing device
dmix:CARD=HDMI,DEV=11
    HDA ATI HDMI, HDMI 5
    Direct sample mixing device
dsnoop:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 0
    Direct sample snooping device
dsnoop:CARD=HDMI,DEV=7
    HDA ATI HDMI, HDMI 1
    Direct sample snooping device
dsnoop:CARD=HDMI,DEV=8
    HDA ATI HDMI, HDMI 2
    Direct sample snooping device
dsnoop:CARD=HDMI,DEV=9
    HDA ATI HDMI, HDMI 3
    Direct sample snooping device
dsnoop:CARD=HDMI,DEV=10
    HDA ATI HDMI, HDMI 4
    Direct sample snooping device
dsnoop:CARD=HDMI,DEV=11
    HDA ATI HDMI, HDMI 5
    Direct sample snooping device
hw:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 0
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=7
    HDA ATI HDMI, HDMI 1
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=8
    HDA ATI HDMI, HDMI 2
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=9
    HDA ATI HDMI, HDMI 3
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=10
    HDA ATI HDMI, HDMI 4
    Direct hardware device without any conversions
hw:CARD=HDMI,DEV=11
    HDA ATI HDMI, HDMI 5
    Direct hardware device without any conversions
plughw:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 0
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=7
    HDA ATI HDMI, HDMI 1
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=8
    HDA ATI HDMI, HDMI 2
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=9
    HDA ATI HDMI, HDMI 3
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=10
    HDA ATI HDMI, HDMI 4
    Hardware device with all software conversions
plughw:CARD=HDMI,DEV=11
    HDA ATI HDMI, HDMI 5
    Hardware device with all software conversions
usbstream:CARD=HDMI
    HDA ATI HDMI
    USB Stream Output
```

DarkWav, 2020-03-20:
Well.......  The issue might very well be that you forgot re-building or re-installing libasound2-data.
The problem is that the patch is shipped in form of a modification of the following files:
```
/usr/share/alsa/pcm/surround21.conf
/usr/share/alsa/pcm/surround40.conf
/usr/share/alsa/pcm/surround41.conf
/usr/share/alsa/pcm/surround50.conf
/usr/share/alsa/pcm/surround51.conf
/usr/share/alsa/pcm/surround71.conf
```
Debian's alsa-lib package is a source package that gets divided into multiple sub-packages at build time.
The main sub-package is libasound2.
However, in order to get the patch working, **you also need to build AND install libasound2-data from the latest git master as it contains the actual patch**. Once debian rolls out the next version of alsa the problem will be fixed, but for now make sure you rebuild and re-install both libasound2 and libasound2-data from latest git master.

vv221, 2020-03-20:
During my tests I’m actually not installing the new libraries, but preloading them using `$LD_LIBRARY_PATH`. So I guess I might not be loading changes related to configuration.
I see a directory `src/conf` in my local git clone, I guess I can load it using `$ALSA_CONFIG_DIR`…

---

That was it, well spotted @DarkWav ;)

Sorry for the noise, the fix in current master is indeed working as expected.
Well, at least I learnt a bit more about using custom builds of alsa-lib… I hope at some point it will help me in making useful reports for real issues.

DarkWav, 2020-03-20:
Nice! Then this issue is truly fixed now xD