alsa-project/alsa-lib #39: [hints] arecord -L show entries which are not capture pcms

sylware, 2020-03-26:


sylware, 2020-03-27:
On Fri, Mar 27, 2020 at 04:01:57AM -0700, Jaroslav Kysela wrote:
> Closed #39 via d434638a683c01bd6decf8f41863044055237a50.

Tested with the mic from my usb webcam, and it actually needs a little bit more
tidying.

--- a/share/alsa/pcm/front.conf	2020-03-27
+++ b/share/alsa/pcm/front.conf	2020-03-27
@@ -53,6 +53,7 @@
 			name defaults.namehint.basic
 		}
 		description "Front speakers"
-		device $DEV
+		device_output $DEV
+		omit_noargs true
 	}
 }

--- a/share/alsa/pcm/iec958.conf
+++ b/share/alsa/pcm/iec958.conf
@@ -78,6 +78,6 @@
 			name defaults.namehint.basic
 		}
 		description "IEC958 (S/PDIF) Digital Audio Output"
-		device $DEV
+		device_output $DEV
 	}
 }

tiwai, 2020-05-06:
This seems breaking quite a few applications.  They use "front" for capturing the primary stream, too, so handling front as output-only makes the stream unavailable.

Also, I don't think iec958 is output only.  There can be iec958 capture stream, too.

perexg, 2020-05-06:
Then the usage does not follow the description (speakers, S/PDIF output).

tiwai, 2020-05-06:
The fact is that this commit broke things, so it has to be fixed in some way.
IMO, we need to rather update the description instead of hiding it.  iec958 might be let as output-only, as the SPDIF input is a rare occasion, though.  But front capture is a must for now.

I guess the front is used for capture by many apps just because there was no primary capture stream definition.

perexg, 2020-05-06:
> I guess the front is used for capture by many apps just because there was no primary capture stream definition.

The primary device should be 'default'. What apps are affected?

tiwai, 2020-05-06:
The report was about Tracktion Waveform, but I know this being used by some others, too.

"default" would be redirected to dsnoop or PA, so some apps were (mis)advised to use "front" as the primary hardware capture stream in the past.

perexg, 2020-05-07:
Note that this commit does not change the behaviour of the capture 'front' devices but only the device hint list. The Trancktion Waveform seems like a binary only application, thus it's a bit difficult to see what they cooked.

If we accept the front device as the default input device, we should probably redefine it using asym plugin (and add support for the asym plugin configuration to src/control/namehint.c). But ... we can do this ... later.

closed 2020-05-07 by perexg