alsa-project/alsa-lib #41: snd_pcm_drain() infinite loop

sylware, 2020-04-11:


sylware, 2020-04-12:
ALSA:HW_PARAMS START------------------------------------------------------------
ACCESS:  RW_NONINTERLEAVED
FORMAT:  FLOAT_LE
SUBFORMAT:  STD
SAMPLE_BITS: [0 0]
FRAME_BITS: [64 64]
CHANNELS: [2 2]
RATE: [44100 44100]
PERIOD_TIME: [21333 21333]
PERIOD_SIZE: [940 940]
PERIOD_BYTES: [0 0]
PERIODS: [0 0]
BUFFER_TIME: [0 0]
BUFFER_SIZE: [7526 7526]
BUFFER_BYTES: [60208 60208]
TICK_TIME: [0 0]
ALSA:HW_PARAMS END--------------------------------------------------------------
ALSA:SW_PARAMS START------------------------------------------------------------
tstamp_mode: NONE
tstamp_type: MONOTONIC_RAW
period_step: 1
avail_min: 940
start_threshold: 1
stop_threshold: 7526
silence_threshold: 0
silence_size: 0
boundary: 4236761349448794112
ALSA:SW_PARAMS END--------------------------------------------------------------
ALSA PCM DUMP START-------------------------------------------------------------
Plug PCM: Linear Integer <-> Linear Float conversion PCM (S32_LE)
Its setup is:
  stream       : PLAYBACK
  access       : RW_NONINTERLEAVED
  format       : FLOAT_LE
  subformat    : STD
  channels     : 2
  rate         : 44100
  exact rate   : 44100 (44100/1)
  msbits       : 32
  buffer_size  : 7526
  period_size  : 940
  period_time  : 21333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC_RAW
  period_step  : 1
  avail_min    : 940
  period_event : 1
  start_threshold  : 1
  stop_threshold   : 7526
  silence_threshold: 0
  silence_size : 0
  boundary     : 4236761349448794112
Slave: Rate conversion PCM (48000, sformat=S32_LE)
Converter: linear-interpolation
Protocol version: 10002
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 2
  rate         : 44100
  exact rate   : 44100 (44100/1)
  msbits       : 32
  buffer_size  : 7526
  period_size  : 940
  period_time  : 21333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC_RAW
  period_step  : 1
  avail_min    : 940
  period_event : 1
  start_threshold  : 1
  stop_threshold   : 7526
  silence_threshold: 0
  silence_size : 0
  boundary     : 4236761349448794112
Slave: Soft volume PCM
Control: PCM Playback Volume
min_dB: -51
max_dB: 0
resolution: 256
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 32
  buffer_size  : 8192
  period_size  : 1024
  period_time  : 21333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC_RAW
  period_step  : 1
  avail_min    : 1024
  period_event : 1
  start_threshold  : 1
  stop_threshold   : 8192
  silence_threshold: 0
  silence_size : 0
  boundary     : 4611686018427387904
Slave: Direct Stream Mixing PCM
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 32
  buffer_size  : 8192
  period_size  : 1024
  period_time  : 21333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC_RAW
  period_step  : 1
  avail_min    : 1024
  period_event : 1
  start_threshold  : 1
  stop_threshold   : 8192
  silence_threshold: 0
  silence_size : 0
  boundary     : 4611686018427387904
Hardware PCM card 0 'HDA ATI SB' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S32_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits

sylware, 2020-04-12:
in the function pcm_rate.c:snd_pcm_rate_drain, the while loop which is supposed to write the remaining audio data to the slave using snd_pcm_rate_commit_area function, does not seem to update rate->appl_ptr and rate->rate->last_commit_ptr, which are then "stuck", hence the never ending writting of data to the slave.

sylware, 2020-04-12:
I wrote a quick and dirty patch which updates last_commit_ptr: it seems it fixes the issues.
That said I have no idea how to update properly this pointer in regards of boundary (and maybe other critical values).

perexg, 2020-04-17:
The drain is not supposed to update last_commit_ptr. Also, unless the drain is not called multiple times, the while loop takes last_commit_ptr only once. Do you call drain in the non-block mode?

sylware, 2020-04-19:
Sorry, I did not insist enough: yes, in non-block mode.

I was looking at how to update consistently the rate plugin state using some
already coded functions but as far I could understand the code, I found none.
Then I tried to modify the plugin state with the least amount of consistent
modifications, but I could not be so sure.

sylware, 2020-04-23:
did you see my patch on the mailing list?

perexg, 2020-04-23:
https://mailman.alsa-project.org/pipermail/alsa-devel/2020-April/165848.html

The patch looks basically good (we can fix it later if we encounter any issues). Create PR or resend with proper description and signed-off-by line. Thanks.

sylware, 2020-04-24:
This is an on-the-spot patch, only tested with the software showing the issue.
If you don't see any obvious issues with it, I'll engage in some broader
testing on top of "now-git" alsa-lib.
If I am not able to break it after some time, I'll post it properly formated on
the mailing list.

sylware, 2020-04-30:
Could not break it in my usage. I posted the patch on the mailing-list.
I hope I kept the state consistent enough for the draining to reach the end in intended use cases.

perexg, 2020-04-30:
Applied in 29041c522071172af2c6bf086b3a0c19510a6660 . Thank you.

closed 2020-04-30 by perexg