alsa-project/alsa-lib #47: helem (MIXER,'Master Playback Volume',0,1,0) appears twice or more cannot load mixer controls: Invalid argument

igrekster, 2020-05-03:
I have a PCIE Asus Sound Card  (Asus Xonar AE) that uses USB interface:
```
Bus 003 Device 002: ID 0b05:180f ASUSTek Computer, Inc.
```
It's been working fine on earlier versions of  kernel/alsa-lib but at some point the volume control stopped working (the playback still works fine):

```
$ alsamixer -c 1
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.2-r1/work/alsa-lib-1.2.2/src/mixer/simple_none.c:1544:(simple_add1) helem (MIXER,'Master Playback Volume',0,1,0) appears twice or more
cannot load mixer controls: Invalid argument
```
This is on kernel 5.4.37/alsa 1.2.2. Unfortunately I cannot downgrade to the point when it was working to pinpoint the issue. I tried kernels 5.3.11 and alsa 1.1.8.
I've tried building the latest alsa-lib, and it is still showing the error.
I'm happy to try patches to alsa or kernel modules to help debug this.

A few more details:

[alsa-info.txt](https://github.com/alsa-project/alsa-lib/files/4569764/alsa-info.txt)
[proc-asound.txt](https://github.com/alsa-project/alsa-lib/files/4569766/proc-asound.txt)

perexg, 2020-05-03:
It appears like a driver issue:

````
numid=20,iface=MIXER,name='Master Playback Volume'
numid=19,iface=MIXER,name='Master Playback Volume',device=1
````

The index field should be one not the device field. @tiwai ?

igrekster, 2020-05-03:
I believe the second volume control is for the SPDIF optical output, which is a device in alsa lingo.

tiwai, 2020-05-03:
> It appears like a driver issue:
> 
> ```
> numid=20,iface=MIXER,name='Master Playback Volume'
> numid=19,iface=MIXER,name='Master Playback Volume',device=1
> ```
> 
> The index field should be one not the device field. @tiwai ?

Those are user controls, so it's likely no kernel problem, but some applications created the controls wrongly.

Try to unload the all sound modules once, then remove your saved alsactl file (e.g. /var/lib/alsa/asound.state) and restart.

igrekster, 2020-05-03:
@twai, I didn't think `asound.state` would impact it, but  it did. After removing the file I got it working, thank you.

closed 2020-05-03 by igrekster