alsa-project/alsa-lib #49: Alsa doesn't respect XDG Base Directory standard?

luxintra, 2020-05-16:
My .config is set to $XDG_CONFIG_HOME and moving asoundrc to $XDG_CONFIG_HOME/alsa ...does not work for me.

perexg, 2020-05-16:
alsa-lib is not a desktop library, it's system library.  You may use ALSA_CONFIG_DIR to override the /usr/share/alsa directory.

tiwai, 2020-05-16:
Maybe we can add $XDG_CONFIG_HOME/asoundrc to the default search path in /usr/share/alsa/alsa.conf?

luxintra, 2020-05-16:
Thanks! That did the trick. And yea it should be default. I changed the search path in /usr/share/alsa/alsa.conf to $XDG_CONFIG_HOME/alsa/asoundrc

perexg, 2020-05-16:
Hmm, it seems that even git added the XDG_CONFIG_HOME support, so it's on my radar now.

We should load also ~/.config/alsa/asoundrc if XDG_CONFIG_HOME is not defined.

perexg, 2020-06-04:
Commited to master.

closed 2020-06-04 by perexg