alsa-project/alsa-lib #55: Missing hotplug support

wm4, 2020-05-31:
Hotplug support is mandatory in any modern system. But in ALSA there is no way to be notified by audio device changes.

I'm aware that as a hack, one could listen to udev notifications. But this is not appropriate, because:
- udev is a different API, which makes this is inelegant and messy
- it requires a dependency on udev, which may not be a workable solution even if the host OS is Linux
- there may be devices that do not have anything to do with udev (for example network audio interfaces implemented in userspace)
- changes in udev may not be reflected in ALSA at the same time (creating the need to avoid race conditions with timeouts or whatever)
- it's ALSA's responsibility to provide this; I don't know any OS where you have to go to a completely different API for this

perexg, 2020-06-04:
It's not as easy. The hardware based device changes can be really tracked using the /dev/snd/control* device node events, but the other "virtual" devices are mostly based on the configuration files and evaluated when the application asks for them.

We have already alsactl state management daemon, so we can probably send some notifications over DBus at least for the hardware devices.

wm4, 2020-06-04:
A DBus requirement  wouldn't be very welcome, to be honest. It's hard to use and uses another socket connection, which needs to be managed.

perexg, 2020-06-04:
Any idea / code / patch is welcome.

JoseTomasTocino, 2021-03-18:
What's a good workaround for this? I've been doing what the OP said for some time: checking for udev events and then waiting 6 seconds for the device to actually be available to ALSA... but it's such a hack.