alsa-project/alsa-lib #56: 1.2.3: alsa-l;ibs is not LTO ready

kloczek, 2020-06-08:
When build with LTO test suite is failing
```
[tkloczko@barrel alsa-lib-1.2.3]$ make check
Making check in doc
make[1]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/doc'
Making check in pictures
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/doc/pictures'
make[2]: Nothing to be done for 'check'.
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/doc/pictures'
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/doc'
make[2]: Nothing to be done for 'check-am'.
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/doc'
make[1]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/doc'
Making check in include
make[1]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/include'
Making check in sound
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/include/sound'
Making check in uapi
make[3]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/include/sound/uapi'
make[3]: Nothing to be done for 'check'.
make[3]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/include/sound/uapi'
make[3]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/include/sound'
make[3]: Nothing to be done for 'check-am'.
make[3]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/include/sound'
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/include/sound'
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/include'
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/include'
make[1]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/include'
Making check in src
make[1]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src'
Making check in control
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/control'
make[2]: Nothing to be done for 'check'.
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/control'
Making check in mixer
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/mixer'
make[2]: Nothing to be done for 'check'.
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/mixer'
Making check in pcm
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/pcm'
make[3]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/pcm'
make[3]: Nothing to be done for 'check-am'.
make[3]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/pcm'
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/pcm'
Making check in timer
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/timer'
make[2]: Nothing to be done for 'check'.
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/timer'
Making check in rawmidi
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/rawmidi'
make[2]: Nothing to be done for 'check'.
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/rawmidi'
Making check in hwdep
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/hwdep'
make[2]: Nothing to be done for 'check'.
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/hwdep'
Making check in seq
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/seq'
make[2]: Nothing to be done for 'check'.
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/seq'
Making check in ucm
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/ucm'
make[2]: Nothing to be done for 'check'.
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/ucm'
Making check in conf
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/conf'
Making check in cards
make[3]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/conf/cards'
make[3]: Nothing to be done for 'check'.
make[3]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/src/conf/cards'
[... 38 lines cut ...]
Making check in .
make[2]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/test'
make  control pcm pcm_min latency seq playmidi1 timer rawmidi midiloop oldapi queue_timer namehint client_event_filter chmap audio_time user-ctl-element-set pcm-multi-thread
make[3]: Entering directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/test'
make[3]: 'control' is up to date.
make[3]: 'pcm' is up to date.
make[3]: 'pcm_min' is up to date.
make[3]: 'latency' is up to date.
make[3]: 'seq' is up to date.
make[3]: 'playmidi1' is up to date.
make[3]: 'timer' is up to date.
make[3]: 'rawmidi' is up to date.
make[3]: 'midiloop' is up to date.
/bin/sh ../libtool  --tag=CC   --mode=link gcc -Wall -pipe -g -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto=auto -flto-partition=none  -Wl,-z,relro -Wl,--as-needed  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -flto=auto -flto-partition=none -fuse-linker-plugin -o oldapi oldapi.o ../src/libasound.la
libtool: link: gcc -Wall -pipe -g -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto=auto -flto-partition=none -Wl,-z -Wl,relro -Wl,--as-needed -Wl,-z -Wl,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -flto=auto -flto-partition=none -fuse-linker-plugin -o .libs/oldapi oldapi.o  ../src/.libs/libasound.so -lm -ldl -lpthread -lrt
/usr/bin/ld: oldapi.lto.o: in function `main':
/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/test/oldapi.c:39: undefined reference to `snd_pcm_hw_params_get_access@ALSA_0.9'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:529: oldapi] Error 1
make[3]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/test'
make[2]: *** [Makefile:796: check-am] Error 2
make[2]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/test'
make[1]: *** [Makefile:645: check-recursive] Error 1
make[1]: Leaving directory '/home/tkloczko/rpmbuild/BUILD/alsa-lib-1.2.3/test'
```

perexg, 2020-07-02:
gcc-10 symver references:

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=48200
https://sourceware.org/pipermail/elfutils-devel/attachments/20200414/1c0c2903/attachment.bin

perexg, 2020-10-13:
Fixed in 2a204a5412ee956ede792a6872ff3d0516d0f0df .

closed 2020-10-13 by perexg

kloczek, 2020-10-13:
Yep it works.

Thx :)