alsa-project/alsa-lib #57: We were woken up with POLLOUT set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail. Jun 09 14:31:33 elrond pulseaudio[11933]: Most likely this is a bug in the ALSA driver 'snd_usb_audio'. Please report this issue to the ALSA developers. Jun 09 14:31:33 elrond pulseaudio[11933]: ALSA woke us up to write new data to the device, but there was actually nothing to write.

danieloppenlander, 2020-06-09:
I ran across an error in my journalctl on Ubuntu 20.04. It said to report a bug. Here is my alsa-info.sh output: http://alsa-project.org/db/?f=6b505aa5138f2a974037ab5c3a89e1605807df97

dpavlushko, 2020-06-14:
I have the same message with new SSL2 USB interface:
```log
Jun 14 02:58:14 [pulseaudio] [alsa-sink-USB Audio] alsa-sink.c: ALSA woke us up to write new data to the device, but there was actually nothing to write.
Jun 14 02:58:14 [pulseaudio] [alsa-sink-USB Audio] alsa-sink.c: Most likely this is a bug in the ALSA driver 'snd_usb_audio'. Please report this issue to the ALSA developers.
Jun 14 02:58:14 [pulseaudio] [alsa-sink-USB Audio] alsa-sink.c: We were woken up with POLLOUT set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
```

http://alsa-project.org/db/?f=d3e9f14ca919442570599e72a2670ce8b46bdf51

Swaphelaer, 2020-07-15:
I've got the same in Linux Mint 20.

AndersonGI, 2020-08-07:
I'am having the same on xubuntu focal 20.04

PopeRigby, 2020-12-19:
I have the same issue on Arch Linux. 

Output of `alsa-info.sh`: https://alsa-project.org/db/?f=ae34ff08c9404efa7d0ad3db0aa1e505bfd99865

perexg, 2020-12-29:
Dup of #90.

closed 2020-12-29 by perexg