alsa-project/alsa-lib #63: File Plugin seems to drop data when writing to FIFO

riggiding, 2020-06-18:
Hi,

we're using the file plugin to pipe data for use in a vu-meter application. This worked well but since updating to 1.1.4.1 the data received is incomplete (tested using sine test tones and Audacity to playback / inspect the data that went through the fifo).
**When writing to a normal file all is fine.**

Also tried using tee, same result.
Any hints or ideas would be great!

**The fifo is created by:**
#! /bin/sh

if [ ! -p "$ALSA_FIFO" ]; then
	# create a fifo for alsa -> vumeter
	mkfifo "$ALSA_FIFO"                                           
	chmod 666 "$ALSA_FIFO"
fi

cat <>"$ALSA_FIFO"

**asound.conf:**

pcm.hiface {
	type hw
	card 0
	device 0
}

pcm.vumeter {
	type rate
	slave {
		pcm "alsaFifo"
		rate 44100
		format "S16_LE"
	}
}

pcm.alsaFifo {
	type file
	slave.pcm "null"
	file "/tmp/alsa-fifo"
}

pcm.alsaFifoTee {
       type empty
       slave.pcm "tee:null,'/tmp/alsa-fifo',raw"
}

pcm.direct {
	type route
	slave.pcm "split_direct"
	ttable.0.0 1
	ttable.1.1 1
	ttable.0.2 1
	ttable.1.3 1
}

pcm.split_direct {
	type multi
	slaves.a.pcm "hiface"
	slaves.a.channels 2
	slaves.b.pcm "vumeter"
	slaves.b.channels 2
	bindings.0.slave a
	bindings.0.channel 0
	bindings.1.slave a
	bindings.1.channel 1
	bindings.2.slave b
	bindings.2.channel 0
	bindings.3.slave b
	bindings.3.channel 1
}

perexg, 2020-06-18:
https://github.com/alsa-project/alsa-lib/commit/6ecd330fb8a4063f814e4828f017ade20c6003e6

riggiding, 2020-06-18:
Thx @perexg! Unfortunately the patch did not help :-/
Normal playback was never affected so I think there has to be a different cause

perexg, 2020-06-18:
Sorry, it's probably the patch which introduced this issue (the write error code for the output file descriptor is handled differently than in the previous version). I do not have a proper fix yet.

JanM-cz, 2020-09-16:
Hi @perexg any news on a fix or a possible workaround? I think we are influenced by the same problem.

adjiandov, 2020-09-16:
Hi @perexg, hi all. In case someone finds a solution please share since I am affected by this problem as well.

anton87ken, 2020-09-25:
Hello @perexg, we're experiencing the same problem. Any news on a this?
Cheers!

mrakks, 2020-09-25:
Hi, we encountered the same issue. It would be much appreciated if someone could find a solution. Best regards! @perexg

paulbalomiri, 2020-10-15:
Hello @perexg,

> Sorry, it's probably the patch which introduced this issue (the write error code for the output file descriptor is handled differently than in the previous version). I do not have a proper fix yet.

Do you still believe that the mentioned patch is causing this or is that idea dismissed?

riggiding, 2020-10-15:
@paulbalomiri,

> Do you still believe that the mentioned patch is causing this or is that idea dismissed?

I don't think so - the problem was evident before I applied the path (which I did believe to be a fix)

mbuchner, 2022-04-05:
Any update on this one?