alsa-project/alsa-lib #64: Regression in Dell WD19TB headphone out volume between 1.2.2 and 1.2.3

ruuda, 2020-06-25:
After updating alsa-lib from 1.2.2 to 1.2.3, alsa-ucm-conf from 1.2.2 to 1.2.3, and alsa-topology-conf from 1.1.1 to 1.2.3, the maximum volume of the headphone out on my Dell WD19TB dock became much lower, to the point where it can’t be set loud enough for normal listening.

The output devices reported in Gnome Settings are also different.

1.2.2:

[image]

1.2.3:

[image]

The dock does indeed have a headphone out on the front, and a line out on the back, so the line out being detected now is good. The volume range of the line out on 1.2.3 is similar to that of the headphone out on 1.2.2, but the volume range of the headphone out on 1.2.3 is much lower.

The card itself as reported by `pactl list-cards` on 1.2.2:

```
    index: 0
	name: <alsa_card.usb-Generic_USB_Audio_200901010001-00>
	driver: <module-alsa-card.c>
	owner module: 6
	properties:
		alsa.card = "1"
		alsa.card_name = "WD19 Dock"
		alsa.long_card_name = "Dell-WD15-Dock"
		alsa.driver_name = "snd_usb_audio"
		device.bus_path = "pci-0000:0b:00.0-usb-0:2.3.4:1.0"
		sysfs.path = "/devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:01.0/0000:09:00.0/0000:0a:02.0/0000:0b:00.0/usb3/3-2/3-2.3/3-2.3.4/3-2.3.4:1.0/sound/card1"
		udev.id = "usb-Generic_USB_Audio_200901010001-00"
		device.bus = "usb"
		device.vendor.id = "0bda"
		device.vendor.name = "Realtek Semiconductor Corp."
		device.product.id = "402e"
		device.product.name = "USB Audio"
		device.serial = "Generic_USB_Audio_200901010001"
		device.string = "1"
		device.description = "USB Audio"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card-usb"
	profiles:
		input:analog-stereo: Analog Stereo Input (priority 65, available: unknown)
		output:analog-stereo: Analog Stereo Output (priority 6500, available: unknown)
		output:analog-stereo+input:analog-stereo: Analog Stereo Duplex (priority 6565, available: unknown)
		off: Off (priority 0, available: unknown)
	active profile: <output:analog-stereo+input:analog-stereo>
	sinks:
		alsa_output.usb-Generic_USB_Audio_200901010001-00.analog-stereo/#0: USB Audio Analog Stereo
	sources:
		alsa_output.usb-Generic_USB_Audio_200901010001-00.analog-stereo.monitor/#0: Monitor of USB Audio Analog Stereo
		alsa_input.usb-Generic_USB_Audio_200901010001-00.analog-stereo/#1: USB Audio Analog Stereo
	ports:
		analog-input-mic: Microphone (priority 8700, latency offset 0 usec, available: unknown)
			properties:
				device.icon_name = "audio-input-microphone"
		analog-output-headphones: Headphones (priority 9900, latency offset 0 usec, available: unknown)
			properties:
				device.icon_name = "audio-headphones"
```

And as reported on 1.2.3:

```
    index: 0
	name: <alsa_card.usb-Generic_USB_Audio_200901010001-00>
	driver: <module-alsa-card.c>
	owner module: 6
	properties:
		alsa.card = "1"
		alsa.card_name = "WD19 Dock"
		alsa.long_card_name = "Dell-WD15-Dock"
		alsa.driver_name = "snd_usb_audio"
		device.bus_path = "pci-0000:0b:00.0-usb-0:2.3.4:1.0"
		sysfs.path = "/devices/pci0000:00/0000:00:1d.6/0000:06:00.0/0000:07:01.0/0000:09:00.0/0000:0a:02.0/0000:0b:00.0/usb3/3-2/3-2.3/3-2.3.4/3-2.3.4:1.0/sound/card1"
		udev.id = "usb-Generic_USB_Audio_200901010001-00"
		device.bus = "usb"
		device.vendor.id = "0bda"
		device.vendor.name = "Realtek Semiconductor Corp."
		device.product.id = "402e"
		device.product.name = "USB Audio"
		device.serial = "Generic_USB_Audio_200901010001"
		device.string = "1"
		device.description = "USB Audio"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card-usb"
	profiles:
		HiFi: Default (priority 8000, available: unknown)
		off: Off (priority 0, available: unknown)
	active profile: <HiFi>
	sinks:
		alsa_output.usb-Generic_USB_Audio_200901010001-00.HiFi__hw_Dock_1__sink/#0: USB Audio Line Out
		alsa_output.usb-Generic_USB_Audio_200901010001-00.HiFi__hw_Dock__sink/#1: USB Audio Headphones
	sources:
		alsa_output.usb-Generic_USB_Audio_200901010001-00.HiFi__hw_Dock_1__sink.monitor/#0: Monitor of USB Audio Line Out
		alsa_output.usb-Generic_USB_Audio_200901010001-00.HiFi__hw_Dock__sink.monitor/#1: Monitor of USB Audio Headphones
		alsa_input.usb-Generic_USB_Audio_200901010001-00.HiFi__hw_Dock__source/#2: USB Audio Microphone
	ports:
		[Out] Line: Line Out (priority 200, latency offset 0 usec, available: unknown)
			properties:
				
		[Out] Headphones: Headphones (priority 100, latency offset 0 usec, available: unknown)
			properties:
				
		[In] Mic: Microphone (priority 100, latency offset 0 usec, available: unknown)
			properties:
```

perexg, 2020-06-25:
Could you attach a link to output from 'alsa-info.sh --no-upload' ?

ruuda, 2020-06-25:
* [Output for 1.2.2](https://gist.githubusercontent.com/ruuda/b5bb8ec3f8cf30c841f45dea465d1e87/raw/0aa409e7e7ba78eb27a932e6e63b8e070779d647/alsa-info.txt)
* [Output for 1.2.3](https://gist.githubusercontent.com/ruuda/d90190f02f5f2ce876d23074847ae2f0/raw/eb5c119d4e291f76c2c6c691e192302fc22c28c5/alsa-info.txt)

The diff, with timestamp differences removed:

```diff
@@ -82,7 +82,7 @@ SMP Enabled:       Yes
 !!------------
 
 Driver version:     k5.6.15-arch1-1
-Library version:    1.2.2
+Library version:    1.2.3
 Utilities version:  1.2.3
 
 @@ -670,7 +670,7 @@ card 1: Dock [WD19 Dock], device 0: USB Audio [USB Audio]
   Subdevices: 1/1
   Subdevice #0: subdevice #0
 card 1: Dock [WD19 Dock], device 1: USB Audio [USB Audio #1]
-  Subdevices: 1/1
+  Subdevices: 0/1
   Subdevice #0: subdevice #0
 
 ARECORD
@@ -1629,8 +1629,8 @@ state.Dock {
                iface PCM
                device 1
                name 'Playback Channel Map'
-               value.0 0
-               value.1 0
+               value.0 3
+               value.1 4
                comment {
                        access read
                        type INTEGER
```

ruuda, 2022-03-06:
This resolved itself at some point, I don’t remember what fixed it in the end.

closed 2022-03-06 by ruuda