alsa-project/alsa-lib #65: Cannot use HyperX Cloud Flight S Microphone

DirtyCajunRice, 2020-06-25:
I have a New USB headset that was added/fixed in kernel patch 5.7rc3 (https://lkml.org/lkml/2020/4/4/197). It now loads all the output options but no input options. Is there something i can do to force it?
also have this if it helps: http://alsa-project.org/db/?f=7cf962fe3df280a9a81f5cd1e82633d8cba7b02e

emmanuel099, 2020-06-27:
Please note that my patch was about Cloud **Alpha** S and not Cloud Flight S.

See https://github.com/torvalds/linux/commit/73094608b8e214952444fb104651704c98a37aeb#diff-aba21fac395f457b2f223b8f36e4bde1 for the Cloud Flight S fix.

DirtyCajunRice, 2020-07-23:
For those that come here after me: Even with the fix you to enable the quirk. I explained [here](https://askubuntu.com/a/1253852/1057566)

closed 2020-07-27 by perexg