alsa-project/alsa-lib #68: snd_pcm_avail() returned a value that is exceptionally large

reezer, 2020-07-18:
I hope this is the right place to report this. This occurred in relation to a crash.
http://alsa-project.org/db/?f=52e8505d884682fc06adac4ab8ce92a4ca3e0870

```
E: [alsa-sink-ALC887-VD Analog] alsa-util.c: snd_pcm_avail() returned a value that is exceptionally large: 177312 bytes (923 ms).
E: [alsa-sink-ALC887-VD Analog] alsa-util.c: Most likely this is a bug in the ALSA driver 'snd_hda_intel'. Please report this issue to the ALSA developers.
E: [alsa-sink-ALC887-VD Analog] alsa-util.c: snd_pcm_dump():
E: [alsa-sink-ALC887-VD Analog] alsa-util.c: Soft volume PCM
E: [alsa-sink-ALC887-VD Analog] alsa-util.c: Control: PCM Playback Volume
E: [alsa-sink-ALC887-VD Analog] alsa-util.c: min_dB: -51
E: [alsa-sink-ALC887-VD Analog] alsa-util.c: max_dB: 0
E: [alsa-sink-ALC887-VD Analog] alsa-util.c: resolution: 256
E: [alsa-sink-ALC887-VD Analog] alsa-util.c: Its setup is:
E: [alsa-sink-ALC887-VD Analog] alsa-util.c:   stream       : PLAYBACK
E: [alsa-sink-ALC887-VD Analog] alsa-util.c:   access       : MMAP_INTERLEAVED
E: [alsa-sink-ALC887-VD Analog] alsa-util.c:   format       : S16_LE
E: [alsa-sink-ALC887-VD Analog] alsa-util.c:   subformat    : STD
E: [alsa-sink-ALC887-VD Analog] alsa-util.c:   channels     : 2
E: [alsa-sink-ALC887-VD Analog] alsa-util.c:   rate         : 48000
E: [alsa-sink-ALC887-VD Analog] alsa-util.c:   exact rate   : 48000 (48000/1)
E: [alsa-sink-ALC887-VD Analog] alsa-util.c:   msbits       : 16
E: [alsa-sink-ALC887-VD Analog] alsa-util.c:   buffer_size  : 4800
```

kyrawertho, 2020-12-06:
I have the same issue and logs say it should be reported (but it doesn't say where). It happened several times before; my whole system is unusable for some time when this happens.

RogueScholar, 2020-12-11:
This started occurring to me as well, on a system (Kubuntu 20.10 on Lenovo ThinkPad T420s) where my configuration of both ALSA and Pulseaudio has been rock solid for years. Does anyone have an idea of how to approach debugging it?

perexg, 2020-12-14:
The PA check is not 100%. The real time measurement is missing, so it may be possible, that there's a task scheduling glitch (gap).

This suggest also the log in the second comment which clearly identifies that the system is too much busy:

  `Web Content invoked oom-killer: gfp_mask=0x100cca(GFP_HIGHUSER_MOVABLE), order=0, oom_score_adj=0` 

PA disables the stream stop (see stop_threshold) settings in the kernel, thus this situation may occur. Discuss this with PA developers.

tanuk, 2020-12-17:
Maybe we shouldn't tell users to report these problems to the ALSA developers if they are not actionable, or have those reports ever resulted in any kernel driver fixes?

perexg, 2020-12-17:
The problem with those redirections is that there are many false-positive reports (they are here, in kernel bugzilla and ocassionally on the alsa-devel mailing list). If PA does not measure the real time for the ALSA calls, the behaviour is correct when the system is overloaded (no real time processing).

I suggest to print the error (or warning) and add the real time measurement (to exclude /or report/ the scheduler problems) when PA runs in the debug mode without the redirection. The drivers are more stable than they were a decade ago.

tanuk, 2020-12-18:
I created this issue for PulseAudio: https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/1078

Jeronimo17, 2021-06-01:
In my case it seems that the problem was derived from thermald.service, for now I have no problem since disabling this service

lafrech, 2022-01-28:
My laptop just froze for the second time in a few days. This time I checked the syslog and found the same message, which brought me here.

Just to be sure I understand correctly, this is a WONTFIX because the issue reported by this message is caused by some latency issue but is not necessarily an issue in ALSA itself, right?

I don't find any other lead in the syslog, so I can't do much more.

Should I report this to my disro (Debian)? Is ther even a point doing so?

Jeronimo17, 2022-01-28:
I finally disabled tsched=0 and the kernel and Nvidia updates and removed Xfwm4 compositor options.

[image]

I did not get this error and the laptop keyboard did not start up again.

I don't really know how it all relates but with this configuration it has been working fine for months now