alsa-project/alsa-lib #76: No sound with Realtek ALC287

BLuFeNiX, 2020-07-27:
I get no sound via the speakers of my Lenovo Legion 7i laptop, which alsamixer tells me is using a Realtek ALC287.

Here is my alsa-info.sh output: http://alsa-project.org/db/?f=4272343a3590cc08f192f98113dedfc0418afe52

I made sure to disable Auto-Mute in alsamixer, and turn all volume levels to maximum. In all cases, I get no sounds from the speakers (running speaker-test, playing music, etc.). I *am* able to get sound via headphones and HDMI (though I believe HDMI is via a different sound card).

I previously reported this here: https://bugzilla.kernel.org/show_bug.cgi?id=208555
but I received no response in almost 3 weeks, so I think maybe I have reported it to the wrong place.

Please let me know if I can provide any assistance or additional information.

veraguido, 2020-07-31:
Same thing for me, I can confirm. At the moment using Manjaro with kernel 5.6.19

vtavernier, 2020-08-01:
With the same hardware config as @BLuFeNiX running Fedora 32 I encounter the same issue, kernel 5.7.10-201.fc32.x86_64

perexg, 2020-10-13:
It's the kernel issue. Keep the bug on kernel.org until someone has a time to look into. Note that we don't have the detailed information about hardware and it seems that the hardware vendors are using the amplifier chips for the integrated speakers on recent hardware which must be initialized, too. The linux support depends on the BIOS initialization (if any) at the moment.

closed 2020-10-13 by perexg

ianrh808, 2021-03-14:
does anyone know if this has been solved yet? Im on Ubuntu 20.10, and am experiencing the same issue.

lmontigny, 2021-03-17:
> does anyone know if this has been solved yet? Im on Ubuntu 20.10, and am experiencing the same issue.

Yes, updated the kernel to 5.11, worked for me.
Check https://itsfoss.com/upgrade-linux-kernel-ubuntu/ for the installation of the .deb package.

AndreiMiculita, 2021-03-27:
@lmontigny Just a quick question, `alsa-info` says ALC287, but for my model, the Lenovo PSREF says ALC3306 instead of ALC287. Do you know if it's the same for your model? Otherwise could you please check? [Link to PSREF](https://psref.lenovo.com/).

Centralobe, 2021-03-29:
@lmontigny I updated my kernel to 5.11 on a Legion 7i with the instructions you provided. However, I still have no sound. Did you do something else as well? 
I have a fresh Ubuntu 20.04.2, I switched back to the "old" kernel, which is 5.8.0-48.

lmontigny, 2021-03-29:
@AndreiMiculita I have ALC287, I don't know the difference with ALC3306.
@Centralobe , I am running Ubuntu **20.10** with kernel 5.11, you may try to use Ubuntu 21.04 it has 5.11 kernel by default.

AndreiMiculita, 2021-03-30:
> @AndreiMiculita I have ALC287, I don't know the difference with ALC3306.

Yes, but did you obtain this information from alsa-info, or did you check the Product Specification Reference? As I said, alsa-info is not reliable in this case as it conflicts with the Lenovo Specification Reference.

Or to put it more simply, do you also have a Legion 7i, or another laptop?

Centralobe, 2021-03-30:
@lmontigny I tried Ubuntu 21.04 from a USB Boot device and yet no sound for me. Can you tell me about the product name and your bios version? May be our BIOS or hardware slightly differs?
(You access your bios by hitting f2 on startup)
My Legion is a 15IMH05 and my BIOS is a E9CN62WW(v4.07).

@AndreiMiculita  I also have ALC287

dantahoua, 2021-05-18:
Hello, same problem here with a Lenovo 13s Gen 2. Chipset is ALC3306 (sated on the spec list of the computer). No sound from speaker in Linux, tried many kernel, many distribution. Saw somewhere might be a "pin" problem. Anyway, headphone working but no speaker sound... :(

NeveIsa, 2021-05-21:
Confirmed on Lenovo Thinkbook 13s G2.
No sound on any linux distro with variety of Kernels.

Hosar, 2021-06-06:
Yeap same here tried several linux distro, laptop Legion 7i not getting sound from the speaker. But I do get it with the headphones.

jacktheripper19, 2021-06-07:
I wonder if this issue will ever get solved. There are many users facing the same issue in this ticket that was opened a year ago: https://bugzilla.kernel.org/show_bug.cgi?id=208555

gitZui, 2022-07-13:
I did some updates today and the sound finally works on my Lenovo Legion 7.
I use Linux Mint 20.2 Cinnamon and the 5.15.0-41-generic kernel.
Can anybody confirm that it works with their setup too?

KeithG33, 2022-07-17:
Whoa, just updated and it works!

fdiniello, 2022-07-25:
> Whoa, just updated and it works! Running 5.17.15-76051715-generic kernel on Pop OS 22.04 LTS

@KeithG33 did you have any custom setup? I'm still having the issue with 5.18.10-76051810-generic kernel on Pop OS.

tjarross, 2022-10-11:
Hello,

Any news on this ?

RustemProg, 2024-06-11:
Same issue, kernel 6.5.13

tomasvalek, 2024-06-14:
Same issue kernel 6.8.0-35

denniskuligin, 2025-02-20:
2025 xD And the issue still persist http://alsa-project.org/db/?f=8f20f33efefb66870ad0c95434c5819a7fa1d136

Hope1337, 2025-07-28:
same issue, kernel 6.8.0, legion pro 5