alsa-project/alsa-lib #86: Release change info in release [atom feed] messages?

mxmilkiib, 2020-10-15:
I'm currently working on an software update aggregation system - https://libreav.org/updates - and I'd like to add ALSA itself, but I don't think there is a feed for this other than https://github.com/alsa-project/alsa-lib/releases.atom.

However, the releases.atom entries do not contain the Changes summary that is used on [the wiki](https://www.alsa-project.org/wiki/Changes_v1.2.3.1_v1.2.3.2).

Might it be possible for the release messages to contain those texts?

perexg, 2020-10-15:
I'm afraid, wiki is currently the main source for the releases for alsa-lib. I don't suggest to rely on the github service.

mxmilkiib, 2020-10-15:
I understand the wiki is the main source currently, but my proposal/request was for a slight alteration in the release process; to copy and paste the release change text into the commit message (which would then flow into the GH releases.atom).

perexg, 2021-06-18:
The annotated tags was extended since release 1.2.5.

closed 2021-06-18 by perexg