alsa-project/alsa-lib #87: Release 1.2.4 has been modified

dvzrv, 2020-10-19:
Hi! I package alsa-lib for Arch Linux.

I have noticed, that the tag for 1.2.4 has been re-done. I have downloaded and used a tarball from October 16th:

```
$ sha512sum alsa-lib-1.2.4.tar.gz
cacff66b888ebd9646a2d1794613b38c2fdacc290f4c71e0f53b25d53372af977d3a01bed323f58fd87e79c4556e8166d242fe7632e2a65dda4fd26b466a4a5c  alsa-lib-1.2.4.tar.gz
```
while it is now

```
$ sha512sum alsa-lib-1.2.4.tar.gz                                                                                                                                                                                                                                    [130]
2dc8d026e42d444e23feaff28cf87ece5577bcb61984d6c690c3a0c848f7206cb9c8838fccaf8e9ea58f32fce7f66a88da2381f8795d52f52662739da0414176  alsa-lib-1.2.4.tar.gz
```

The difference is:

```
$diffoscope old/alsa-lib-1.2.4.tar.gz alsa-lib-1.2.4.tar.gz

--- old/alsa-lib-1.2.4.tar.gz
+++ alsa-lib-1.2.4.tar.gz
├── alsa-lib-1.2.4.tar
│ ├── file list
│ │ @@ -1,415 +1,415 @@
│ │ -drwxrwxr-x   0 root         (0) root         (0)        0 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/
│ │ -drwxrwxr-x   0 root         (0) root         (0)        0 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/.github/
│ │ -drwxrwxr-x   0 root         (0) root         (0)        0 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/.github/workflows/
│ │ --rw-rw-r--   0 root         (0) root         (0)     2685 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/.github/workflows/build.yml
│ │ --rw-rw-r--   0 root         (0) root         (0)      934 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/.gitignore
│ │ --rw-rw-r--   0 root         (0) root         (0)      136 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/.travis.yml
│ │ --rw-rw-r--   0 root         (0) root         (0)    26440 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/COPYING
│ │ --rw-rw-r--   0 root         (0) root         (0)     1347 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/ChangeLog
│ │ --rw-rw-r--   0 root         (0) root         (0)     3866 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/INSTALL
│ │ --rw-rw-r--   0 root         (0) root         (0)      469 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/MEMORY-LEAK
│ │ --rw-rw-r--   0 root         (0) root         (0)      819 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/Makefile.am
│ │ --rw-rw-r--   0 root         (0) root         (0)     2089 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/NOTES
│ │ --rw-rw-r--   0 root         (0) root         (0)      926 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/README.md
│ │ --rw-rw-r--   0 root         (0) root         (0)      246 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/TODO
│ │ --rw-rw-r--   0 root         (0) root         (0)     1244 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/acinclude.m4
│ │ -drwxrwxr-x   0 root         (0) root         (0)        0 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/alsalisp/
│ │ --rw-rw-r--   0 root         (0) root         (0)      173 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/alsalisp/Makefile.am
│ │ --rw-rw-r--   0 root         (0) root         (0)     2684 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/alsalisp/alsalisp.c
│ │ --rw-rw-r--   0 root         (0) root         (0)     2700 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/alsalisp/hctl.lisp
│ │ --rw-rw-r--   0 root         (0) root         (0)      863 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/alsalisp/hello.lisp
│ │ --rw-rw-r--   0 root         (0) root         (0)       38 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/alsalisp/itest.lisp
│ │ --rw-rw-r--   0 root         (0) root         (0)     9787 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/alsalisp/test.lisp
│ │ -drwxrwxr-x   0 root         (0) root         (0)        0 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/aserver/
│ │ --rw-rw-r--   0 root         (0) root         (0)    18010 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/aserver/COPYING
│ │ --rw-rw-r--   0 root         (0) root         (0)      244 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/aserver/Makefile.am
│ │ --rw-rw-r--   0 root         (0) root         (0)    25870 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/aserver/aserver.c
│ │ --rw-rw-r--   0 root         (0) root         (0)    26195 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/configure.ac
│ │ -drwxrwxr-x   0 root         (0) root         (0)        0 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/doc/
│ │ --rw-rw-r--   0 root         (0) root         (0)      750 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/doc/Makefile.am
│ │ --rw-rw-r--   0 root         (0) root         (0)       84 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/doc/README.1st
│ │ --rw-rw-r--   0 root         (0) root         (0)    11473 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/doc/asoundrc.txt
│ │ --rw-rw-r--   0 root         (0) root         (0)     4432 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/doc/doxygen.cfg.in
│ │ --rw-rw-r--   0 root         (0) root         (0)     2338 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/doc/index.doxygen
│ │ -drwxrwxr-x   0 root         (0) root         (0)        0 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/doc/pictures/
│ │ --rw-rw-r--   0 root         (0) root         (0)       31 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/doc/pictures/Makefile.am
│ │ --rw-rw-r--   0 root         (0) root         (0)     1715 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/doc/pictures/wave1.gif
│ │ --rw-rw-r--   0 root         (0) root         (0)     1020 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/doc/pictures/wave2.gif
│ │ --rwxrwxr-x   0 root         (0) root         (0)     2056 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/gitcompile
│ │ -drwxrwxr-x   0 root         (0) root         (0)        0 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/include/
│ │ --rw-rw-r--   0 root         (0) root         (0)     2751 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/include/Makefile.am
│ │ --rw-rw-r--   0 root         (0) root         (0)     2210 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/include/alisp.h
│ │ --rw-rw-r--   0 root         (0) root         (0)     2169 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/include/alsa-symbols.h
│ │ --rw-rw-r--   0 root         (0) root         (0)     4096 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/include/aserver.h
│ │ --rw-rw-r--   0 root         (0) root         (0)    20709 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/include/asoundef.h
│ │ --rw-rw-r--   0 root         (0) root         (0)     1325 2020-10-15 17:15:15.000000 alsa-lib-1.2.4/include/asoundlib-head.h
[... 1030 lines cut ...]
│ │  {
│ │ +	int err;
│ │ +	char *value;
│ │ +
│ │ +	err = get_value(uc_mgr, "Linked", &value, NULL, NULL, 1);
│ │ +	if (err >= 0) {
│ │ +		err = strcasecmp(value, "true") == 0 ||
│ │ +		      strcmp(value, "1") == 0;
│ │ +		free(value);
│ │ +		if (err)
│ │ +			return 0;
│ │ +	}
│ │  	if (!list_empty(&uc_mgr->verb_list))
│ │  		return 0;
│ │  	if (!list_empty(&uc_mgr->boot_list))
│ │  		return 0;
│ │  	return -ENXIO;
│ │  }

```

While I understand that you would like to ship a fix for a release, **please** do not retag. Just add a new tag.
From a downstream perspective retagging is very problematic, as investigation in regards to a possible [supply chain attack](https://en.wikipedia.org/wiki/Supply_chain_attack) has to be done and it of course breaks the reproducibility of the package.

For the future I would like to encourage you to maybe consider doing detached PGP signatures for your release tarballs.

perexg, 2020-10-19:
The problem is that github is not our release place. The packagers should wait to the final tar ball on alsa-project.org. I will reconsider the release process to not create the annotate tag until the package passed all tests.

> For the future I would like to encourage you to maybe consider doing detached
> PGP signatures for your release tarballs.

I will consider also this option.

closed 2020-10-19 by perexg

dvzrv, 2020-10-19:
> The packagers should wait to the final tar ball on alsa-project.org.

Please document this. I was not aware of this requirement, especially as this github project is advertised on the project home page (and I am accustomed to custom release tarballs by projects being out-of-date due to manual release processes - which is why auto-generated tarballs based on tags are definitely preferred).

The current advertised download locations on alsa-project.org are all FTP-based without TLS by default, which makes them less attractive to use for packaging purposes due to the higher possibility of [man-in-the-middle attacks](https://en.wikipedia.org/wiki/Man-in-the-middle_attack).

> I will reconsider the release process to not create the annotate tag until the package passed all tests.

Thanks! As mentioned via e-mail this might also be solvable by doing pre-releases (e.g. alpha, beta, rc).