alsa-project/alsa-lib #88: ASoC: error at snd_soc_pcm_dai_trigger on acp_rn_pdm_dma.0

JcdeA, 2020-10-19:
I am on Artix and this has been an issue for a while. Pauvcontrol shows Dummy Output as the only output available.

https://www.reddit.com/r/archlinux/comments/e73u68/dummy_audio_output_after_update/  
I saw this post and downgraded alsa-lib-1.2.4 to alsa-lib-1.2.3 and it works perfectly now. 

 I tried all of the solutions on the fourms, but none worked except downgrading. Seems like a issue with the latest version.

perexg, 2020-10-20:
Dup of https://github.com/alsa-project/alsa-ucm-conf/issues/55 .

closed 2020-10-20 by perexg

dvzrv, 2020-10-20:
For reference: https://github.com/alsa-project/alsa-lib/issues/87

Will downgrade things later today, unless a release is close.