alsa-project/alsa-lib #89: ALSA error on finding "working profile" for a USB MIDI device.

radiobla, 2020-10-20:
TL;DR: After upgrading alsa-* packages yesterday, on the next reboot, systemd (apparently, pulseaudio.service) begun to show an error message related to "module-alsa-card" not finding a "working profile" for a USB MIDI device I have connected. Despite the error message, the device is working fine (through "raw MIDI" and also with jack2). Anyway, I want go further and find the root cause of this problem, since it's very annoying to see the Sound Service failing on every reboot.

Here is my pacman.log from yesterday:

**[ALPM] upgraded alsa-topology-conf (1.2.3-1 -> 1.2.4-1)
[ALPM] upgraded alsa-ucm-conf (1.2.3-2 -> 1.2.4-1)
[ALPM] upgraded alsa-lib (1.2.3.2-1 -> 1.2.4-1)
[ALPM] upgraded alsa-utils (1.2.3-2 -> 1.2.4-1)
[ALPM] upgraded lib32-alsa-lib (1.2.3.2-1 -> 1.2.4-1)**

After a reboot, the Sound Service could start, but begun to show an error message.

Taking a look on the systemd logs with

journalctl -xe --user-unit pulseaudio

I found this:

systemd[1027]: Starting Sound Service...
...
**pulseaudio[1037]: Failed to find a working profile.
pulseaudio[1037]: Failed to load module "module-alsa-card" (argument: "device_id="3" name="usb-Roland_A->
pulseaudio[1037]: module-rescue-stream is obsolete and should no longer be loaded. Please remove it from>
systemd[1027]: Started Sound Service.**

I disabled the "load-module module-rescue-streams" on the default.pa, but the other issue persists. They seem not be related, IMO. I wanted to rule out that chance anyway.

By stopping pulseaudio service through systemd and running

pulseaudio -vvv

manually, I got more verbose information.

**E: [pulseaudio] module-alsa-card.c: Failed to find a working profile.
E: [pulseaudio] module.c: Failed to load module "module-alsa-card" (argument: "device_id="1" name="usb-Roland_A-Series_Keyboard-00" card_name="alsa_card.usb-Roland_A-Series_Keyboard-00" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed.  
I: [pulseaudio] module-udev-detect.c: Card /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/sound/card1 (alsa_card.usb-Roland_A-Series_Keyboard-00) failed to load module.**

Regarding specifically to the following error message...

**E: [pulseaudio] module-alsa-card.c: Failed to find a working profile.**

What does it mean it is not finding a "working profile"? How could I investigate it further?

It worth mentioning that the device (an USB Keyboard Controller) is working
perfectly with jack2. I tested it with an "aseqdump".

$ aseqdump -l
 Port    Client name                      Port name
  0:0    System                           Timer
  0:1    System                           Announce
 14:0    Midi Through                     Midi Through Port-0
 20:0    A-Series Keyboard                A-Series Keyboard Keyboard

$ aseqdump -p20:0
Waiting for data. Press Ctrl+C to end.
Source  Event                  Ch  Data
 20:0   Note on                 0, note 41, velocity 76
 20:0   Note on                 0, note 40, velocity 82

Although the device is working even with the error message, it's very annoying to see a "failed service" on initialization.

Thanks!

More occurrences of this error were provided on Arch Linux forum: [](https://bbs.archlinux.org/viewtopic.php?pid=1932801)

perexg, 2020-10-20:
Show the full PA debug log, please.

radiobla, 2020-10-20:
Do you mean the full "pulseaudio -v"?

If not, how do I get these specific logs you asked?

perexg, 2020-10-20:
> Do you mean the full "pulseaudio -v"?

Yes, 'pulseaudio -vvv' log.

radiobla, 2020-10-20:
Here it goes:

https://gist.github.com/radiobla/f823db57885ce65360826d2e143b7877#file-gistfile1-txt

If you need anything else, just tell me.

perexg, 2020-10-20:
Thank you. Could you also provide output from 'alsa-info.sh --no-upload'? It appears like a driver issue.

radiobla, 2020-10-20:
Sure.

For pa.log (pulseaudio -vvv output), consider this latest gist:

https://gist.github.com/radiobla/900bf9cc705dea6d6b4876308a0e57e3

The alsa-info.sh.txt is here:

https://gist.github.com/radiobla/8e8830bed6330c67738a173c379e8f85

Thanks

perexg, 2020-10-20:
Thanks. Do you have this control with the previous kernel, too?

```
!!-------Mixer controls for card Keyboard

Card hw:3 'Keyboard'/'Roland A-Series Keyboard at usb-0000:00:14.0-10, full speed'
  Mixer name	: ''
  Components	: 'USB0582:0156'
  Controls      : 1
  Simple ctrls  : 1
Simple mixer control 'MIDI Input Mode',0
  Capabilities: enum
  Items: 'High Load' 'Light Load'
  Item0: 'Light Load'
```

radiobla, 2020-10-20:
Yes. It begun after the the alsa-* packages upgrade, but Linux Kernel was 5.8.x and was upgraded to 5.9.x (details in the provided gist).

In the previous kernel, it was working fine with no error messages. It keeps working with the current 5.9 kernel, although it show the error message.

perexg, 2020-10-23:
The alsa-info output from the previous kernel may be helpful. Thanks.

radiobla, 2020-10-23:
I'll downgrade the kernel version and provide you the alsa-info soon.

radiobla, 2020-11-11:
The issue keeps happening on Linux kernel 5.9.8.

I was unable to downgrade the kernel to the previous (supposedly) working version past days, but anyway it seems not related with the kernel directly, IMO.

```
-- Reboot --
nov 11 14:27:40 - systemd[996]: Starting Sound Service...
nov 11 14:27:41 - pulseaudio[1006]: Failed to find a working profile.
nov 11 14:27:41 - pulseaudio[1006]: Failed to load module "module-alsa-card" (argument: "device_id="3" name="usb-Roland_A-Series_Keyboard-00" card_name="alsa_card.usb-Roland_A-Series_Keyboard-00" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed.
nov 11 14:27:41 - systemd[996]: Started Sound Service.
```
Is one able to create the "profile" for this specific device? Where would I begin? Is there anyway I can contribute to make this "profile" to work?

As I said before, the device is working flawlessly, but the error message keeps happening. It bugs me somehow, even if I don't read the Pulseaudio journal everytime. :)

mbenkmann, 2020-12-12:
See here for how to make pulseaudio ignore your device:

https://unix.stackexchange.com/questions/473846/how-does-pulseaudio-determine-which-alsa-devices-to-make-available-or-not

perexg, 2020-12-29:
Stalled bug. Closing.

closed 2020-12-29 by perexg

marian556, 2021-04-02:
please reopen this

PrismaPhonic, 2021-06-09:
I agree, please re-open this. I'm hitting this as well now.

perexg, 2021-06-09:
It's unlikely alsa-lib issue. The kernel driver does not provide any PCM device and pulseaudio thinks that it's an error. I suggest to create a bug for pulseaudio: https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues

perexg, 2021-06-09:
And you can create an udev entry in `/lib/udev/rules.d/*pulseaudio.rules` file (or in /etc if you like) - look for `ENV{PULSE_IGNORE}="1"` lines.

JHW46, 2021-06-10:
I'm getting this intermittently (about 50% of the time) since upgrading Kubuntu to 21.04.

fredbr, 2021-06-28:
Downgrading all alsa related packages to 1.2.3-2 solved this issue for me

PrismaPhonic, 2021-12-14:
So frustrating this is still an open issue. If users are solving this by downgrading ALSA, then this is an ALSA issue. The redirection of blame to other projects is absurd.

Eggbertx, 2022-01-01:
That doesn't really sound like the issue is solved since that's just a workaround for an issue that still exists in the latest version.

samueldy, 2022-07-14:
Just want to report that I'm encountering the same issue with `alsa-lib` 1.2.7.2 on Manjaro on a Thinkpad P17. The device in question is a MOTU M2 over USB-A and the "failure to load profile" message is eventually followed by a core dump of the `pulseaudio` service. Downgrading `alsa-lib` and `lib32-alsa-lib` to 1.2.6.1 fixed the issue.

Banbeucmas, 2022-07-15:
Confirm the exact same problem with @samueldy on the exact same device and exact `alsa-lib` with Arch Linux. Same solution applies and solved the issue

MolarFox, 2022-08-21:
Also reporting a near identical scenario as @samueldy: 
- MOTU M2 audio interface over USB-A, Arch Linux on a Dell Inspiron 15 7559, latest versions of `alsa-lib` (1.2.7.2-1) and `lib32-alsa-lib` (1.2.7.2-1)
- Results in pulseaudio service crashing w/ core dump if the audio interface is present on boot, or if interface is connected while system is live

Downgrading has also been a successful workaround for me, though this bug still occurs when using latest versions of alsa-lib

thewriteway, 2022-11-01:
latest kernel: Linux 5.15.74-v8+

I get a similar issue with no devices plugged in:

pulseaudio
Failed to load module "module-alsa-card" (argument: "device_id="2" name="platform-fef05700.hdmi" card_name="alsa_card.platform-fef05700.hdmi" namereg_fail=false tsched=no fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed.
CODE_FILE	pulsecore/module.c
CODE_FUNC	pa_module_load
CODE_LINE	196
PULSE_BACKTRACE	
SYSLOG_IDENTIFIER	pulseaudio
_CMDLINE	/usr/bin/pulseaudio --daemonize=no --log-target=journal
_COMM	pulseaudio
_EXE	/usr/bin/pulseaudio
_SOURCE_REALTIME_TIMESTAMP	1667303144015320
_SYSTEMD_CGROUP	/user.slice/user-1000.slice/user@1000.service/app.slice/pulseaudio.service
_SYSTEMD_USER_UNIT	pulseaudio.service
_TRANSPORT	journal

hbarnard, 2023-02-09:
I'm getting this from pulseaudio on:
`Linux 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux`

```
module-alsa-card.c: Failed to find a working profile.
E: [pulseaudio] module.c: Failed to load module "module-alsa-card" (argument: "device_id="3" name="platform-soc_sound" card_name="alsa_card.platform-soc_sound" namereg_fail=false tsched=no fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed
```
I'm trying to use libav-audio with libcamera-vid to record an mp4.