alsa-project/alsa-lib #91: src/dlmisc.c:snd_dlpath_unlock() doesn't build if DL_ORIGIN_AVAILABLE is not defined

ollieparanoid, 2020-10-22:
Hey all, thanks for maintaining alsa.

In musl libc based Alpine Linux, the build of v1.2.4 fails with:
```
dlmisc.c:95:7: error: 'snd_plugin_dir_set' undeclared (first use in this function)
   95 |  if (!snd_plugin_dir_set) {
      |       ^~~~~~~~~~~~~~~~~~
dlmisc.c:95:7: note: each undeclared identifier is reported only once for each function it appears in
dlmisc.c:98:4: error: 'snd_plugin_dir' undeclared (first use in this function)
   98 |    snd_plugin_dir = strdup(env);
      |
```

This is a regression from https://github.com/alsa-project/alsa-lib/commit/8580c081c25678d11278efcb61bd15cf44d0a225, where the #ifdef...#else...#endif was removed.

perexg, 2020-10-22:
Oops. Fixed in ad8c8e5503980295dd8e5e54a6285d2d7e32eb1e .

closed 2020-10-22 by perexg

ollieparanoid, 2020-10-22:
Patch works as expected, thanks for the quick fix!

wymiatacz, 2020-10-29:
Hi,
What is the purpose of snd_plugin_dir[_set] when DL_ORIGIN_AVAILABLE is undefined?
Maybe handling this case in snd_dlpath() would be more appropriate?
I mean:
```
int snd_dlpath(char *path, size_t path_len, const char *name)
{
#ifdef DL_ORIGIN_AVAILABLE
...
#else
	snprintf(path, path_len, "%s/%s", ALSA_PLUGIN_DIR, name);
#endif
	return 0;
}

```

perexg, 2020-10-29:
Check the recent code. Those variables are also used for the getenv() output.

bramtayl, 2021-04-16:
Hey all, thanks for maintaining alsa (to echo ollieparanoid above)

I think I'm running into this here: https://github.com/JuliaPackaging/Yggdrasil/pull/2849 on the musl builds. It looks like this hasn't been part of a release yet? Will there be a release soon?