alsa-project/alsa-lib #93: how to properly release all memory after use?

xatian, 2020-10-27:
```C++
int main () {
    snd_pcm_t* handle;
    int err;
    err = ::snd_pcm_open (&handle, "default", SND_PCM_STREAM_PLAYBACK, SND_PCM_NONBLOCK);
    ::printf ("snd_pcm_open: %d\n", err);
    err = ::snd_pcm_close (handle);
    ::printf ("snd_pcm_close: %d\n", err);
    ::snd_config_update_free_global ();
    return 0;
}
```

```
==51583== Memcheck, a memory error detector
==51583== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==51583== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==51583== Command: ./alsa-leak
==51583== 
snd_pcm_open: 0
snd_pcm_close: 0
==51583== 
==51583== HEAP SUMMARY:
==51583==     in use at exit: 64,140 bytes in 140 blocks
==51583==   total heap usage: 2,937 allocs, 2,797 frees, 635,030 bytes allocated
==51583== 
==51583== 4 bytes in 1 blocks are definitely lost in loss record 3 of 71
==51583==    at 0x483AB65: calloc (vg_replace_malloc.c:760)
==51583==    by 0x58E0A3E: pa_xmalloc0 (in /usr/lib/x86_64-linux-gnu/libpulse.so.0.21.1)
==51583==    by 0x58B5F9E: pa_context_new_with_proplist (in /usr/lib/x86_64-linux-gnu/libpulse.so.0.21.1)
==51583==    by 0x4848216: ???
==51583==    by 0x4904ABD: ??? (in /usr/lib/x86_64-linux-gnu/libasound.so.2.0.0)
==51583==    by 0x4904F24: ??? (in /usr/lib/x86_64-linux-gnu/libasound.so.2.0.0)
==51583==    by 0x4905849: snd_config_searcha_hooks (in /usr/lib/x86_64-linux-gnu/libasound.so.2.0.0)
==51583==    by 0x4905940: snd_config_searchva_hooks (in /usr/lib/x86_64-linux-gnu/libasound.so.2.0.0)
==51583==    by 0x4905A42: ??? (in /usr/lib/x86_64-linux-gnu/libasound.so.2.0.0)
==51583==    by 0x49042D8: snd_config_search_definition (in /usr/lib/x86_64-linux-gnu/libasound.so.2.0.0)
==51583==    by 0x4920255: ??? (in /usr/lib/x86_64-linux-gnu/libasound.so.2.0.0)
==51583==    by 0x4923227: snd_pcm_open (in /usr/lib/x86_64-linux-gnu/libasound.so.2.0.0)
==51583== 
==51583== 4 bytes in 1 blocks are definitely lost in loss record 4 of 71
==51583==    at 0x483AB65: calloc (vg_replace_malloc.c:760)
==51583==    by 0x58E0A3E: pa_xmalloc0 (in /usr/lib/x86_64-linux-gnu/libpulse.so.0.21.1)
==51583==    by 0x58B5F9E: pa_context_new_with_proplist (in /usr/lib/x86_64-linux-gnu/libpulse.so.0.21.1)
==51583==    by 0x5892577: ???
==51583==    by 0x5891F42: ???
==51583==    by 0x491FC84: ??? (in /usr/lib/x86_64-linux-gnu/libasound.so.2.0.0)
==51583==    by 0x49202E6: ??? (in /usr/lib/x86_64-linux-gnu/libasound.so.2.0.0)
==51583==    by 0x4923227: snd_pcm_open (in /usr/lib/x86_64-linux-gnu/libasound.so.2.0.0)
==51583==    by 0x404A5B: main (src/main.cpp:15)
==51583== 
==51583== LEAK SUMMARY:
==51583==    definitely lost: 8 bytes in 2 blocks
==51583==    indirectly lost: 0 bytes in 0 blocks
==51583==      possibly lost: 0 bytes in 0 blocks
==51583==    still reachable: 64,132 bytes in 138 blocks
==51583==         suppressed: 0 bytes in 0 blocks
==51583== Reachable blocks (those to which a pointer was found) are not shown.
==51583== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==51583== 
==51583== For lists of detected and suppressed errors, rerun with: -s
==51583== ERROR SUMMARY: 2 errors from 2 contexts (suppressed: 0 from 0)
```

Did I do something wrong?
What makes matters worse is that if I duplicate the code more leaks get reported so this is not just some global state that gets left behind. And what about the 64kB still reachable?
I am using libasound2 1.2.3.2-1 amd64

Thank you!

xatian, 2020-10-27:
I tried different devices but haven't managed to get no leaks or a leak without libpulse.so.0.21.1 in it. (The device you suggested also produces a leak where libpulse is part of the stacktrace.)

Your guess might be correct ... but it might also be the case that you call pa_context_new_with_proplist() but forgot a call to pa_context_unref() when closing the pcm.

perexg, 2020-10-27:
Your setup is suspicious:

<pre>
$ valgrind --leak-check=full --show-leak-kinds=all aplay -d 1 -D plughw:0  /dev/zero
==279218== Memcheck, a memory error detector
==279218== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==279218== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==279218== Command: aplay -d 1 -D plughw:0 /dev/zero
==279218== 
Playing raw data '/dev/zero' : Unsigned 8 bit, Rate 8000 Hz, Mono
==279218== 
==279218== HEAP SUMMARY:
==279218==     in use at exit: 0 bytes in 0 blocks
==279218==   total heap usage: 2,244 allocs, 2,244 frees, 137,194 bytes allocated
==279218== 
==279218== All heap blocks were freed -- no leaks are possible
==279218== 
==279218== For lists of detected and suppressed errors, rerun with: -s
==279218== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
</pre>

xatian, 2020-10-28:
very interesting. on my machine I get the same memory leak and the same stacktrace with your command-line.
So you can confirm that I did not miss some function-call (like snd_config_update_free_global() which is at first looks not obvious) "during" cleanup?
Any would you make another guess as to why/how this might be happening on my setup. I am using a default debian 11 installation with a few extra dev-packages installed.
Thank you!

perexg, 2020-10-28:
If I compile your simple code:

<pre>
$ cat a.c
#include <stdio.h>
#include <alsa/asoundlib.h>

int main () {
    snd_pcm_t* handle;
    int err;
    err = snd_pcm_open (&handle, "plughw:0", SND_PCM_STREAM_PLAYBACK, SND_PCM_NONBLOCK);
    printf ("snd_pcm_open: %d\n", err);
    err = snd_pcm_close (handle);
    printf ("snd_pcm_close: %d\n", err);
    snd_config_update_free_global ();
    return 0;
}
$ gcc -g -lasound -o a a.c
$ valgrind --leak-check=full --show-leak-kinds=all ./a
$ valgrind --leak-check=full --show-leak-kinds=all ./a
==293640== Memcheck, a memory error detector
==293640== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==293640== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==293640== Command: ./a
==293640== 
snd_pcm_open: 0
snd_pcm_close: 0
==293640== 
==293640== HEAP SUMMARY:
==293640==     in use at exit: 0 bytes in 0 blocks
==293640==   total heap usage: 2,183 allocs, 2,183 frees, 121,149 bytes allocated
==293640== 
==293640== All heap blocks were freed -- no leaks are possible
==293640== 
==293640== For lists of detected and suppressed errors, rerun with: -s
==293640== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
</pre>

xatian, 2020-10-29:
I see ... thank you for your time and effort!

closed 2020-10-29 by xatian