alsa-project/alsa-lib #95: No more sound through HDMI / DisplayPort with 1.2.4

fkroener, 2020-11-02:
The recent update might have broken HDMI / DisyplayPort sound on my Thinkpad T14 (AMD), using ArchLinux with 5.9.2 kernel.
[old-alsa.log](https://github.com/alsa-project/alsa-lib/files/5474401/old-alsa.log) has the log of pulseaudio -vvvvv after the downgrade; [new-alsa.log](https://github.com/alsa-project/alsa-lib/files/5474402/new-alsa.log) is 1.2.4 - where the device name as well changed. 
Strangely only after a reboot the previous alsa-lib appears to come in effect and working, see: 
[rebooted-alsa.log](https://github.com/alsa-project/alsa-lib/files/5474440/rebooted-alsa.log)

Maybe this is a bug on my side, any pointers would be highly appreciated. I'll happily supply more logs or information if necessary. Thanks.

perexg, 2020-11-02:
Can you select the HDMI device manually? Does it work? The HDMI outputs are detected with UCM. Please, show also output from 'alsa-info.sh --no-upload' and 'pactl list'.

fkroener, 2020-11-02:
Here my alsa-info:
[alsa-info.txt](https://github.com/alsa-project/alsa-lib/files/5474668/alsa-info.txt) 
and 
[pactl-list.log](https://github.com/alsa-project/alsa-lib/files/5474678/pactl-list.log)
with alsa-lib pre-1.2.4.
Selection is possible, I just get no sound.

perexg, 2020-11-02:
Please, show the 'pactl list' with the problematic alsa-lib/alsa-ucm-conf.

fkroener, 2020-11-02:
alsa-ucm has been 1.2.4 in all cases.

now with alsa-lib 1.2.4 I have the following pactl list output: [pactl-list-1.2.4.txt](https://github.com/alsa-project/alsa-lib/files/5474923/pactl-list-1.2.4.txt)

alsa-lib 1.2.4 changes the output name from "HDMI / DisplayPort" to "HDMI 1" and "HDMI 2".

Interestingly now with rebooted 1.2.4 I DO get audio through DisplayPort (USB-C). But no audio through the HDMI port anymore.

perexg, 2020-11-02:
It's a bit problematic to dig through logs with the various hw configs. It seems that you connected two devices now.

Please, show clearly the problem with one HDMI device attached. Provide all logs for working and non-working case for the comparison ('alsa-info.sh --no-upload', 'pactl list', pulseaudio debug logs). Thank you. Please, also provide all logs when you activated the HDMI output (something is playing) to see the "runtime" state while you're collecting the logs.

fkroener, 2020-11-02:
Yes, there was HDMI and DisplayPort connected.
Attached files show HDMI and internal speaker only, pulseaudio verbose output while switching between HDMI and internal.
[pulseaudio-verbose-hdmi-only-switching.log](https://github.com/alsa-project/alsa-lib/files/5475715/pulseaudio-verbose-hdmi-only-switching.log)
[pactl-list-1.2.4-hdmi-only-playing.txt](https://github.com/alsa-project/alsa-lib/files/5475717/pactl-list-1.2.4-hdmi-only-playing.txt)
[alsa-info.txt](https://github.com/alsa-project/alsa-lib/files/5475719/alsa-info.txt)

perexg, 2020-11-02:
Could you run `amixer -c 0 set IEC958,1 on` on the command line when something is playing via HDMI connector with 1.2.4?

perexg, 2020-11-02:
I believe that the above change https://github.com/alsa-project/alsa-ucm-conf/commit/33be660e4b1e75c19d5332556c3d2636dd3344bf should fix the issue. Thank you for the report.

fkroener, 2020-11-02:
Thank you. Let's see when archlinux gets these, so I can test. :-)

Strange though, at least to my limited understanding, that a change in ucm-conf appears to be affected by alsa lib version.

perexg, 2020-11-02:
It's easy to test the change (as root):

````
cd /usr/share/alsa
wget -O a.patch https://github.com/alsa-project/alsa-ucm-conf/commit/33be660e4b1e75c19d5332556c3d2636dd3344bf.patch
cat a.patch | patch -p1
````

... and reboot or restart pulseaudio.

fkroener, 2020-11-02:
I needed to reboot, probably because the device was already unmuted by alsa-lib 1.2.3.

Upgrade to 1.2.4 -> Reboot -> Sound still not available through HDMI2, patch, restart pulseaudio -> Sound available. :-)

So yes, that patch appears to fix my issue. Thanks a lot.

Can we have "HDMI / DisplayPort" back for the name of the device?

perexg, 2020-11-02:
Change device name: https://github.com/alsa-project/alsa-ucm-conf/commit/f8c2428ccc42488c5e96801cce9f7e28080ecb71

Yes, we need a bugfix release.

closed 2020-11-02 by perexg

The-Compiler, 2021-03-06:
Any update regarding a bugfix release? I have the same issue with a T14 as well, and I can confirm the patch above works.