alsa-project/alsa-lib #96: (Regression) Broken pipe when streaming music from bandcamp with mpv

orbea, 2020-11-04:
OS: `Slackware64-current`
alsa-lib: `1.2.4`
mpv: https://github.com/mpv-player/mpv/commit/172146e9f7a231b5de21921d883612d18b13a717
yt-dlc: `2020.10.31`

When streaming music from bandcamp.com with alsa it will fail with a broken pipe when automatically changing tracks at the end of a track. This will not happen if the track is skipped.
```
$ mpv https://actnoir1.bandcamp.com/album/shape-a-new-start 

Playing: https://actnoir1.bandcamp.com/track/shatterproof-beauty
 (+) Audio --aid=1 (mp3 2ch 44100Hz)
AO: [alsa] 44100Hz stereo 2ch float
A: 00:05:20 / 00:05:21 (100%) Cache: 0.0s

Playing: https://actnoir1.bandcamp.com/track/set-fire
A: 00:05:20 / 00:05:21 (100%) Cache: 0.0s
[ao/alsa] snd_pcm_status: Broken pipe
A: 00:05:20 / 00:05:21 (100%) Cache: 0.0s
 (+) Audio --aid=1 (mp3 2ch 44100Hz)
[ao/alsa] snd_pcm_status: Broken pipe
[ao/alsa] snd_pcm_status: Broken pipe
[ao/alsa] snd_pcm_status: Broken pipe
mpv: ../audio/out/buffer.c:585: _Bool ao_play_data(struct ao *): Assertion `space >= 0' failed.
The process was killed by SIGABRT: Aborted
```
mpv log: [mpv.log](https://github.com/alsa-project/alsa-lib/files/5485135/mpv.log)
alsa-info log: [alsa-info.log](https://github.com/alsa-project/alsa-lib/files/5485141/alsa-info.log)

I bisected this and found the first bad commit.
```
$ git bisect good
4f90392f07e8822d1984ed990f622ad36022a4a3 is the first bad commit
commit 4f90392f07e8822d1984ed990f622ad36022a4a3
Author: Jaroslav Kysela <perex@perex.cz>
Date:   Fri Oct 9 19:57:57 2020 +0200

    pcm: fix the snd_pcm_plugin_status() avail and delay fields
    
    The avail and delay fields in the returned status structure does not
    reflect the actual hw_ptr/appl_ptr. This change correct this.
    
    TODO: Unfortunately, the delay might contain also information about
    extra hardware / buffering delay which is hidden with this change.
    
    Link: https://lore.kernel.org/alsa-devel/d9c1f37e-5c8d-f289-270e-c6cda7a56ce3@axis.com/
    Reported-by: Jonas Holmberg <jonashg@axis.com>
    Tested-by: Jonas Holmberg <jonashg@axis.com>
    Signed-off-by: Jaroslav Kysela <perex@perex.cz>

 src/pcm/pcm_plugin.c | 8 ++++++--
 1 file changed, 6 insertions(+), 2 deletions(-)
```
4f90392f07e8822d1984ed990f622ad36022a4a3

perexg, 2020-11-09:
You should consult this with the author of the mpv code. The free_samples variable is already in range 0..buffer_size - https://github.com/mpv-player/mpv/blob/master/audio/out/ao_alsa.c#L999 . It appears that recover_and_get_state() function https://github.com/mpv-player/mpv/blob/master/audio/out/ao_alsa.c#L920 returns early and the -1 value is used from https://github.com/mpv-player/mpv/blob/master/audio/out/buffer.c#L104 . There is no error checking in audio_get_state() https://github.com/mpv-player/mpv/blob/master/audio/out/ao_alsa.c#L1013.

closed 2020-11-09 by perexg

orbea, 2020-11-09:
@perexg This seems to have been fixed in mpv PR https://github.com/mpv-player/mpv/pull/8229 which is now merged, thanks for the feedback!