alsa-project/alsa-lib #98: USB Audio device: Shure MV7 not recognized

vchuravy, 2020-11-09:
Hello! i hope this is the right venue for reporting not working audio devices.

```
> alsa-capabilities
9: `USB Audio' Digital USB Audio Class audio output on device Shure MV7
    -hardware address:               hw:3,0  (or hw:MV7,0)
    -device label:                   Shure MV7
    -formats:                        (n/a)
    -usb audio class:                UAC2 (isochronous asynchronous)
    -streamfile:                     /proc/asound/card3/stream0
    -character device:               /dev/snd/pcmC3D0p
    -monitor file:                   /proc/asound/card3/pcm0p/sub0/hw_params
```

`alsa-info.sh`: http://alsa-project.org/db/?f=1fb51d03454930fce4d6164323cfb7b3c7c88dda
`pulseaudio -vvv` https://gist.github.com/vchuravy/f5f8ee972d5e2179387c8c53423379f6

During the pulseaudio session I plugged the device in and then unplugged it. I am happy to dig in, but I would need some guidance on where to start. As a side-note the microphone seems to work on Android so I am wondering if they are using a different driver.

perexg, 2020-12-29:
`arecord -d 3  -f U8 sample.mp3` should not use 'dsnoop' plugin as default. Show your asoundrc .

Shosta, 2021-01-14:
I got that same issue.
Found a workaround, but just for testing.
Unplug the mic
Kill the pulseaudio process pulseaudio --kill
replug the pic while pulseuaudio is not on yet

You can select your Shure MV7 in the Ubuntu Sound settings.
Nevertheless, as soon as pulseaudio is on. Shure MV7 is not working.

That's why I think it is a pulseaudio related issue.
Nevertheless, I don't know how to solve it ?

vchuravy, 2021-01-14:
I managed to get the microphone portion of this working by adding https://gist.github.com/vchuravy/a25ea9533099224a0c4604191f6c668d in /usr/share/alsa-card-profile/mixer/profile-sets/ (Archlinux)

and adding:

```
 cat /lib/udev/rules.d/91-shuremv7.rules 
SUBSYSTEM!="sound", GOTO="pulseaudio_end"
ACTION!="change", GOTO="pulseaudio_end"
KERNEL!="card*", GOTO="pulseaudio_end"

SUBSYSTEMS=="usb", ATTRS{idVendor}=="14ed", ATTRS{idProduct}=="1012", ENV{PULSE_PROFILE_SET}="shure-mv7.conf"

LABEL="pulsaudio_end"
```

It seems like the device freezes when being probed for Audio out.

perexg, 2021-01-14:
I would report this at https://bugzilla.kernel.org . Also, if you can, try the latest kernel where the support for USB implicit feedback devices are better supported. @tiwai - FYI.

closed 2021-01-14 by perexg

mc0de, 2021-01-22:
can confirm this is big problem, got MV7 especially for videos on linux... sadness, using Fedora 33, same stuff hapenning, mic is just looped in restart, @vchuravy solution didn't work either

ntl, 2021-02-03:
@mc0de I've made a very tiny bit of progress. First, I had to _remove_ an entry in `/etc/pulse/daemon.conf` that I had added from somewhere I found on the internet: `default-sample-rate = 48000`. Then I restart pulse audio. After pulse audio restarts, I remove my laptop's on-board device ("Family 17h", it's an Asus laptop) from pulseaudio using `pactl unload-module`. Then, my microphone "works" -- but there's a 50/50 chance that it doesn't get the sample rate correct, and the audio modulates down and everyone's voices sound off. When that happens, I restart pulse audio, unplug and plug the MV7 back in, re-do the `pactl unload-module` step, and hope it works correctly the next time.

I can't say for sure, but it seems like the zoom program might be making matters worse, as the microphone seems more likely to come up with the correct sample rate when zoom isn't running.

I wish I knew where and how I could direct my efforts to get to the bottom of this. Between alsa and pulseaudio I'm not really sure what tools I need to learn to use to even capture error telemetry.

perexg, 2021-02-03:
There are some improvements in the USB kernel driver which may fix this issue. You may try 5.11 vanilla kernels for Fedora (https://fedoraproject.org/wiki/Kernel_Vanilla_Repositories) or compile the latest upstream kernel yourself.

ntl, 2021-02-04:
Thanks for the tip. I tried 5.11.0-rc6, and unfortunately the behavior remains the same.

```
> uname -r
5.11.0-0.rc6.20210203git3aaf0a27ffc2.143.vanilla.1.fc33.x86_64
```

tlhunter, 2021-02-18:
For what it's worth, I have two Linux laptops, one works pretty well with the MV7 and the other does not. Here's some output from `dmesg` and `lsusb`:
https://askubuntu.com/questions/1317537/shure-mv7-usb-microphone-seems-works-with-thinkpad-but-not-xps

buytenh, 2021-02-21:
Yesterday, I looked with @mc0de into why his Shure MV7 doesn't seem to
be working so well under Linux (but seems to work OK under Windows on
the same hardware), and we found some interesting things.

For one, the microphone sometimes goes into a seemingly infinite
disconnect/reconnect loop when you first plug it into the USB port:

```
[5964855.233514] usb 1-3: new full-speed USB device number 11 using xhci_hcd
[5964855.595101] usb 1-3: New USB device found, idVendor=14ed, idProduct=1012, bcdDevice= 0.02
[5964855.595110] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=128
[5964855.595114] usb 1-3: Product: Shure MV7
[5964855.595118] usb 1-3: Manufacturer: Shure Inc
[5964855.653510] hid-generic 0003:14ED:1012.02B8: hiddev98,hidraw5: USB HID v1.11 Device [Shure Inc Shure MV7] on usb-0000:12:00.0-3/input3
[5964855.672598] input: Shure Inc Shure MV7 as /devices/pci0000:00/0000:00:01.2/0000:12:00.0/usb1/1-3/1-3:1.4/0003:14ED:1012.02B9/input/input1255
[5964855.724738] hid-generic 0003:14ED:1012.02B9: input,hidraw6: USB HID v1.11 Device [Shure Inc Shure MV7] on usb-0000:12:00.0-3/input4
[5964855.757278] mc: Linux media interface: v0.10
[5964855.964436] usbcore: registered new interface driver snd-usb-audio
[5964857.845174] usb 1-3: USB disconnect, device number 11
[5964861.056660] usb 1-3: 1:1: usb_set_interface failed (-110)
[5964861.300521] usb 1-3: new full-speed USB device number 12 using xhci_hcd
[5964861.663105] usb 1-3: New USB device found, idVendor=14ed, idProduct=1012, bcdDevice= 0.02
[5964861.663113] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=128
[5964861.663117] usb 1-3: Product: Shure MV7
[5964861.663120] usb 1-3: Manufacturer: Shure Inc
[5964861.883274] hid-generic 0003:14ED:1012.02BA: hiddev98,hidraw5: USB HID v1.11 Device [Shure Inc Shure MV7] on usb-0000:12:00.0-3/input3
[5964861.902183] input: Shure Inc Shure MV7 as /devices/pci0000:00/0000:00:01.2/0000:12:00.0/usb1/1-3/1-3:1.4/0003:14ED:1012.02BB/input/input1256
[5964861.954683] hid-generic 0003:14ED:1012.02BB: input,hidraw6: USB HID v1.11 Device [Shure Inc Shure MV7] on usb-0000:12:00.0-3/input4
[5964864.080443] usb 1-3: USB disconnect, device number 12
[5964867.200865] usb 1-3: 1:1: usb_set_interface failed (-110)
[5964867.446443] usb 1-3: new full-speed USB device number 13 using xhci_hcd
[5964867.807994] usb 1-3: New USB device found, idVendor=14ed, idProduct=1012, bcdDevice= 0.02
[5964867.808000] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=128
[5964867.808004] usb 1-3: Product: Shure MV7
[5964867.808006] usb 1-3: Manufacturer: Shure Inc
[5964868.028423] hid-generic 0003:14ED:1012.02BC: hiddev98,hidraw5: USB HID v1.11 Device [Shure Inc Shure MV7] on usb-0000:12:00.0-3/input3
[5964868.047096] input: Shure Inc Shure MV7 as /devices/pci0000:00/0000:00:01.2/0000:12:00.0/usb1/1-3/1-3:1.4/0003:14ED:1012.02BD/input/input1257
[5964868.098839] hid-generic 0003:14ED:1012.02BD: input,hidraw6: USB HID v1.11 Device [Shure Inc Shure MV7] on usb-0000:12:00.0-3/input4
[5964869.875728] usb 1-3: USB disconnect, device number 13
[5964873.344792] usb 1-3: 1:1: usb_set_interface failed (-110)
[5964873.584422] usb 1-3: new full-speed USB device number 14 using xhci_hcd
[5964873.946045] usb 1-3: New USB device found, idVendor=14ed, idProduct=1012, bcdDevice= 0.02
[5964873.946052] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=128
[5964873.946055] usb 1-3: Product: Shure MV7
[5964873.946058] usb 1-3: Manufacturer: Shure Inc
[5964874.166142] hid-generic 0003:14ED:1012.02BE: hiddev98,hidraw5: USB HID v1.11 Device [Shure Inc Shure MV7] on usb-0000:12:00.0-3/input3
[5964874.185227] input: Shure Inc Shure MV7 as /devices/pci0000:00/0000:00:01.2/0000:12:00.0/usb1/1-3/1-3:1.4/0003:14ED:1012.02BF/input/input1258
[5964874.236692] hid-generic 0003:14ED:1012.02BF: input,hidraw6: USB HID v1.11 Device [Shure Inc Shure MV7] on usb-0000:12:00.0-3/input4
[5964876.044858] usb 1-3: USB disconnect, device number 14
[5964879.488523] usb 1-3: 1:1: usb_set_interface failed (-110)
[5964879.734364] usb 1-3: new full-speed USB device number 15 using xhci_hcd
[5964880.095930] usb 1-3: New USB device found, idVendor=14ed, idProduct=1012, bcdDevice= 0.02
[5964880.095937] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=128
[5964880.095940] usb 1-3: Product: Shure MV7
[5964880.095943] usb 1-3: Manufacturer: Shure Inc
[5964880.316435] hid-generic 0003:14ED:1012.02C0: hiddev98,hidraw5: USB HID v1.11 Device [Shure Inc Shure MV7] on usb-0000:12:00.0-3/input3
[5964880.335012] input: Shure Inc Shure MV7 as /devices/pci0000:00/0000:00:01.2/0000:12:00.0/usb1/1-3/1-3:1.4/0003:14ED:1012.02C1/input/input1259
[5964880.386565] hid-generic 0003:14ED:1012.02C1: input,hidraw6: USB HID v1.11 Device [Shure Inc Shure MV7] on usb-0000:12:00.0-3/input4
[5964882.162390] usb 1-3: USB disconnect, device number 15
[5964885.632606] usb 1-3: 1:1: usb_set_interface failed (-110)
[5964885.883301] usb 1-3: new full-speed USB device number 16 using xhci_hcd
[...]
```

But sometimes, it only disconnects and reconnects a few times before
eventually stabilizing (and very rarely, there are no disconnect events
after you plug it in and it just works on the first try).

When the microphone does eventually stabilize, changing the device
profile in `pavucontrol`'s Configuration tab has a fair chance to send
[... 1 lines cut ...]
reliably happen for every profile change, and it doesn't reliably
[... cut ...]
the firmware crashes and the microphone reboots.

We tried adding extra hubs between the USB root hub and the microphone,
and that seems to make occurence of the disconnect/reconnect loops less
likely.  On a random test PC (running a 5.9 kernel), connecting the
microphone to one of the motherboard's front panel USB3 ports would
cause it to go into a reconnect loop almost every time, but when
connected to that same port through two hubs, we couldn't make it go
into a reconnect loop even after dozens of profile changes.

kevindherdt, 2021-02-22:
I did get the shure mv7 recognizable using vchuravy conf and his tweaks but it does not record any sound for me in ubuntu. I'm not that knowledge in linux but if there is anything I can do to test stuff feel free to ask me.

S-o-m-e-0-n-e, 2021-03-10:
> Yesterday, I looked with @mc0de into why his Shure MV7 doesn't seem to
> be working so well under Linux (but seems to work OK under Windows on
> the same hardware), and we found some interesting [...]

My "fix" was using an USB 2.x port (I had the reconnecting loop on USB 3.x ports):
https://askubuntu.com/a/1322557/1190923

ntl, 2021-04-05:
This week I tested the MV7 over a USB2 port. Wasn't easy since my laptop only has USB3 ports, so I had to dig up a USB dock that happened to have one USB2 connection on it. Happy to say it appears to work flawlessly. Thanks @S-o-m-e-0-n-e !

mc0de, 2021-04-05:
@ntl you're welcome

Olegblow, 2021-04-12:
oh it's work. 1 of 6 usb ports approached. I used Windows 6 months. today i can remove windows!!!!!

Shatur, 2021-06-12:
> My "fix" was using an USB 2.x port (I had the reconnecting loop on USB 3.x ports):

This not worked for me.

But I managed it to work using @vchuravy method + some manual steps:

1. Stop pulseaudio (`systemctl --user stop pulseaudio`.
2. Plug microphone.
3. Wait for initialization.
4. Enable pulseaudio again (`systemctl --user start pulseaudio`).

This should be done on every boot.
Pipiwire works the same, but `PULSE_PROFILE_SET` should be replaced with `ACP_PROFILE_SET` in `shure-mv7.conf`
For unknown reasons on Plasma with pulseaudio, when I open the volume settings in the tray, the microphone starts rebooting again. But with pipiwire it works correctly.

VadimOnix, 2021-07-04:
Use Ubuntu 21.04
Fix problem with recognize Shure mv7 on Ubuntu 21.04. I swapped pulseaudio on PipeWire.
https://github.com/pipewire-debian/pipewire-debian#1-ppa-configuration

and pasted @vchuravy [config file](https://gist.github.com/vchuravy/a25ea9533099224a0c4604191f6c668d) in
```/usr/share/alsa-card-profile/mixer/profile-sets```

After reboot system with unpluged device, and plug it back.

[image]