alsa-project/alsa-lib #101: SB Omni Surround 5.1

witoldo7, 2020-11-24:
Hi,
SB Omni Surround 5.1 use default hw 1 for iec958.0 please update USB-Audio.conf, 0 is used for dac(jack connectors)

perexg, 2021-01-07:
What's the exact string to be added? Or show the output from alsa-info.sh script. Thank you.

witoldo7, 2021-04-04:
I'm lazy so I just modified : 
```
USB-Audio.pcm.iec958.0 {
        @args [ CARD AES0 AES1 AES2 AES3 ]
        @args.CARD { type string }
        @args.AES0 { type integer }
        @args.AES1 { type integer }
        @args.AES2 { type integer }
        @args.AES3 { type integer }
        @func refer
        name {
                @func concat
                strings [
                        "cards.USB-Audio."
                        { @func card_name card $CARD }
                        ".pcm.iec958:CARD=" $CARD
                        ",AES0=" $AES0 ",AES1=" $AES1 ",AES2=" $AES2 ",AES3=" $AES3
                ]
        }
        default {
                # FIXME: we cannot set the AES parameters
                type hw
                card $CARD
                device {
                        @func refer
                        name {
                                @func concat
                                strings [
                                        "cards.USB-Audio.pcm.iec958_device."
                                        { @func card_name card $CARD }
                                ]
                        }
here -->>                        default 0  to 1
                }
        }
}
```
[https://pastebin.com/EmXLjVeq](url)

perexg, 2021-04-13:
Fixed in 8f2b64f0d7ff65d48359d6f786a17f6fd6688862 . Thank you.

closed 2021-04-13 by perexg