alsa-project/alsa-lib #104: module-alsa-card.c: Failed to find a working profile.

mhwombat, 2020-12-01:
I can get audio through my headphones, but not through my speakers.

Basic checks:

- Speakers work on another machine
- Speakers are plugged in and turned on
- Headphones are unplugged when using speakers
- My user account is in the audio group.
- Nothing is muted in alsamixer

I think the cause might be this `module-alsa-card.c: Failed to find a working profile.`

```
$ journalctl --user -u pulseaudio | tail -n 20
-- Reboot --
Nov 28 22:35:15 nixos systemd[2142]: Starting Sound Service...
Nov 28 22:35:15 nixos pulseaudio[2160]: E: [pulseaudio] module-alsa-card.c: Failed to find a working profile.
Nov 28 22:35:15 nixos pulseaudio[2160]: E: [pulseaudio] module.c: Failed to load module "module-alsa-card" (argument: "device_id="0" name="pci-0000_22_00.4" card_name="alsa_card.pci-0000_22_00.4" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed.
Nov 28 22:35:17 nixos pulseaudio[2160]: E: [pulseaudio] module-jackdbus-detect.c: Unable to contact D-Bus session bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Nov 28 22:35:17 nixos pulseaudio[2160]: E: [pulseaudio] module.c: Failed to load module "module-jackdbus-detect" (argument: "channels=2"): initialization failed.
Nov 28 22:35:17 nixos pulseaudio[2160]: W: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Nov 28 22:35:17 nixos pulseaudio[2160]: W: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Nov 28 22:35:17 nixos systemd[2142]: Started Sound Service.
Nov 28 22:35:17 nixos pulseaudio[2160]: E: [pulseaudio] bluez5-util.c: GetManagedObjects() failed: org.freedesktop.DBus.Error.ServiceUnknown: The name org.bluez was not provided by any .service files
Nov 29 17:24:50 nixos systemd[2142]: pulseaudio.service: Current command vanished from the unit file, execution of the command list won't be resumed.
Nov 29 17:25:00 nixos pulseaudio[2160]: E: [pulseaudio] module-alsa-card.c: Failed to find a working profile.
Nov 29 17:25:00 nixos pulseaudio[2160]: E: [pulseaudio] module.c: Failed to load module "module-alsa-card" (argument: "device_id="0" name="pci-0000_22_00.4" card_name="alsa_card.pci-0000_22_00.4" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed.
Nov 29 17:25:00 nixos pulseaudio[2160]: E: [pulseaudio] module-alsa-card.c: Failed to find a working profile.
Nov 29 17:25:00 nixos pulseaudio[2160]: E: [pulseaudio] module.c: Failed to load module "module-alsa-card" (argument: "device_id="0" name="pci-0000_22_00.4" card_name="alsa_card.pci-0000_22_00.4" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed.
Nov 29 17:25:01 nixos pulseaudio[2160]: E: [pulseaudio] module-alsa-card.c: Failed to find a working profile.
Nov 29 17:25:01 nixos pulseaudio[2160]: E: [pulseaudio] module.c: Failed to load module "module-alsa-card" (argument: "device_id="0" name="pci-0000_22_00.4" card_name="alsa_card.pci-0000_22_00.4" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed.
Dec 01 10:53:45 nixos pulseaudio[2160]: E: [alsa-sink-USB Audio] alsa-sink.c: ALSA woke us up to write new data to the device, but there was actually nothing to write.
Dec 01 10:53:45 nixos pulseaudio[2160]: E: [alsa-sink-USB Audio] alsa-sink.c: Most likely this is a bug in the ALSA driver 'snd_usb_audio'. Please report this issue to the ALSA developers.
Dec 01 10:53:45 nixos pulseaudio[2160]: E: [alsa-sink-USB Audio] alsa-sink.c: We were woken up with POLLOUT set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
```

Here is the [output of alsa-info_alsa-info.sh](https://alsa-project.org/db/?f=c5a24a41c9ab6f2b320362b6b0d3afee507d6cab).

perexg, 2020-12-01:
Please, show the PA debug output: https://fedoraproject.org/wiki/How_to_debug_PulseAudio_problems

mhwombat, 2020-12-01:
Output of `pulseaudio -vvvvv` is [here](https://gist.github.com/mhwombat/16ec1e3a850e2475ad5331f1d4a9d379)

Output of `pacmd ls` is [here](https://gist.github.com/mhwombat/12c17e4004cc7fb4ed7952e69bfdc4c1)

```
# pulseaudio --system
W: [pulseaudio] main.c: Running in system mode, but --disallow-exit not set.
W: [pulseaudio] main.c: Running in system mode, but --disallow-module-loading not set.
N: [pulseaudio] main.c: Running in system mode, forcibly disabling SHM mode.
N: [pulseaudio] main.c: Running in system mode, forcibly disabling exit idle time.
E: [pulseaudio] main.c: Failed to find user 'pulse'.
```

perexg, 2020-12-02:
Ok, this hardware consists from four cards:

HDA analog - which is not used at all (no codec is connected to the PCI bridge)
HDA HDMI
USB Front Audio
USB Main Audio

The 'Failed to find a working profile' belongs to the empty HDA card which should not active at the kernel driver level in my opinion. @tiwai ?

Could you switch between 'USB Front' and 'USB Main' card manually in PA (use pavucontrol app - settings / profiles or the pactl command line).

tiwai, 2020-12-02:
Yes, the board is TRX40 and the recent AMD platform, and there are already a few TRX40 devices in the deny list of snd-hda-intel.
But this machine seems to have a wrong PCI SSID (that isn't board-unique but keeps the chipset generic one from AMD).
Maybe the BIOS update will fix that PCI SSID?

mhwombat, 2020-12-02:
> Ok, this hardware consists from four cards:
>
> HDA analog - which is not used at all (no codec is connected to the PCI bridge)
> HDA HDMI
> USB Front Audio
> USB Main Audio
>
> The 'Failed to find a working [...]

I will try that. But just so you know, my speakers aren't plugged into a USB port; they're plugged into the "line out" jack.

mhwombat, 2020-12-02:
> Could you switch between 'USB Front' and 'USB Main' card manually in PA (use pavucontrol app - settings / profiles or the pactl command line).

@perexg Sorry, but I'm not clear on what you're suggesting that I do, or how to do it in either pactl or pavucontrol. I'm really confused between cards, drivers, sinks, modules, streams, etc.

perexg, 2020-12-03:
The TRX40 boards use 2 integrated USB audio chips which leads to this confusion. The USB driver represents this hardware as two ALSA USB cards (Main and Front). Just try to select the "Aorus Master Main Audio" card and corresponding profile in pavucontrol. It seems that PA creates only iec958 sink for it, so you may give a try.

Note that there are three playback devices for the main usb card:

```
card 3: Audio [Aorus Master Main Audio], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: Audio [Aorus Master Main Audio], device 1: USB Audio [USB Audio #1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: Audio [Aorus Master Main Audio], device 3: USB Audio [USB Audio #3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

So it may be possible that another PCM device is connected to Line Out for this hardware. You may try to find the right path using the direct ALSA commands. Try `speaker-test -D hw:3,1` and `speaker-test -D hw:3,3` on the command line, too. The PA uses only 'hw:3,0' PCM device. The mixer settings may be changed via `alsamixer -c 3`.

tiwai, 2020-12-03:
Actually I have forgotten to create two UCM profiles for those Gigabyte Aorus Master boards.
They should be similar like USB-Audio/Realtek-ALC1220-VB-Desktop, but there will be two for both main and front headphone.
Maybe it's a good chance to write and test such ones...

perexg, 2020-12-03:
It's possible to "merge" those two cards into one profile via UCM now. I believe that it may be more understandable for users.

mhwombat, 2020-12-03:
**Success!**

> The TRX40 boards use 2 integrated USB audio chips which leads to this confusion. The USB driver represents this hardware as two ALSA USB cards (Main and Front). Just try to select the "Aorus Master [...]

In pavucontrol, on the "Configuration" tab, I have three cards (I think the icon represents a card) listed. None of them are named "Aorus Master Main Audio". The cards and profile options are listed below.

- USB Audio
- USB Audio
- Navi 10 HDMI Audio

**Here's are the profile settings that worked for me:**

USB Audio: Off
USB Audio: Analog Stereo Output
Navi 10 HDMI Audio: Off

**The issue is resolved to my satisfaction, so please feel free to close it. But if you do more work related to this motherboard, and I can help in any way, please let me know. I can experiment with other settings for you.**

perexg, 2020-12-04:
Just a note: The 'Line Out Jack' control / presence detection does not work properly in this case.