alsa-project/alsa-lib #105: Bug in the ALSA driver 'snd_usb_audio'

rombru, 2020-12-12:
Hello,
I'm not an expert in linux systems.
I will try to give a maximum of information.

I've bought Bowers & Wilkins PX7 headphones. They doesn't work using USB on Ubuntu.
Using Ubuntu and bluetooth, or audio jack, the headphones work well. 
Using Windows 10 and USB, the headphones work well.

I d'like too use them with USB on Ubuntu because that recharge them in parallel.

Problems while the headphons are connected :
On Chrome : The videos are played too fast and there's no sound (Youtube & Twitch).
On Firefox : The videos are played too fast and there's a crackling sound (Youtube & Twitch).
On VLC : No sound

I've tried with Ubuntu 20.04 & 18.04.
They are set as default audio device
I've played with PulseAudio and alsamixer (reinstall, mute/unmute, ...)
The headphones are set as fallback in PulseAudio.

Trying to find the issue, i've make some test found on Google :
I've added pcie_aspm=off in GRUB_CMDLINE_LINUX_DEFAULT
Setting devices as default in /etc/asound.conf

--- The ALSA part

I think the problem come from ALSA, i've found the errors :
`Failed to load module "module-alsa-card`
followed few lines later by
`Most likely this is a bug in the ALSA driver 'snd_usb_audio'. Please report this issue to the ALSA developers.`

Output of `journalctl --user -u pulseaudio` is [here](https://gist.githubusercontent.com/rombru/eb4ca14168494e95c62a961f92e91a90/raw/52c93f01b9c617f823308183d32d07f8e9033bc1/journalctl.txt)
Output of `pulseaudio -vvvvv` is [here](https://gist.githubusercontent.com/rombru/eb4ca14168494e95c62a961f92e91a90/raw/52c93f01b9c617f823308183d32d07f8e9033bc1/pulseaudio.txt)
Output of `pacmd ls` is [here](https://gist.githubusercontent.com/rombru/eb4ca14168494e95c62a961f92e91a90/raw/52c93f01b9c617f823308183d32d07f8e9033bc1/pacmd.txt)

Please, tell me if you need something more to analyse the problem.

rombru, 2020-12-14:
Here is alsa-info, maybe it can help more.
Output of alsa-info is (here)[https://gist.githubusercontent.com/rombru/eb4ca14168494e95c62a961f92e91a90/raw/8dbe498aee573e8f3af2e5c617be46e8ee9fbc9e/alsa-info]

perexg, 2020-12-29:
Dup of #90.

closed 2020-12-29 by perexg