alsa-project/alsa-lib #109: dmix + dshare and changes of period_size buffer_size

pietercardoen, 2021-01-06:
Dear

I'm trying to create my audio configuration using dmix or dshare plugins but i'm facing some issues.

Currently, if I don't use dmix or dshare, I'm able to specify the period_size using aplay and changes are populated to the hardware. However if I include a dmix/dshare plugin, I'm  not able to configure the period_time anymore.

To configure dmix, one needs to specify this:
slave {			# Slave definition
  pcm STR		# slave PCM name
  # or
  pcm { }		# slave PCM definition
  format STR	# format definition
  rate INT	# rate definition
  channels INT
  period_time INT	# in usec
  # or
  period_size INT	# in frames
  buffer_time INT	# in usec
  # or
  buffer_size INT # in frames
  periods INT	# when buffer_size or buffer_time is not specified
}

Is it possible in any way to avoid the need to specify buffer_size/periods/... and to avoid the system to fall back on defaults?

Kind regards
Pieter Cardoen

perexg, 2021-01-07:
It's correct. The dmix is sharing the hardware. In your case, the first application will setup the parameters and others will share this settings which may be problematic.

aditpape, 2021-03-26:
You may use option 'var_periodsize true' in DSHARE/DSNOOP plugin.
This allows clients to configure 'exact multiple' of the slave period size.
Thus configure e.g slave to 2ms period and 64ms buffer size then each client can chose any period matching multiple of 2ms up to period of 32ms to meet requirement buffer >= 2periods...

perexg, 2021-05-25:
It's the dmix design thing.

closed 2021-05-25 by perexg

pietercardoen, 2021-05-25:
@aditpape: Thanks for the help. I will try this option!