alsa-project/alsa-lib #112: Audio over HDMI via USB-C Dock broke recently - module-alsa-card.c: Failed to find a working profile

asgrim, 2021-01-14:
Hi folks, not sure if this is related to #104 since I get the same error, but slightly different characteristics. Running Ubuntu 20.04.1 here.

I've got an XPS 17 9700, which is plugged into a Dell WD19TB dock via USB-C. Off that, I have a Dell 34" screen via HDMI, into which my speakers are plugged in. My headphones are plugged directly into the WD19TB dock. I do have a Yeti mic as well, which is plugged via USB into the dock also. I often switch between my speakers/headphones using a Gnome extension (I select either `Headphones (USB Audio)` or HDMI something (can't remember what it was called exactly).  However, I noticed recently (maybe last week or two?) I wasn't able to switch. Looked into it today, it's because the HDMI audio output seems completely missing. Does not show in regular Ubuntu sound settings, nor in `pavucontrol`. Checked /var/log/syslog, and found this `Failed to find a working profile` message.

Some Googling bought me here, hopefully you lovely folks can help!

* `pacmd ls` output is [here](https://gist.github.com/asgrim/1788fcae11cd1d0ced106c5d94199f29#file-pacmd)
* `pulseaudio -vvvvv` output is [here](https://gist.github.com/asgrim/1788fcae11cd1d0ced106c5d94199f29#file-pulse)

Hopefully this is just user error, but this did used to work, and I don't know where to ask for help as I know nothing about Linux audio stuff :D

Side note: I'm aware there's an issue with my XPS 17 "onboard" speakers not working, they've never worked; I believe there's a fix out there, but it looks complicated so I have so far avoided it. I'm ok with this as I've got the external speakers anyway.

perexg, 2021-01-14:
```
I: [pulseaudio] (alsa-lib)pcm_hw.c: open '/dev/snd/pcmC0D0p' failed (-2)
I: [pulseaudio] alsa-util.c: Error opening PCM device hw:sofhdadsp: No such file or directory
```

Could you also share a link to the output from alsa-info.sh ?

kv2019i, 2021-01-14:
@asgrim alsa-info.s output would be needed. Most likely explanantion is that some update to your system has broken audio support for the XPS17 audio chip, which also handles HDMI/DP audio. This is probably either kernel driver bug or a bug in Ubuntu.

asgrim, 2021-01-14:
Ah sorry, I think this is the info you need? http://alsa-project.org/db/?f=401784c355bd8a2158faefdd21311ab6960664dc

perexg, 2021-01-14:
It's weird. There are no capture devices for card 0 (sofhdadsp):

```
ARECORD

**** List of CAPTURE Hardware Devices ****
card 1: Dock [WD19 Dock], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Microphone [Yeti Stereo Microphone], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

!!Amixer output
```

It looks like wrong permissions or so. What's output of `ls -la /dev/snd/`?

asgrim, 2021-01-14:
```
$ ls -la /dev/snd
total 0
drwxr-xr-x   4 root root      360 Jan 14 08:19 .
drwxr-xr-x  24 root root     6020 Jan 14 08:19 ..
drwxr-xr-x   2 root root       80 Jan 14 08:19 by-id
drwxr-xr-x   2 root root      100 Jan 14 08:19 by-path
crw-rw----+  1 root audio 116,  6 Jan 14 08:19 controlC0
crw-rw----+  1 root audio 116, 10 Jan 14 08:19 controlC1
crw-rw----+  1 root audio 116, 13 Jan 14 08:19 controlC2
crw-rw----+  1 root audio 116,  5 Jan 14 08:19 hwC0D2
crw-rw----+  1 root audio 116,  2 Jan 14 08:19 pcmC0D1p
crw-rw----+  1 root audio 116,  3 Jan 14 08:19 pcmC0D2p
crw-rw----+  1 root audio 116,  4 Jan 14 08:27 pcmC0D3p
crw-rw----+  1 root audio 116,  8 Jan 14 08:34 pcmC1D0c
crw-rw----+  1 root audio 116,  7 Jan 14 10:11 pcmC1D0p
crw-rw----+  1 root audio 116,  9 Jan 14 08:34 pcmC1D1p
crw-rw----+  1 root audio 116, 12 Jan 14 08:59 pcmC2D0c
crw-rw----+  1 root audio 116, 11 Jan 14 08:34 pcmC2D0p
crw-rw----+  1 root audio 116,  1 Jan 14 08:19 seq
crw-rw----+  1 root audio 116, 33 Jan 14 08:19 timer
```

asgrim, 2021-01-14:
And the subdirectories there too, if it's any help:

```
$ ls -la /dev/snd/by-id/
total 0
drwxr-xr-x 2 root root  80 Jan 14 08:19 .
drwxr-xr-x 4 root root 360 Jan 14 08:19 ..
lrwxrwxrwx 1 root root  12 Jan 14 08:19 usb-Blue_Microphones_Yeti_Stereo_Microphone_REV8-00 -> ../controlC2
lrwxrwxrwx 1 root root  12 Jan 14 08:19 usb-Generic_USB_Audio_200901010001-00 -> ../controlC1
$ ls -la /dev/snd/by-path/
total 0
drwxr-xr-x 2 root root 100 Jan 14 08:19 .
drwxr-xr-x 4 root root 360 Jan 14 08:19 ..
lrwxrwxrwx 1 root root  12 Jan 14 08:19 pci-0000:00:1f.3-platform-skl_hda_dsp_generic -> ../controlC0
lrwxrwxrwx 1 root root  12 Jan 14 08:19 pci-0000:a8:00.0-usb-0:2.3.3.2:1.0 -> ../controlC2
lrwxrwxrwx 1 root root  12 Jan 14 08:19 pci-0000:a8:00.0-usb-0:2.3.4:1.0 -> ../controlC1
```

perexg, 2021-01-14:
Please, link to full `dmesg` and output from `ls -la /proc/asound/card0`.

asgrim, 2021-01-14:
Sure - [dmesg](https://gist.github.com/asgrim/0445642fc3a4994c1cda514ae1c04413#file-dmesg), and [/proc/asound/card0 contents](https://gist.github.com/asgrim/0445642fc3a4994c1cda514ae1c04413#file-proc-asound-card0) :)

kv2019i, 2021-01-15:
It seems this is the root of the problem:
[    3.430229] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 4
[    3.430230] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.430233] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 0

... only the HDMI/DP codec is found. So the integrated audio chip is only used for HDMI. This confuses Pulseaudio as we have no UCM config for this kind of system and it doesn't fir the generic HDA profile.

It's hard to imagine a laptop shipped without any audio codec. In theory you could disable the codec in BIOS, but I don't think Dell BIOS let's you do this. 

So I wonder if this actually a product that uses Soundwire audio codecs and the Ubuntu kernel does not yet have soundwire support (a very new thing with support just merged upstream).

@plbossart ping, have you seen this. Dell XPS17 9700 SKU 098F ?

perexg, 2021-01-15:
It's looks like a driver issue. The workaround is to use this settings which will force to use the legacy HDA driver for HDMI:

```
echo "options snd-intel-dspcfg dsp_driver=1" > /etc/modprobe.d/alsa.conf
```

plbossart, 2021-01-15:
https://elixir.bootlin.com/linux/latest/source/sound/soc/intel/boards/sof_sdw.c#L83

This is a SoundWire-based device, so likely the OEM image was not installed or was replaced by a different distro which doesn't include kernel support (and likely UCM) for SoundWire.

asgrim, 2021-01-15:
Adding `options snd-intel-dspcfg dsp_driver=1` into `/etc/modprobe.d/alsa.conf` worked. I guess an Ubuntu update broke something somewhere?

perexg, 2021-01-15:
The new SOF SoundWire driver behaves differently. The dspcfg option just forces to use the legacy (previous) HDA driver. Closing, it's not an alsa-lib issue and we have a workaround until the ucm / driver is stable for this machine.

closed 2021-01-15 by perexg

asgrim, 2021-01-15:
Thank you for your time folks, much appreciated! :+1: