alsa-project/alsa-lib #123: Fiio M3K: Failed to find a working profile

x1unix, 2021-03-03:
Hello, I have a [Fiio M3K](https://fiio.com/m3k) player which I often use as an external USB DAC.

It worked well with Pipewire, but I had some other issues with Pipewire so I had to rollback to PulseAudio.
Unfortunately, PulseAudio and module-alsa-card cannot setup this device, so it doesn't work.

Here is a small portion of PulseAudio log (full log is available at link below):

```
E: [pulseaudio] module-alsa-card.c: Failed to find a working profile.
E: [pulseaudio] module.c: Failed to load module "module-alsa-card" (argument: "device_id="2" name="usb-GuangZhou_FiiO_Electronics_Technology_Co.__Ltd._FiiO_M3K_FiiO_M3K-00" card_name="alsa_card.usb-GuangZhou_FiiO_Electronics_Technology_Co.__Ltd._FiiO_M3K_FiiO_M3K-00" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed.
I: [pulseaudio] module-udev-detect.c: Card /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb5/5-1/5-1:1.0/sound/card2 (alsa_card.usb-GuangZhou_FiiO_Electronics_Technology_Co.__Ltd._FiiO_M3K_FiiO_M3K-00) failed to load module.
D: [pulseaudio] module-udev-detect.c: /dev/snd/controlC2 is accessible: yes
D: [pulseaudio] module-udev-detect.c: /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb5/5-1/5-1:1.0/sound/card2 is busy: no
```

I'm using ArchLinux and Linux 5.11.2.

**Logs and debug info**

aplay, pulseaudio debug log and other info I've posted as gist:
https://gist.github.com/x1unix/dae3fb810ac499cc63362a805220968f

x1unix, 2021-05-25:
@perexg found some interesting effect. Device is detected and configured successfully if it's connected during boot time.

perexg, 2021-05-26:
Then it appears like a device initialization issue. It's not easy to find the culprit then (udev / pulseaudio handshake).