alsa-project/alsa-lib #127: Very unstable and broken after audio device switching, reboots, multimedia app usage etc etc etc

oliiix, 2021-03-19:
To me this seems to be a huge issue for years now. Running Ubuntu 20.10 very latest.
I'm using an Astro A50 and two HDMI Displays for playing audio plus there's two more outputs I'm not really using. But as soon as the computer ran a little long (yes, that's actually enough to cause problems, probably due to certain apps like Unreal engine and games!), came out of sleep or I switch playback devices (which I'm doing very often and that shouldnt be ANY issue), things get seriously mixed up. Suddenly the routing goes all wrong (i select one output device, another plays instead, or just no sound at all). Sometimes "no sound" problem gets fixed by just switching devices about 10-20! times until sound is back (it either works or doesn't but that it starts working again after ten times switching devices is very weird behavior). Reinstalling as well as resetting alsa doesn't do anything to remedy no sound, restarting computer doesn't fix sound problems, only switching devices ten times does! Alsa Mixer doesn't show any controls for my A50, yet it often works even if some profiles are to low volume and the one that has good volume there I can't hear audio that comes from behind me (I have the no sound issue also on my HDMI outputs, just saying). Pavucontrol on the other hand lists my A50 but doesn't show one of my HDMI displays. I mean what is going wrong here (for years)??? I mean I found forum posts about these  issues that reach back years in time -.- When will we have proper working audio where audio device switching on linux is possible without any problems and the sound doesn't just "give up" after some time?

perexg, 2021-04-13:
If you have an issue, try to describe it seriously to get it fixed. One by one.

closed 2021-04-13 by perexg

oliiix, 2021-04-14:
i'd say i pretty much described it all... and ye, it's not about a simple single error message that needs fixing, you see that correctly. the routing is problematic, the routing including those "profiles" (ubuntu calls it "configuration") get mixed up, suddenly profiles are even attached to devices they don't belong to and then it comes to the no sound at all anymore issue, just USE IT YOURSELF. It's really not hard to repeat this!

perexg, 2021-04-14:
The profiles are managed by the pulseaudio or pipewire. We cannot do much here. You can do tests with the direct ALSA devices, then we can work on the fix.