alsa-project/alsa-lib #134: pulse.conf overrides ~/.asoundrc for default

flatmax, 2021-05-13:
On many systems pulse audio sets the default device to pulse. This overrides the settings in ~/.asoundrc.

This is further complicated by using a tool like pasuspender and finding that pulse's default device is still set and overriding ~/.asoundrc

It would seem more empowering to let the user override the default device in ~/.asoundrc and have that take effect over pulse.conf which resides in /usr/share/alsa/alsa.conf.d

perexg, 2021-05-13:
It's already possible.

````
# Override default PCM device back to the native ALSA device

pcm.!default cards.pcm.default
````

flatmax, 2021-05-13:
As an example, when I set default in my ~/.asoundrc like so :
```
pcm.!default {
  type hw
  card "HARDWARE"
}
```
the pulse.conf default still takes over. In otherwords, aplay file.wav => pcm.pulse rather them my pcm.default from .asoundrc.

The only way I have found to override this is to put the following in my .asoundrc :
```
@hooks [
	{
		func load
		files [
			"~/.asoundrc"
		]
		errors false
	}
]
```

perexg, 2021-05-13:
Something is bad in your system:

````
$ aplay -L
default
    Default ALSA Output (currently PulseAudio Sound Server)
````

Modified `~/.asoundrc`:

````
pcm.!default cards.pcm.default
````

Result:

````
$ aplay -L
default
    Default Audio Device
````

With the recent `alsactl` (not released yet):

````
$ alsactl dump-cfg | grep -E "^pcm.default"
pcm.default 'cards.pcm.default'
````

flatmax, 2021-05-13:
I am a little lost !

Lets say my setup has pulse as the last loaded default from /etc/alsa/conf.d/99-pulseaudio-default.conf.example This runs through a HDMI connector to the monitor with speakers as it is card 0.

I have a second USB soundcard which comes up as card 1. How do I setup my asoundrc without having to use the @hooks to force audio through the USB card 1?

Setting the following makes the audio still come through card 0 :
`pcm.!default cards.pcm.default`

perexg, 2021-05-14:
`cards.pcm.default` redirects to the generic device definition which has card 0 as default (defined at `defaults.pcm.card 0` line in `/usr/share/alsa/alsa.conf`)

Possibilities:

`pcm.!default "hw:1"` or `pcm.!default "plughw:1"`
```
defaults.pcm.card 1
pcm.!default cards.pcm.default
```
```
pcm.!default {
  type hw
  card 1
}
```

The card may be identified by the number or ASCII identifier (see /proc/asound/cards [] brackets).

perexg, 2021-11-05:
Answered.

closed 2021-11-05 by perexg