alsa-project/alsa-lib #139: No sound with Realtek ALC3306

dantahoua, 2021-05-27:
Realtek ALC3306 (Lenovo 13s Gen2 ITL i7), but seems common problem with any computer using this codec.
-Headphone, HDMI, BT working great.
-Speaker not working (but computer seeing audio output and sending audio, meter even showing audio in GUI).
Seems like something (amp?) is not turned, like missing something between speaker and codec...
Works perfect in Windows 10.
Tested lots different distro and kernel...

pablitar, 2021-05-29:
Same issue here. Everything working except for the speakers.

My computer is a Lenovo Legion 7 16 inches, model 16ACHg6.

Here is my info: 
[lenovo-16ACHg6-alsainfo.txt](https://github.com/alsa-project/alsa-lib/files/6565353/lenovo-16ACHg6-alsainfo.txt)

As happened to @dantahoua, it reports ALC287, when the specs of the model say it is ALC3306.

Hope this helps!

henryhormaza, 2021-05-30:
I have the same issue running fedora 34 (5.12.7-300.fc34.x86_64).

Manufacturer:      LENOVO
Product Name:      82N6
Product Version:   Legion 7 16ACHg6
Firmware Version:  GKCN36WW
System SKU:        LENOVO_MT_82N6_BU_idea_FM_Legion 7 16ACHg6
Board Vendor:      LENOVO
Board Name:        LNVNB161216

[alsa-info.txt](https://github.com/alsa-project/alsa-lib/files/6566375/alsa-info.txt)

perexg, 2021-05-30:
It seems like a kernel issue. The bug tracker for the kernel is at https://bugzilla.kernel.org .

closed 2021-05-30 by perexg