alsa-project/alsa-lib #140: 1.2.5?

kloczek, 2021-05-29:
Looks like 1.2.5 has been released but I dond see related changes and git tags not only here but in other alsa git repos as well.

perexg, 2021-05-29:
Sorry, the reltest branch is release testing. I've not announced the release nor tags until my tests are finished.

kloczek, 2021-05-29:
Funny .. in Fedoa are alredy all alsa* 1.2.5 packages and that cunfuset me a bit :)
I'v been only thinking that you forgot to push your local repo changes to github :)

Closing

closed 2021-05-29 by kloczek

perexg, 2021-05-30:
Yes, Fedora rawhide updates are part of my tests.

kloczek, 2021-05-30:
So why would you not version that like for example 1.2.4.99?

perexg, 2021-05-30:
It's very unlikely that the packages will change when they're built. I don't think that this extra step is necessary.