alsa-project/alsa-lib #142: alsa-lib 1.2.5 no devices on Lenovo Carbon (gen 7)

sambo57u, 2021-05-31:
I use Fedora 34 and updated alsa-lib and all other alsa products to version 1.2.5. They work fine on my desktop machines but on Lenovo laptop I am getting the message alsactl cannot find hw:0 and daemon deactivates with no devices showing. Downgrading
to 1.2.4 solves the problem.

perexg, 2021-05-31:
Could you show me the error lines ?

sambo57u, 2021-05-31:
alsactl[2087]: alsactl 1.2.5 daemon started
systemd[1]: Reached target Sound Card.
alsactl[2087]: /usr/sbin/alsactl: snd_card_iterator_sinit:252: Cannot find soundcard 'hw:0'...
alsactl[2087]: alsactl daemon stopped

foutrelis, 2021-06-01:
I think I'm seeing a similar issue on Arch Linux. After upgrading to alsa-lib 1.2.5 Chromium stopped playing audio until pulseaudio was restarted. Upon closer inspection, `pactl list cards` is now missing the `HDA Intel HDMI` card and only shows the `HDA Intel PCH` one.

Some error messages I've gathered:

- After upgrading to alsa-lib and before restarting pulseaudio, attempting to play a video in Chromium would result in:
  `pulseaudio[625]: Error opening PCM device front:0: Invalid argument`

- Restarting pulseaudio with alsa-lib 1.2.5 installed logs the following to the journal:
  ```
  pulseaudio[66345]: Failed to find a working profile.
  pulseaudio[66345]: Failed to load module "module-alsa-card" (argument: "device_id="1" name="pci-0000_00_03.0"
  card_name="alsa_card.pci-0000_00_03.0" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no
  deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""):
  initialization failed.
  ```

Git bisect points to commit 63f7745be504e447923f2cde421177a2fca99340 as the first commit that results in the `Failed to find a working profile` error.

After restarting pulseaudio with alsa-lib 1.2.5 installed, it appears that `pactl list cards` shows one fewer cards on my laptop which has this audio hardware:

```
$ lspci | grep Audio
00:03.0 Audio device: Intel Corporation Haswell-ULT HD Audio Controller (rev 0b)
00:1b.0 Audio device: Intel Corporation 8 Series HD Audio Controller (rev 04)
```

[alsa-lib-1.2.4-pactl-list-cards.txt](https://github.com/alsa-project/alsa-lib/files/6575475/alsa-lib-1.2.4-pactl-list-cards.txt)
[alsa-lib-1.2.5-pactl-list-cards.txt](https://github.com/alsa-project/alsa-lib/files/6575476/alsa-lib-1.2.5-pactl-list-cards.txt)

aminvakil, 2021-06-01:
I'm having the exact same issue with @foutrelis, I only faced this issue using [strawberry](https://github.com/strawberrymusicplayer/strawberry/).

joakim-tjernlund, 2021-06-01:
Same problem on Gentoo

Polynomial-C, 2021-06-01:
> Same problem on Gentoo

https://bugs.gentoo.org/793410

joakim-tjernlund, 2021-06-01:
killing PA and restarting it may fix the problem, in case that makes it easier to diagnose.

perexg, 2021-06-01:
Note that those are probably two bugs - PA sound device detection and alsactl `hw:#` problem. The alsactl problem should be fixed with the above commit.

foutrelis, 2021-06-01:
@perexg Would you say the sound device detection is a PA bug? Or should I open a new bug in alsa-lib?

perexg, 2021-06-01:
```
lenovo-x1.localdomain pipewire[4675]: '_ucm0001.hw:sofhdadsp,5': playback open failed: No such device
lenovo-x1.localdomain pipewire[4675]: adapter 0x5617b566a460: can't get format: No such device
```

The new /var/lib/alsa/card0 directory does not exists with the ucm config (it should be created also in alsactl). Digging more.. It's problem for X1 Carbon, the HDA HDMI issue seems different. You should check the full pulseaudio / pipewire logs. The short version is not very helpful.

foutrelis, 2021-06-01:
Any chance this `pulseaudio -v` output is more useful? 🐈

[pulseaudio-verbose.txt](https://github.com/alsa-project/alsa-lib/files/6577650/pulseaudio-verbose.txt)

perexg, 2021-06-01:
Yes:

```
Parse arguments error: Invalid argument
I: [pulseaudio] (alsa-lib)confmisc.c: Unable to find definition 'cards.HDA-Intel.pcm.hdmi.common:CARD=1,DEVICE=3,CTLINDEX=0,AES0=4,AES1=130,AES2=0,AES3=2cards.HDA-Intel.pcm.hdmi.common:CARD=1,DEVICE=3,CTLINDEX=0,AES0=4,AES1=130,AES2=0,AES3=2'
```

foutrelis, 2021-06-01:
So it's complaining about `AES3=2cards` because `Parameter AES3 must be an integer`. Have you figured out how it relates to commit 63f7745be504e447923f2cde421177a2fca99340?

perexg, 2021-06-01:
The config reference string is wrongly doubled. It should not happen. Trying to reproduce here.

foutrelis, 2021-06-01:
> The config reference string is wrongly doubled.

Ouch, it's indeed doubled! I opened a new issue to better track this.

perexg, 2021-06-01:
The original issue is for pipewire. It seems that this daemon opens the devices outside the UCM manager, thus the private alsa-lib configuration (which is available only in memory), is not available for the forked process.

Trying to discuss this in the pipewire tracker:

https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/1251

perexg, 2021-06-02:
Workaround:

````
# diff -u /usr/share/alsa/ucm2/ucm.conf~ /usr/share/alsa/ucm2/ucm.conf
--- /usr/share/alsa/ucm2/ucm.conf~	2021-05-27 10:14:01.000000000 +0200
+++ /usr/share/alsa/ucm2/ucm.conf	2021-06-02 16:39:34.093331137 +0200
@@ -18,7 +18,7 @@
 Define.V2Module yes	# non-empty string to enable module name lookups (obsolete)
 Define.V2Name yes	# non-empty string to enable driver & card name lookups (obsolete)
 
-Include.libgen.File "/lib/generic.conf"  # private alsa-lib configuration
+#Include.libgen.File "/lib/generic.conf"  # private alsa-lib configuration
 
 If.driver {
 	Condition {
````

sambo57u, 2021-06-02:
I patched the rpms with all the recent changes including the one above and it works now with pipewire. I see that you are also
building the rpms on koji.

dvzrv, 2021-06-04:
@perexg Would it be possible to release bug fix version with the current fixes or is pipewire integration not fixed by their release of 0.3.29 and applying https://github.com/alsa-project/alsa-lib/commit/0e4ba2ea8c0402f12a645032a14693eb9b1278e6 and https://github.com/alsa-project/alsa-lib/commit/9621d0bff2e60b43e329ffa5059ab19f2914ec14?

From a packager perspective it is a bit unclear at this point, which patches to apply for this specific issue.

perexg, 2021-06-04:
@dvzrv : Sorry about that. Yes, I plan to release 1.2.5.1 soon, but all changes in alsa-lib and alsa-utils repos are bug fixes right now.

dvzrv, 2021-06-04:
I understand.

It seems the first three commits since 1.2.5 in combination with https://github.com/alsa-project/alsa-utils/commit/78212445de4c8e07873cbc7dff2abcacd031f151 and pipewire 0.3.29 (or pulseaudio 14.2) fixed it for the affected users on Arch Linux: https://bugs.archlinux.org/task/71077

tramtrist, 2021-06-14:
My Lenovo X13 AMD lost Microphone with 1.2.5.
1.2.4 working just fine.

perexg, 2021-06-14:
ALSA bugfix packages 1.2.5.1 were released. I believe that the original issue was resolved. Closing.

closed 2021-06-14 by perexg

miquecg, 2021-06-15:
@tramtrist You can join this [Arch Linux forum thread](https://bbs.archlinux.org/viewtopic.php?id=266981). Main issue is solved but other problems are still present.