alsa-project/alsa-lib #143: Incomplete sound device detection in PulseAudio with alsa-lib 1.2.5

foutrelis, 2021-06-01:
(Copying my comments over from #142 since it appears to be a different issue to that one. Sorry for the noise there!)

After upgrading to alsa-lib 1.2.5 Chromium stopped playing audio until pulseaudio was restarted. Upon closer inspection, `pactl list cards` is now missing the `HDA Intel HDMI` card and only shows the `HDA Intel PCH` one.

Some error messages I've gathered:

- After upgrading to alsa-lib and before restarting pulseaudio, attempting to play a video in Chromium would result in:
  `pulseaudio[625]: Error opening PCM device front:0: Invalid argument`

- Restarting pulseaudio with alsa-lib 1.2.5 installed logs the following to the journal:
  ```
  pulseaudio[66345]: Failed to find a working profile.
  pulseaudio[66345]: Failed to load module "module-alsa-card" (argument: "device_id="1" name="pci-0000_00_03.0"
  card_name="alsa_card.pci-0000_00_03.0" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no
  deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""):
  initialization failed.
  ```

Git bisect points to commit 63f7745be504e447923f2cde421177a2fca99340 as the first commit that results in the `Failed to find a working profile` error.

After restarting pulseaudio with alsa-lib 1.2.5 installed, it appears that `pactl list cards` shows one fewer cards on my laptop which has this audio hardware:

```
$ lspci | grep Audio
00:03.0 Audio device: Intel Corporation Haswell-ULT HD Audio Controller (rev 0b)
00:1b.0 Audio device: Intel Corporation 8 Series HD Audio Controller (rev 04)
```

[alsa-lib-1.2.4-pactl-list-cards.txt](https://github.com/alsa-project/alsa-lib/files/6575475/alsa-lib-1.2.4-pactl-list-cards.txt)
[alsa-lib-1.2.5-pactl-list-cards.txt](https://github.com/alsa-project/alsa-lib/files/6575476/alsa-lib-1.2.5-pactl-list-cards.txt)

`pulseaudio -v` output:

[pulseaudio-verbose.txt](https://github.com/alsa-project/alsa-lib/files/6577650/pulseaudio-verbose.txt)

perexg, 2021-06-01:
I found the issue. Here's a quick fix (this config file is included multiple times, so it's required to clean the updated tree to avoid merging of arrays):

```
# diff -u /usr/share/alsa/cards/HDA-Intel.conf~ /usr/share/alsa/cards/HDA-Intel.conf
--- /usr/share/alsa/cards/HDA-Intel.conf~	2021-05-31 12:29:51.000000000 +0200
+++ /usr/share/alsa/cards/HDA-Intel.conf	2021-06-01 21:01:13.347658405 +0200
@@ -4,6 +4,8 @@
 
 <confdir:pcm/front.conf>
 
+!HDA-Intel {}
+
 HDA-Intel.pcm.front.0 {
 	@args [ CARD ]
 	@args.CARD {
```

I am evaluating other solutions, too.

perexg, 2021-06-02:
Fixed in the above commit without the requirement to modify the driver specific configuration files.

closed 2021-06-02 by perexg

foutrelis, 2021-06-02:
Hi @perexg, thank you for looking into this issue!

> `pactl list cards` is now missing the `HDA Intel HDMI` card and only shows the `HDA Intel PCH` one.

This is now fixed.

> After upgrading to alsa-lib and before restarting pulseaudio, attempting to play a video in Chromium would result in:
>   `pulseaudio[625]: Error opening PCM device front:0: Invalid argument`

This is still happening, preventing a smooth upgrade from 1.2.4 to 1.2.5 without having to restart pulseaudio.

perexg, 2021-06-02:
> > After upgrading to alsa-lib and before restarting pulseaudio, attempting to play a video in Chromium would result in:
> > `pulseaudio[625]: Error opening PCM device front:0: Invalid argument`
> 
> This is still happening, preventing a smooth upgrade from 1.2.4 to 1.2.5 without having to restart pulseaudio.

Unfortunately, the config is not backward compatible, so the older alsa-lib (which pulseaudio uses until restarted) cannot handle it.

foutrelis, 2021-06-02:
> Unfortunately, the config is not backward compatible, so the older alsa-lib (which pulseaudio uses until restarted) cannot handle it.

Ah, that's all right. We'll add a post installation note about it in Arch.

FWIW, I'm attaching `pulseaudio -v` output after upgrading alsa-lib to 1.2.5 (with this patch) without restarting PA and attempting to play an audio file in mpv. These lines stood out to me:

```
I: [alsa-sink-ALC3227 Analog] (alsa-lib)dlmisc.c: unable to verify version for symbol snd_func_private_integer
I: [alsa-sink-ALC3227 Analog] (alsa-lib)conf.c: symbol snd_func_private_integer is not defined inside (null)
```

Based on what you said, it does appear that the older libasound loaded by pulseaudio can't find the newly-added `snd_func_private_integer` symbol within itself (which is referenced in the config as `private_integer`).

I appreciate the explanation and quick fix of the device detection issue. Thanks again! 🦔

[alsa-lib-upgrade-without-pulseaudio-restart.txt](https://github.com/alsa-project/alsa-lib/files/6582917/alsa-lib-upgrade-without-pulseaudio-restart.txt)

iiv3, 2021-06-03:
Would you please release alsa-lib-1.2.5.1 containing this fix?

Without this fix even native alsa programs fail to work...