alsa-project/alsa-lib #146: amixer and alsamixer unaware of controls for ThinkPad T14's 'acp' card

jwatt, 2021-06-05:
I have the AMD Ryzen 7 Pro 4750U version of the [ThinkPad T14 gen 1 type 20UD](https://pcsupport.lenovo.com/gb/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t14-type-20ud-20ue/20ud/?linkTrack=Homepage%3ABody_Search%20Products&searchType=4&keyWordSearch=T14%20%20Gen%201%20%28type%2020UD,%2020UE%29%20Laptop%20%28ThinkPad%29%20-%20Type%2020UD) ([tech specs](https://www.lenovo.com/gb/en/laptops/thinkpad/t-series/ThinkPad-T14-AMD-G1/p/22TPT14T4A2?notPassDisplayRuleRedirectFlag=notPassDisplayRuleRedirectFlag)) with Fedora 34 installed. I am unable to control the built-in microphone using `amixer`/`alsamixer`.

```
$ cat /proc/asound/modules
   0 snd_hda_intel
   1 snd_hda_intel
   2 snd_acp3x_rn
```

As best I can tell (I could be mistaken!), 0 is the HDMI port, 1 is the build-in speakers and headphones jack, and 2 is the Ryzen "Audio Co-Processor" (i.e. "acp" / "AMD TrueAudio") which is (or includes) the built-in microphone. To be specific, I only get microphone input if I select 'acp' as the input.

For the 'acp' card, AlsaMixer v1.2.4 reports "This sound device does not have any controls."

It appears support for the Ryzen 4000 mobile series Audio Co-Processor was added in kernel 5.8 (see the section "Add Renoir ACP driver" on the [5.8 release notes](https://kernelnewbies.org/Linux_5.8#Audio)). If the kernel supports it, it's unclear to me why the ALSA tools wouldn't allow me to control it. Is this currently expected?

I'm happy to provide any other information that would be helpful - just let me know what you need.

jwatt, 2021-06-05:
I should also add: although I can select 'acp' as the input, even after boosting the gain via PulseAudio audio from the built-in microphone is extremely quiet. On Windows 10 the input volume is fine.

perexg, 2021-06-07:
The AMD engineers says that there are no hardware volume controls for the digital microphones connected to the AMD ACP bridge. It's definitely not an alsa-lib bug. If you have questions, the driver authors contacts are in the source: https://github.com/torvalds/linux/commit/1eb2852efe05abfa94cd78cc9865389643726ee9

closed 2021-06-07 by perexg

jwatt, 2021-06-07:
Oh, surprising. Thank you for checking with the AMD engineers and replying, @perexg!