alsa-project/alsa-lib #148: alsa-lib 1.2.5(-2 on arch) only one of two connected hdmi/displayport outputs visible at once. My mic is completely MIA

fkroener, 2021-06-07:
Since the recent update to 1.2.5-2 on archlinux, only one of my two usb-c/displayport/hdmi devices show up in pavucontrol and kde mixer. I am able to switch between hdmi0 and hdmi1, by changing the Profile in pavucontrol of HD-Audio Generic. 

The microphone doesn't show up at all now.

[image]

This doesn't appear related to #142 or #143, since I have revision 2 from arch installed, which should include the fixes.

This is a Lenovo T14 AMD.

```
pactl list cards 
Card #0
        Name: alsa_card.pci-0000_07_00.1
        Driver: module-alsa-card.c
        Owner Module: 6
        Properties:
                alsa.card = "0"
                alsa.card_name = "HD-Audio Generic"
                alsa.long_card_name = "HD-Audio Generic at 0xfd3c8000 irq 91"
                alsa.driver_name = "snd_hda_intel"
                device.bus_path = "pci-0000:07:00.1"
                sysfs.path = "/devices/pci0000:00/0000:00:08.1/0000:07:00.1/sound/card0"
                device.bus = "pci"
                device.vendor.id = "1002"
                device.vendor.name = "Advanced Micro Devices, Inc. [AMD/ATI]"
                device.product.id = "1637"
                device.string = "0"
                device.description = "HD-Audio Generic"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card-pci"
        Profiles:
                output:hdmi-stereo: Digital Stereo (HDMI) Output (sinks: 1, sources: 0, priority: 5900, available: yes)
                output:hdmi-stereo-extra1: Digital Stereo (HDMI 2) Output (sinks: 1, sources: 0, priority: 5700, available: yes)
                output:hdmi-stereo-extra2: Digital Stereo (HDMI 3) Output (sinks: 1, sources: 0, priority: 5700, available: no)
                output:hdmi-surround-extra2: Digital Surround 5.1 (HDMI 3) Output (sinks: 1, sources: 0, priority: 600, available: no)
                output:hdmi-surround71-extra2: Digital Surround 7.1 (HDMI 3) Output (sinks: 1, sources: 0, priority: 600, available: no)
                off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
        Active Profile: output:hdmi-stereo
        Ports:
                hdmi-output-0: HDMI / DisplayPort (type: HDMI, priority: 5900, latency offset: 0 usec, available)
                        Properties:
                                device.icon_name = "video-display"
                                device.product.name = "P27h-20"
                        Part of profile(s): output:hdmi-stereo
                hdmi-output-1: HDMI / DisplayPort 2 (type: HDMI, priority: 5800, latency offset: 0 usec, available)
                        Properties:
                                device.icon_name = "video-display"
                                device.product.name = "P27h-20"
                        Part of profile(s): output:hdmi-stereo-extra1
                hdmi-output-2: HDMI / DisplayPort 3 (type: HDMI, priority: 5700, latency offset: 0 usec, not available)
                        Properties:
                                device.icon_name = "video-display"
                        Part of profile(s): output:hdmi-stereo-extra2, output:hdmi-surround-extra2, output:hdmi-surround71-extra2

Card #1
        Name: alsa_card.pci-0000_07_00.6
        Driver: module-alsa-card.c
        Owner Module: 8
        Properties:
                alsa.card = "1"
                alsa.card_name = "HD-Audio Generic"
                alsa.long_card_name = "HD-Audio Generic at 0xfd3c0000 irq 92"
                alsa.driver_name = "snd_hda_intel"
                device.bus_path = "pci-0000:07:00.6"
                sysfs.path = "/devices/pci0000:00/0000:00:08.1/0000:07:00.6/sound/card1"
                device.bus = "pci"
                device.vendor.id = "1022"
                device.vendor.name = "Advanced Micro Devices, Inc. [AMD]"
                device.product.id = "15e3"
                device.product.name = "Family 17h (Models 10h-1fh) HD Audio Controller"
                device.string = "1"
                device.description = "Family 17h (Models 10h-1fh) HD Audio Controller"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card-pci"
        Profiles:
                input:analog-stereo: Analog Stereo Input (sinks: 0, sources: 1, priority: 65, available: no)
                output:analog-stereo: Analog Stereo Output (sinks: 1, sources: 0, priority: 6500, available: yes)
                output:analog-stereo+input:analog-stereo: Analog Stereo Duplex (sinks: 1, sources: 1, priority: 6565, available: no)
                off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
        Active Profile: output:analog-stereo
        Ports:
                analog-input-mic: Microphone (type: Mic, priority: 8700, latency offset: 0 usec, not available)
                        Properties:
                                device.icon_name = "audio-input-microphone"
                        Part of profile(s): input:analog-stereo, output:analog-stereo+input:analog-stereo
                analog-output-speaker: Speakers (type: Speaker, priority: 10000, latency offset: 0 usec, availability unknown)
                        Properties:
                                device.icon_name = "audio-speakers"
                        Part of profile(s): output:analog-stereo, output:analog-stereo+input:analog-stereo
                analog-output-headphones: Headphones (type: Headphones, priority: 9900, latency offset: 0 usec, not available)
                        Properties:
                                device.icon_name = "audio-headphones"
                        Part of profile(s): output:analog-stereo, output:analog-stereo+input:analog-stereo
```

If you need any more information, I'm gladly at your disposal. Thanks!

perexg, 2021-06-07:
Could you select "Analog Stereo Duplex" profile?

For HDMI, the profiles are different, HDMI 3 is not available. No idea what you expect here. You may post the previous output.

It would be nice to consult this with PA developers.

perexg, 2021-06-07:
It seems that UCM is not used. I believe that this fix is missing in the ucm config files: https://github.com/alsa-project/alsa-ucm-conf/commit/3f34021beffba4e39f064a14c5faceeaa224b766

closed 2021-06-07 by perexg

fkroener, 2021-06-07:
You are right! Thanks and sorry for the noise.

dvzrv, 2021-06-10:
@perexg I have just applied https://github.com/alsa-project/alsa-ucm-conf/commit/3f34021beffba4e39f064a14c5faceeaa224b766 to our alsa-ucm-conf package on Arch Linux in the hopes that it will fix the issue we are seeing there: https://bugs.archlinux.org/task/71180

It would be great to leave a ticket such as this open until there is release out that fixes it. Given the amount of patches that have to be applied on top of 1.2.5 already it is otherwise hard to track what can be fixed how. Thanks! :)

sanketh1691, 2021-11-30:
Hello, 
can someone guide on how to install the alsa-utils, lib, ucm-conf etc on ARM architecture or aarch64. Like after downloading these, what to do to install these?