alsa-project/alsa-lib #149: opensuse tumbleweed no sound after alsa lib update (tumbleweed 2010605)

pawanyadav, 2021-06-09:
sound is not working alsa (never installed pipewire and pulseaudio)

asound.conf
defaults.pcm.card 1
defaults.pcm.device 1

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: HDMI [HDA ATI HDMI], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Generic [HD-Audio Generic], device 0: ALC887-VD Analog [ALC887-VD Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Generic [HD-Audio Generic], device 1: ALC887-VD Digital [ALC887-VD Digital]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

cat /proc/asound/cards
 0 [HDMI           ]: HDA-Intel - HDA ATI HDMI
                      HDA ATI HDMI at 0xfcf40000 irq 67
 1 [Generic        ]: HDA-Intel - HD-Audio Generic
                      HD-Audio Generic at 0xfce00000 irq 69

speaker-test 1.2.4

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
ALSA lib confmisc.c:855:(parse_card) cannot find card '1dmix:1'
ALSA lib conf.c:5091:(_snd_config_evaluate) function snd_func_card_id returned error: No such device
ALSA lib confmisc.c:422:(snd_func_concat) error evaluating strings
ALSA lib conf.c:5091:(_snd_config_evaluate) function snd_func_concat returned error: No such device
ALSA lib confmisc.c:1334:(snd_func_refer) error evaluating name
ALSA lib conf.c:5091:(_snd_config_evaluate) function snd_func_refer returned error: No such device
ALSA lib conf.c:5579:(snd_config_expand) Evaluate error: No such device
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM dmix:1dmix:1
Playback open error: -19,No such device

perexg, 2021-06-09:
It's already fixed in the current alsa-lib master git repo (commit ddfc32abf5697de1618b9e7ffdf57a0f97013090).

closed 2021-06-09 by perexg

JonMagon, 2021-06-14:
Fix merged to https://build.opensuse.org/package/show/openSUSE%3AFactory/alsa but it still doesn't work