alsa-project/alsa-lib #150: Current version not working with Focusrite Scarlett 18i20

TheBitFighter, 2021-06-09:
Hello, I hope I am in the right repository for this. 

My arch system updated alsa-card-profiles and alsa-utils today which caused my interface, a Focusrite Scarlett 18i20 1st gen to stop working. The card was still showing up in pulse and showed output, however no sound was actually played. Input was still working fine. It sounded like the output stage of the interface had been disabled, and the problem could only be fixed by booting Windows and setting new routing settings in Scarlett MixControl. The interface had exhibited some resetting and popping behavior for a few versions now (when stopping playback, changing application and playing something else there), but continued to work (given a few instances where sample rate played mismatched sample rate sent which caused weird audio slowdowns). It only stopped working today.

What I did initially was reinstall the older versions of these two packages, which did not fix the problem. I now downgraded alsa-libs (1.2.5-2 -> 1.2.4-2), alsa-card-profiles (1:0.3.29-2 -> 1:0.3.28-1), lib32-alsa-lib (1.2.5-2 -> 1.2.4-1), alsa-topology-conf (1.2.5-1 -> 1.2.4-1), and alsa-ucm-conf (1.2.5-1 -> 1.2.4-1) which fixed the problem. (alsa-utils is at 1.2.5-2)
`cat /etc/asound/version` now reports `Advanced Linux Sound Architecture Driver Version k5.12.9-arch1-1.`
This also fixed the popping/resetting issues I had before.

 Thanks for the help! If you need any further information I am happy to help.

perexg, 2021-06-09:
You may try to compare the output from `amixer -c CARDNO contents` for the working / not working case. Replace CARDNO with the appropriate ALSA card number or text id.

TheBitFighter, 2021-06-09:
Sadly when I run it with the old version, I get this error:
```
ALSA lib conf.c:5200:(snd_config_expand) Unknown parameters 2
ALSA lib control.c:1471:(snd_ctl_open_noupdate) Invalid CTL sysdefault:2
amixer: Control sysdefault:2 open error: Invalid argument
```
It works with the current version and I can now confirm that it is indeed reproducible. The output for the current alsa version is attached below.
[after.txt](https://github.com/alsa-project/alsa-lib/files/6625794/after.txt)

perexg, 2021-06-09:
Use `amixer -D hw:CARDNO contents` for the old version.

dvzrv, 2021-06-09:
As background information, here are the build scripts (which reference the currently used patches on top of 1.2.5) for the packages on Arch Linux:

- https://github.com/archlinux/svntogit-packages/blob/packages/alsa-lib/trunk/PKGBUILD
- https://github.com/archlinux/svntogit-packages/blob/packages/alsa-utils/trunk/PKGBUILD
- https://github.com/archlinux/svntogit-packages/blob/packages/alsa-plugins/trunk/PKGBUILD
- https://github.com/archlinux/svntogit-packages/blob/packages/alsa-topology-conf/trunk/PKGBUILD
- https://github.com/archlinux/svntogit-packages/blob/packages/alsa-ucm-conf/trunk/PKGBUILD
- https://github.com/archlinux/svntogit-community/blob/packages/alsa-tools/trunk/PKGBUILD

TheBitFighter, 2021-06-10:
I the -D flag worked for the older alsa version. I have appended the file bellow. However diff shows no difference between the state before and after the update.
[before.txt](https://github.com/alsa-project/alsa-lib/files/6630557/before.txt)

TheBitFighter, 2021-06-10:
Also I wanted to add, the older version has not solved the popping and resetting issues.

perexg, 2021-06-10:
Ok, in this case, I would try the standard ALSA tools like `aplay` or `speaker-test` and check the behaviour for old / new alsa-lib. The PCM device state can be obtained via `/proc/asound/cardNO/pcm*p/*` interface. It may also show something.

TheBitFighter, 2021-06-10:
Once again, the files are identical, however `speaker-test` does not work after the update. I appended both results for good measure.
[amixer_after.txt](https://github.com/alsa-project/alsa-lib/files/6634536/amixer_after.txt)
[amixer_before.txt](https://github.com/alsa-project/alsa-lib/files/6634538/amixer_before.txt)

`/proc/asound/USB/pcm0p/sub0` reported this after the update:
```
format: S32_LE
subformat: STD
channels: 16
rate: 44100 (44100/1)
period_size: 8192
buffer_size: 16384
```
The only interesting thing I can spot is that the interface is supposed to run at 48kHz, which is what I set in MixControl.
`sw_params` just stated `closed` before, now changed to 
```
tstamp_mode: ENABLE
period_step: 1
avail_min: 16033
start_threshold: 18446744073709551615
stop_threshold: 4611686018427387904
silence_threshold: 0
silence_size: 0
boundary: 4611686018427387904
```
same with `status`:
```
state: RUNNING
owner_pid   : 2558
trigger_time: 24.127977530
tstamp      : 534.646497255
delay       : 1218
avail       : 15237
avail_max   : 15237
-----
hw_ptr      : 22513876
appl_ptr    : 22515023
```

perexg, 2021-06-14:
Could you dump the alsa-lib configuration using `aplay -v` (for both working / non-working state) ?

TheBitFighter, 2021-06-15:
I just managed to get noise on the first run, which I attribute to the lack of knowledge about aplay, however it worked on the first run and didn't on the second, same invocation command.
`aplay -v -f cdr Hypnotoad.mp3`
Output was again identical for both runs.
[aplay_after.txt](https://github.com/alsa-project/alsa-lib/files/6654020/aplay_after.txt)
[aplay_before.txt](https://github.com/alsa-project/alsa-lib/files/6654021/aplay_before.txt)

TheBitFighter, 2021-06-15:
I just managed to get noise on the first run, which I attribute to the lack of knowledge about aplay, however it worked on the first run and didn't on the second, same invocation command.
`aplay -v -f cdr Hypnotoad.mp3`
Output was again identical for both runs.
[aplay_after.txt](https://github.com/alsa-project/alsa-lib/files/6654144/aplay_after.txt)
[aplay_before.txt](https://github.com/alsa-project/alsa-lib/files/6654145/aplay_before.txt)
 
Edit: After this upgrade, I was no longer able to fix the issue by downgrading the alsa packages. I also downgraded alsa-utils and pipewire, no change. This is today's pacman log:
[pacman_log.txt](https://github.com/alsa-project/alsa-lib/files/6654168/pacman_log.txt)

perexg, 2021-06-15:
It looks like a driver (kernel) issue then.

TheBitFighter, 2021-06-15:
It very much seems to be a kernel Issue. Can someone point me to the right Mailinglist/Bugtracker to report this issue to before we close this one? I switched to the LTS kernel (5.10.42-1-lts), which works just fine in contrast to the previous kernel (5.12.9.arch1-1).

TheBitFighter, 2021-06-22:
I have an update. I have updated alsa again on the LTS kernel and the issue popped up again. This seems to be more than a kernel issue.

perexg, 2021-06-28:
We need to know the difference between 1.2.4-3 and 1.2.4-2 packages (which upstream source was used).

dvzrv, 2021-06-28:
@perexg In that version I switched from the github provided source tarball (used in `-2`) back to the source tarball provided on the ALSA website (used in `-3`): https://github.com/archlinux/svntogit-packages/commit/d06923812a5244f9431eace2f78b3d3ba0f78dda

IIRC the tags for 1.2.4 were moved (https://github.com/alsa-project/alsa-lib/issues/87#issuecomment-712366525) after creating them, so I believe the version used in `-2` was that before the version had been retagged on or after October 16th 2020 (if it has been retagged multiple times I have no idea which version it may relate to). Maybe you know which commit it has been.

perexg, 2021-11-05:
Does this issue persist with alsa-lib 1.2.5 ?

symdeb, 2022-05-14:
Hi. I have a 18i20 Gen 1 and a 2. They work fine on Ubuntu (now at 22.04).  But you may need to check the USB port. focurite interfaces seem to be very sensitive for the type of USB hardware. for Example, the 18i20 does not work great on a USB 3 port an the Gen 3 seems does not like USB 2. YMMV. I use a low latency kernel. Since past 4 or so Ubuntu versions (from 19.XX or so, I can't remember) the sound is always muted in the Gen 18i20 after boot.  After toggling "sync"  in alsa-mixer the sound is enabled. My PC is now exactly 10 years old  with Jack2 and a buffer of 64 and 3 cycles with zero drops. You could also look for more clues in the LinuxMusician in the computer Related Hardware section as there are some folks who build the Linux drivers for these interfaces.  

P.S: I would like the ALSA team:  How can I change or update  ALSA on my system w/o rebuilding the whole kernel ?

perexg, 2022-05-19:
> P.S: I would like the ALSA team: How can I change or update ALSA on my system w/o rebuilding the whole kernel ?

We don't provide the standalone builds now. Use the standard linux kernel build process.

I'm closing this bug, because it's not confirmed, that it's alsa-lib's related issue.

closed 2022-05-19 by perexg