alsa-project/alsa-lib #151: [Bug] Duplication in Array Element not Handled in Parsing conf

aiChaoSONG, 2021-06-10:
When parsing array element in conf files, an auto-increase number is used as config id(`snd_config_t->id`), and array element as the config value(`snd_config_t->u.string/integer/integer64/real`), duplication is not handle in `parse_array_defs/parse_array_def/parse_value` functions.

Suppose we have a conf:
```
array [
    "a"
    "b"
    "a"
]
```
the parsed result will be:
```
    parent
        |
        ---- snd_config_t {id: "0", u.string: "a"}
        |
        ---- snd_config_t {id: "1", u.string: "b"}
        |
        ---- snd_config_t {id: "2", u.string: "a"}
```
The duplication in "a" may be desired. But please consider this condition: suppose we have four conf files A, B, C and D, where A 
is the conf above, let B include A, C include A, D include B and C, so D include A twice indirectly. File A will be processed twice when parsing D, because duplication is not handled, here comes the problem, the parsed result of A in D is:
```
    parent
        |
        ---- snd_config_t {id: "0", u.string: "a"}
        |
        ---- snd_config_t {id: "1", u.string: "b"}
        |
        ---- snd_config_t {id: "2", u.string: "a"}
        |
        ---- snd_config_t {id: "3", u.string: "a"}
        |
        ---- snd_config_t {id: "4", u.string: "b"}
        |
        ---- snd_config_t {id: "5", u.string: "a"}
```
This is obviously wrong.

I think handle duplication in parse_value is not the solution, because element duplication in an array is wanted. The key problem here is that we parse the same conf more than once.

perexg, 2021-06-10:
The array merging is a function when the parsing mode is _merge_. You can force to override the configuration element using `!` prefix. You're probably talking about recent bug fixed in ddfc32abf5697de1618b9e7ffdf57a0f97013090 .

perexg, 2021-06-10:
So:

```
array [ a b c ]
!array [ d e f ]
```

aiChaoSONG, 2021-06-10:
@perexg Thank you, `!` is a magic operator for conf

closed 2021-06-10 by aiChaoSONG