alsa-project/alsa-lib #152: Logitech C922x microphone intermittently working

sambo57u, 2021-06-10:
I have a Logitech C922X (detected as C922) USB webcam. Running latest updates on Fedora 34. The microphone of this device only works about 1 out of 5 reboots. When not working if I install another microphone, activate that one, the unplug and plug
C922 back and switch to it,  it starts to work. Some kind of a timing issue that is very annoying. Don't have this problem with
Logitech C270. The camera always works fine.

perexg, 2021-06-14:
Please, do tests with the native ALSA applications (like arecord) without pulseaudio / pipewire to confirm that it's an ALSA problem.

sambo57u, 2021-06-14:
Hi, first,
-------------------------------
$ arecord --list-devices
**** List of CAPTURE Hardware Devices ****
card 0: Webcam [C922 Pro Stream Webcam], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: PCH [HDA Intel PCH], device 0: ALC3234 Analog [ALC3234 Analog]
  Subdevices: 1/1
------------------------------

The KDE pa is showing only one microphone

$ arecord test

This creates the file but then played with arecord it is silent (microphone is not muted).

sambo57u, 2021-06-14:
I also see these messages in /var/log/messages (tried analog and iec958 stereo)

alsactl[3674]: alsa-lib parser.c:242:(error_node) UCM is not supported for this HDA model (HDA Intel PCH at 0x38003ff20000 irq 141)
alsactl[3674]: alsa-lib main.c:1405:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -6
alsactl[3674]: alsa-lib parser.c:242:(error_node) UCM is not supported for this HDA model (HDA NVidia at 0xc5080000 irq 142)
alsactl[3674]: alsa-lib main.c:1405:(snd_use_case_mgr_open) error: failed to import hw:2 use case configuration -6
alsactl[3674]: alsa-lib main.c:1405:(snd_use_case_mgr_open) error: failed to import hw:3 use case configuration -2
pipewire[5025]: (alsa_output.pci-0000_d5_00.1.hdmi-stereo-extra2-49) XRun! rate:1024/48000 count:1 time:74327341 delay:16277327 max:16277327
plasmashell[4477]: org.kde.plasma.pulseaudio: No object for name "alsa_input.usb-046d_C922_Pro_Stream_Webcam_269190FF-02.analog-stereo"
plasmashell[4477]: org.kde.plasma.pulseaudio: No object for name "alsa_input.usb-046d_C922_Pro_Stream_Webcam_269190FF-02.iec958-stereo"
plasmashell[4477]: org.kde.plasma.pulseaudio: No object for name "alsa_input.usb-046d_C922_Pro_Stream_Webcam_269190FF-02.iec958-stereo"
plasmashell[4477]: org.kde.plasma.pulseaudio: No object for name "alsa_input.usb-046d_C922_Pro_Stream_Webcam_269190FF-02.analog-stereo"

PranavBhattarai, 2021-07-15:
I do see the same issue daily here in Fedora 34 Workstation.
```
pranav@fedora ~> sudo journalctl -b | grep -i -B 1 -A 1 alsa-lib
Jul 15 09:11:17 fedora alsactl[823]: alsactl 1.2.5.1 daemon started
Jul 15 09:11:17 fedora alsactl[823]: alsa-lib parser.c:242:(error_node) UCM is not supported for this HDA model (HDA Intel HDMI at 0xc1110000 irq 49)
Jul 15 09:11:17 fedora alsactl[823]: alsa-lib main.c:1405:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -6
Jul 15 09:11:17 fedora alsactl[823]: alsa-lib parser.c:242:(error_node) UCM is not supported for this HDA model (HDA Intel PCH at 0xc1114000 irq 50)
Jul 15 09:11:17 fedora alsactl[823]: alsa-lib main.c:1405:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -6
Jul 15 09:11:17 fedora smartd[781]: Configuration file /etc/smartmontools/smartd.conf was parsed, found DEVICESCAN, scanning devices
pranav@fedora ~> arecord --list-devices
**** List of CAPTURE Hardware Devices ****
card 1: PCH [HDA Intel PCH], device 0: ALC3234 Analog [ALC3234 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
GNOME logs also throws similar errors.
[image]

Device: Dell Inspiron 5548
Graphics: Mesa Intel® HD Graphics 5500 (BDW GT2)
Processor: Intel® Core™ i5-5200U CPU @ 2.20GHz × 4 

@perexg What does this error means?

perexg, 2021-11-05:
The UCM errors are harmless (and they are hidden in the current code). There's fallback to the legacy detection in pulseaudio.

The recording should be checked using the direct ALSA devices like: `arecord -Dplughw:0,0 -f dat a.wav` where 0,0 is CARD,DEVICE combination.

PranavBhattarai, 2021-11-05:
_[If I'm not being noob enough, pardon me]_

I downloaded an mp3 and converted that .mp3 to .wav and renamed the file to **a.wav**. Created a file named **dat** using the touch command. The command does work:

```
~/Downloads
  ⟹  arecord -Dplughw:0,0 -f dat a.wav
arecord: main:831: audio open error: No such file or directory
~/Downloads
  ⟹  arecord -Dplughw:0,0 -f a.wav
arecord: main:630: wrong extended format 'a.wav'
```

perexg, 2021-11-05:
You're trying to record (capture) something, so the arecord will capture `a.wav` file for you. You need to replace the zero numbers with the card and device numbers - hint - `arecord -l`.

PranavBhattarai, 2021-11-05:
```
  ⟹  arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: PCH [HDA Intel PCH], device 0: ALC3234 Analog [ALC3234 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
  ⟹  arecord -Dplughw:1,0 -f dat a.wav
Recording WAVE 'a.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
^Zfish: Job 1, 'arecord -Dplughw:1,0 -f d a.w…' has stopped
```

I see it recorded the an audio. What should I do with that audio file?

perexg, 2021-11-05:
> I see it recorded the an audio. What should I do with that audio file?

It's a recording test for ALSA PCM devices - the original issue was about the microphone.

The UCM problem is harmless - just ignore those lines in the log.

sambo57u, 2021-11-05:
Hello all, sorry for the delayed response. I have been struggling with this for months....the fact that one out of 7-8 reboots it worked was suggestive of some timing issue. So, I went into BIOS and instead of quick boot I chose 5s delay option (this is DELL precision and has 5s and 10s delay options, supposedly to see messages). This seems to have solved the problem for me. There was a quirk in the kernel for the webcam but somehow the microphone issue did not have to do with that.
I just discovered this last week....hope it helps others.

perexg, 2021-11-07:
Ok, closing. This problem appears like a kernel issue (initialization timing) so try to enter this kind of bugs to https://bugzilla.kernel.org - audio component - next time - with the kernel log.

closed 2021-11-07 by perexg

PranavBhattarai, 2021-11-09:
Just to make sure:

All the Alsa-Lib log which I mentioned above is **ACTUALLY** a Linux kernel problem, not the ALSA itself!!! 

_I'm preety sure _(like 90%)_ if I make a report in kernel bugzilla, I will be redirected here. And I won't even be surprised._

perexg, 2021-11-09:
> Just to make sure:
> 
> All the Alsa-Lib log which I mentioned above is **ACTUALLY** a Linux kernel problem, not the ALSA itself!!!
> 
> _I'm preety sure _(like 90%)_ if I make a report in kernel bugzilla, I will be redirected here. And I won't even be surprised._

The UCM errors are different issue. And belongs to alsa-lib. They will be fixed in 1.2.6.

PranavBhattarai, 2021-11-09:
Thank you. ciao.